espressif/esp-idf #10343: [Master: /examples/peripherals/i2s/] Example "i2s_recorder" will not compile

GTCLive, 2022-12-10:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

in "esp-idf/examples/peripherals/i2s/i2s_recorder/main)/CMakeLists.txt"
Cannot find: INCLUDE_DIRS "../../common"

1. The problem occurs only in the master branch.
2. Everything works and compile perfectly fine using the same example in "ESP-IDF Release v5.0"

suda-morris, 2022-12-12:
Hi @GTCLive 

That example relies on a header file which is located [here](https://github.com/espressif/esp-idf/tree/master/examples/peripherals/i2s/common). Maybe you copied the example to a new place, so the compiler can't find that header file by the **relative** path. 

For now, you can update your example's main CMakeLists.txt file by using **absolute** path, e.g.:

```diff
diff --git a/examples/peripherals/i2s/i2s_recorder/main/CMakeLists.txt b/examples/peripherals/i2s/i2s_recorder/main/CMakeLists.txt
index 44a620a03e..5c060abbce 100644
--- a/examples/peripherals/i2s/i2s_recorder/main/CMakeLists.txt
+++ b/examples/peripherals/i2s/i2s_recorder/main/CMakeLists.txt
@@ -1,2 +1,2 @@
 idf_component_register(SRCS "i2s_recorder_main.c"
-                    INCLUDE_DIRS "../../common")
+                    INCLUDE_DIRS "$ENV{IDF_PATH}/examples/peripherals/i2s/common")
```

GTCLive, 2022-12-13:
Hi @suda-morris 
Thanks for your help, absolute path you've proposed works perfectly in this case 👍

**Additional info:** I am using Eclipse and keeping this updated with the latest version of everything. I am creating the project using the recommended method: File > New > New IDF Project > Create Project Using one of the Template. 

Is it fair to assume that this "**compiler can't find that header file by the relative path**" issue may be related to the Eclipse Plugin rather? If you suspect so please let me know I'll close this one and pass this over to the Eclipse Plugin repo for them to investigate.

[image]

closed 2022-12-15 by GTCLive