espressif/esp-idf #10344: restful server npm build issue

fabo9573, 2022-12-10:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

does not apply

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

PowerShell

### What is the expected behavior?

clean build of npm generating html files in the front\web-demo\distr folder

### What is the actual behavior?

npm fails at npm run build with the error below.
I tried 
- npm config set legacy-peer-deps true
- tried all nodejs versions: 10.12, 12.19, 14.11, 18.12
 no one works. Attached log is for the 18.12. I tried playing around with the package.json to resolve conflicts but no luck so far.

Does anyone have a reccomendation on what nodejs and what npm version to use? Thank you very much.

 

### Steps to reproduce.

1. download repository
2. install nodejs (includes npm)
3. run npm install
4. run npm run build
...

### Build or installation Logs.

```plain
/  Building legacy bundle for production...

 ERROR  Failed to compile with 3 errors                                                                                                                                                      10:43:01

[eslint] Failed to load plugin 'node' declared in '.eslintrc.js » @vue/eslint-config-standard » C:\Users\Fabian\OneDrive\workspace\07_MCUs\03_Espressif\restful_server\front\web-demo\node_modules\eslint-config-standard\index.js': Cannot find module 'eslint-plugin-node'
Require stack:
- C:\Users\Fabian\OneDrive\workspace\07_MCUs\03_Espressif\restful_server\front\web-demo\__placeholder__.js

You may use special comments to disable some warnings.
Use // eslint-disable-next-line to ignore the next line.
Use /* eslint-disable */ to ignore all warnings in a file.
 error  in ./src/store.js

Module not found: Error: Can't resolve 'core-js/modules/es.array.push.js' in 'C:\Users\Fabian\OneDrive\workspace\07_MCUs\03_Espressif\restful_server\front\web-demo\src'

 error  in ./src/views/Light.vue?vue&type=template&id=71958402&

Module not found: Error: Can't resolve 'core-js/modules/es.array.concat.js' in 'C:\Users\Fabian\OneDrive\workspace\07_MCUs\03_Espressif\restful_server\front\web-demo\src\views'

 ERROR  Error: Build failed with errors.
Error: Build failed with errors.
    at C:\Users\Fabian\OneDrive\workspace\07_MCUs\03_Espressif\restful_server\front\web-demo\node_modules\@vue\cli-service\lib\commands\build\index.js:207:23
    at C:\Users\Fabian\OneDrive\workspace\07_MCUs\03_Espressif\restful_server\front\web-demo\node_modules\webpack\lib\webpack.js:148:8
    at C:\Users\Fabian\OneDrive\workspace\07_MCUs\03_Espressif\restful_server\front\web-demo\node_modules\webpack\lib\HookWebpackError.js:68:3
    at Hook.eval [as callAsync] (eval at create (C:\Users\Fabian\OneDrive\workspace\07_MCUs\03_Espressif\restful_server\front\web-demo\node_modules\tapable\lib\HookCodeFactory.js:33:10), <anonymous>:6:1)
    at Hook.CALL_ASYNC_DELEGATE [as _callAsync] (C:\Users\Fabian\OneDrive\workspace\07_MCUs\03_Espressif\restful_server\front\web-demo\node_modules\tapable\lib\Hook.js:18:14)
    at Cache.shutdown (C:\Users\Fabian\OneDrive\workspace\07_MCUs\03_Espressif\restful_server\front\web-demo\node_modules\webpack\lib\Cache.js:150:23)
    at C:\Users\Fabian\OneDrive\workspace\07_MCUs\03_Espressif\restful_server\front\web-demo\node_modules\webpack\lib\Compiler.js:1225:15
    at Hook.eval [as callAsync] (eval at create (C:\Users\Fabian\OneDrive\workspace\07_MCUs\03_Espressif\restful_server\front\web-demo\node_modules\tapable\lib\HookCodeFactory.js:33:10), <anonymous>:6:1)
    at Hook.CALL_ASYNC_DELEGATE [as _callAsync] (C:\Users\Fabian\OneDrive\workspace\07_MCUs\03_Espressif\restful_server\front\web-demo\node_modules\tapable\lib\Hook.js:18:14)
    at Compiler.close (C:\Users\Fabian\OneDrive\workspace\07_MCUs\03_Espressif\restful_server\front\web-demo\node_modules\webpack\lib\Compiler.js:1218:23)
PS C:\Users\Fabian\OneDrive\workspace\07_MCUs\03_Espressif\restful_server\front\web-demo>
```

### More Information.

{
  "name": "web-demo",
  "version": "0.1.0",
  "private": true,
  "scripts": {
    "serve": "vue-cli-service serve",
    "build": "vue-cli-service build",
    "lint": "vue-cli-service lint"
  },
  "dependencies": {
    "axios": "^0.21.1",
    "core-js": "^2.6.5",
    "eslint-plugin-flowtype": "^8.0.3",
    "eslint-plugin-import": "^2.26.0",
    "eslint-plugin-jsx-a11y": "^6.6.1",
    "eslint-plugin-react": "^7.31.11",
    "react-scripts": "^5.0.1",
    "vue": "^2.6.10",
    "vue-router": "^3.0.3",
    "vuetify": "^1.5.14",
    "vuex": "^3.0.1"
  },
  "devDependencies": {
    "@vue/cli-plugin-babel": "~5.0.0",
    "@vue/cli-plugin-eslint": "~5.0.0",
    "@vue/cli-service": "~5.0.0",
    "@vue/eslint-config-standard": "~5.0.0",
    "babel-eslint": "^10.0.1",
    "eslint": "7.32.0",
    "eslint-plugin-vue": "^7.0.0",
    "stylus": "^0.54.5",
    "stylus-loader": "^3.0.1",
    "vue-cli-plugin-vuetify": "^0.5.0",
    "vue-template-compiler": "^2.5.21",
    "vuetify-loader": "^1.0.5"
  }
}

hmalpani, 2022-12-13:
Hello @fabo9573 
I have following versions:
nodejs: v10.19.0
npm: 6.14.4

I was able to build HTML files successfully. Can you try with the above mentioned versions?

fabo9573, 2022-12-14:
Hello hmalpani,

thank you so much! With using exactly those versions, it also works for me! 

Thanks again for the quick help,

Best Regards

closed 2023-01-12 by Alvin1Zhang