espressif/esp-idf #10345: Windows 11 ARM64 Support

drony, 2022-12-10:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.0

### Operating System used.

Windows

### How did you build your project?

Other (please specify in More Information)

### If you are using Windows, please specify command line type.

CMD

### What is the expected behavior?

"python.exe C:\SysGCC\esp32\esp-idf\v5.0\tools\idf_tools.py install-python-env"

### What is the actual behavior?

ERROR: Platform Windows-ARM64 appears to be unsupported

### Steps to reproduce.

"python.exe C:\SysGCC\esp32\esp-idf\v5.0\tools\idf_tools.py install-python-env"

### Build or installation Logs.

```plain
ERROR: Platform Windows-ARM64 appears to be unsupported
```

### More Information.

ERROR: Platform Windows-ARM64 appears to be unsupported

dobairoland, 2022-12-10:
Hi @drony. I can't tell right now if this will be supported or when it is going to happen. But If you install all the necessary libraries and tools (based on suggestions from error messages) then you should be able to set it up and get it running for you.

drony, 2022-12-13:
I found a temporary solution.

Actually, Windows 11 ARM still emulates x64 applications. It just starts working once you make the necessary redirects in idf_tools.py

dobairoland, 2022-12-13:
@drony Great! Could you please submit a pull request?

dobairoland, 2023-08-30:
This fix will be available soon. It will be in v5.2 only (for now). If we get some requests and confirmation that this works as expected then we can backport it to previous releases (which are still in support period).

closed 2023-09-04 by espressif-bot

Basti77, 2025-03-30:
I'm facing the exact same issue. I'm using a modern MS Surface Pro and wanted to compile a project — but no luck, because there's no ARM64 support.

Seriously, ARM64 should be considered a standard by now. It's not some exotic architecture anymore — Apple has fully embraced it, and more and more devices are running on ARM. It's really frustrating that this still isn't supported.