espressif/esp-idf #10346: ESP32-S3/ESP32-S2的ULP-RISCV I2C中断始终未置位

yangdx2022, 2022-12-10:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1-dev-2186-g454aeb3a48

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

常规开发板

### Power Supply used.

USB

### What is the expected behavior?

使用ULP-RSIC-V中I2C例程操作SD2068

### What is the actual behavior?

我尝试使用ESP-IDF的ULP-RISCV中的ULP-I2C的程序测试低功耗I2C，但是发现在读取和写入的时候会卡在
`while (!REG_GET_FIELD(RTC_I2C_INT_ST_REG, RTC_I2C_TX_DATA_INT_ST)) {
            /* Minimal delay to avoid hogging the CPU */
            printf("waitting... ...\n");
            vTaskDelay(1000);
        }`，我尝试使用多个ESP32-S3发现没有效果，切换到ESP32-S2也没有效果，测试的I2C设备是SD2068，使用常规的I2C接口能够正常读写

### Steps to reproduce.

修改例程中设备地址与寄存器地址
调用写入或读取函数
开始测试

sudeep-mohanty, 2023-03-01:
Hello @yangdx2022,
Thank you for reporting this issue. Apologies for a delayed response.
This issue is similar to the one reported under #9990 and will be investigated under the same. Therefore, closing this issue. Thanks.

closed 2023-03-01 by sudeep-mohanty

sudeep-mohanty, 2025-01-24:
Fix merged at https://github.com/espressif/esp-idf/commit/e9a2f3b098e9a74202707bd2b0f5550812605697. Closing this ticket. Please reopen the ticket or open a new one for further support.