espressif/esp-idf #10349: SPI_HOST_MAX for ESP32C2 is incorrect

nopnop2002, 2022-12-11:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

ESP32C2 has hardware limitations compared to ESP32.

The number of hardware can be checked with the following constants.

But the SPI_HOST_MAX of the ESP32C2 is incorrect.

```
$ idf.py --version
ESP-IDF v5.1-dev-1840-gaad600c9ca-dirty
```

# ESP32
SPI_HOST_MAX=3
I2C_NUM_MAX=2
UART_NUM_MAX=3

# ESP32C2
SPI_HOST_MAX=3 ---> incorrect
I2C_NUM_MAX=1
UART_NUM_MAX=2

suda-morris, 2022-12-12:
Hi @nopnop2002 Thanks for reporting.

Yes, the enum `spi_host_device_t` should reflect the `SOC_SPI_PERIPH_NUM` macro, we will fix it ASAP.

nopnop2002, 2022-12-12:
@suda-morris 

The ESP32C2 and ESP32C3 documents have the following description.

This description may be misleading as SPI3 is available.

[image]

[image]

AxelLin, 2022-12-23:
This was reported 1 year ago: https://github.com/espressif/esp-idf/issues/8144

closed 2023-01-13 by espressif-bot