espressif/esp-idf #10350: what should I do?

FTarpa, 2022-12-11:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1-dev-2186-g454aeb3

### Operating System used.

Linux

### How did you build your project?

Command line with Make

### If you are using Windows, please specify command line type.

CMD

### What is the expected behavior?

Detecting Python interpreter
Checking "python3" ...
Checking "python" ...
Checking "python3.8" ...
Python 3.8.9
"python3.8" has been detected
Installing ESP-IDF tools
Current system platform: linux-amd64
Selected targets are: esp32c2, esp32s2, esp32h4, esp32c6, esp32, esp32s3, esp32c3
Installing tools: xtensa-esp-elf-gdb, riscv32-esp-elf-gdb, xtensa-esp32-elf, xtensa-esp32s2-elf, xtensa-esp32s3-elf, riscv32-esp-elf, esp32ulp-elf, openocd-esp32, esp-rom-elfs
Skipping xtensa-esp-elf-gdb@12.1_20221002 (already installed)
Skipping riscv32-esp-elf-gdb@12.1_20221002 (already installed)
WARNING: directory for tool xtensa-esp32-elf version esp-2022r1-11.2.0 is present, but tool was not found
Installing xtensa-esp32-elf@esp-2022r1-11.2.0
file xtensa-esp32-elf-gcc11_2_0-esp-2022r1-linux-amd64.tar.xz is already downloaded
WARNING: destination path already exists, removing
Extracting /home/futao/.espressif/dist/xtensa-esp32-elf-gcc11_2_0-esp-2022r1-linux-amd64.tar.xz to /home/futao/.espressif/tools/xtensa-esp32-elf/esp-2022r1-11.2.0
Traceback (most recent call last):
  File "/home/futao/linux/ESP32_Workspace/esp-idf/tools/idf_tools.py", line 2610, in <module>
    main(sys.argv[1:])
  File "/home/futao/linux/ESP32_Workspace/esp-idf/tools/idf_tools.py", line 2602, in main
    action_func(args)
  File "/home/futao/linux/ESP32_Workspace/esp-idf/tools/idf_tools.py", line 1860, in action_install
    tool_obj.install(tool_version)
  File "/home/futao/linux/ESP32_Workspace/esp-idf/tools/idf_tools.py", line 812, in install
    unpack(archive_path, dest_dir)
  File "/home/futao/linux/ESP32_Workspace/esp-idf/tools/idf_tools.py", line 359, in unpack
    archive_obj = tarfile.open(filename, 'r:xz')
  File "/usr/local/python/python3.8/lib/python3.8/tarfile.py", line 1621, in open
    return func(name, filemode, fileobj, **kwargs)
  File "/usr/local/python/python3.8/lib/python3.8/tarfile.py", line 1730, in xzopen
    t = cls.taropen(name, mode, fileobj, **kwargs)
  File "/usr/local/python/python3.8/lib/python3.8/tarfile.py", line 1651, in taropen
    return cls(name, mode, fileobj, **kwargs)
  File "/usr/local/python/python3.8/lib/python3.8/tarfile.py", line 1514, in __init__
    self.firstmember = self.next()
  File "/usr/local/python/python3.8/lib/python3.8/tarfile.py", line 2318, in next
    tarinfo = self.tarinfo.fromtarfile(self)
  File "/usr/local/python/python3.8/lib/python3.8/tarfile.py", line 1104, in fromtarfile
    buf = tarfile.fileobj.read(BLOCKSIZE)
  File "/usr/local/python/python3.8/lib/python3.8/lzma.py", line 205, in read
    return self._buffer.read(size)
  File "/usr/local/python/python3.8/lib/python3.8/_compression.py", line 68, in readinto
    data = self.read(len(byte_view))
  File "/usr/local/python/python3.8/lib/python3.8/_compression.py", line 96, in read
    if self._decompressor.needs_input:
[... 177 lines cut ...]
  File "/home/futao/linux/ESP32_Workspace/esp-idf/tools/idf_tools.py", line 1860, in action_install
    tool_obj.install(tool_version)
  File "/home/futao/linux/ESP32_Workspace/esp-idf/tools/idf_tools.py", line 812, in install
    unpack(archive_path, dest_dir)
  File "/home/futao/linux/ESP32_Workspace/esp-idf/tools/idf_tools.py", line 359, in unpack
    archive_obj = tarfile.open(filename, 'r:xz')
  File "/usr/local/python/python3.8/lib/python3.8/tarfile.py", line 1621, in open
    return func(name, filemode, fileobj, **kwargs)
  File "/usr/local/python/python3.8/lib/python3.8/tarfile.py", line 1730, in xzopen
    t = cls.taropen(name, mode, fileobj, **kwargs)
  File "/usr/local/python/python3.8/lib/python3.8/tarfile.py", line 1651, in taropen
    return cls(name, mode, fileobj, **kwargs)
  File "/usr/local/python/python3.8/lib/python3.8/tarfile.py", line 1514, in __init__
    self.firstmember = self.next()
  File "/usr/local/python/python3.8/lib/python3.8/tarfile.py", line 2318, in next
    tarinfo = self.tarinfo.fromtarfile(self)
  File "/usr/local/python/python3.8/lib/python3.8/tarfile.py", line 1104, in fromtarfile
    buf = tarfile.fileobj.read(BLOCKSIZE)
  File "/usr/local/python/python3.8/lib/python3.8/lzma.py", line 205, in read
    return self._buffer.read(size)
  File "/usr/local/python/python3.8/lib/python3.8/_compression.py", line 68, in readinto
    data = self.read(len(byte_view))
  File "/usr/local/python/python3.8/lib/python3.8/_compression.py", line 96, in read
    if self._decompressor.needs_input:
AttributeError: '_lzma.LZMADecompressor' object has no attribute 'needs_input'

dobairoland, 2022-12-11:
The same error can be found several times on the Internet with no clear indication about the root cause. They have solved the issue by changing Python to a newer version. 

I don't see changes in `tarfile.open()` arguments since Python 3.6 and the `needs_input` attribute should be available since 3.5 with no indication about later changes. Your 3.8 should be good enough. So this is a mystery to me.

chipweinberger, 2022-12-12:
I use python 3.9.9. And recommend it.

FTarpa, 2022-12-12:
I know why, and resolved it

before run  **./install.sh** cmd , I run **git branch --track  v5.0 remotes/origin/release/v5.0**  cmd

then everything is all right !

dobairoland, 2022-12-12:
The same archive is in the v5.0 release and there is no significant difference between the tools. So probably recreating the virtual environment helped.

If anyone else encounters this issue in the future then please try to remove `~/.espressif/python_env` and re-run the installer.

There is nothing else left to be done regarding this. I'm closing it.

closed 2022-12-12 by dobairoland