espressif/esp-idf #10352: [Docs] ESP32_WIFI_MGMT_SBUF_NUM is undocumented

chipweinberger, 2022-12-12:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

> WiFi mgmt short buffer number
> 
> Range:
> from 6 to 32
> 
> Default value:
> 32

The docs are very sparse here. 
- How big are these buffer? 
- Where are they allocated? 
- How much does this value affect perf? 
- What are they used for?

Thanks.

hansw123, 2024-03-27:
@chipweinberger 
sorry for late reply
We'll add a more detailed description in a later version.

Alvin1Zhang, 2024-12-17:
Thanks for reporting, the fix is available at https://github.com/espressif/esp-idf/commit/6673376297b921d438790d195014b860eaf8bb17, and has been back ported to branches until 5.0. Feel free to reopen.

closed 2024-12-17 by Alvin1Zhang