espressif/esp-idf #10353: What I can confirm the esp32s3 fpu is running?

WangShuoran, 2022-12-12:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

When I search ESP-IDF doc, it show https://docs.espressif.com/projects/esp-idf/zh_CN/v5.0/esp32s3/api-guides/freertos-smp.html#floating-point-usage , but I don't know the math.h sinf() can or not use FPU, I am write any example try to use FPU:
```c
#define ANGLES_TABLE_SIZE 2000
    clock_t start, stop;
    float tmp = 0;
    float *angle_table_f = (float *)malloc(sizeof(float) * ANGLES_TABLE_SIZE);
    for (size_t i = 0; i < 2000; ++i) {
        esp_fill_random(&angle_table_f[i], 4);
    }
    start = clock();
    for (size_t i = 0; i < 20000; ++i) {
        tmp += sinf(angle_table_f[i%2000]);
    }
    stop = clock();
    printf("test_sin_float:%ld\n", (stop - start));
```
But the time is same with last code snippet:
```c
#define ANGLES_TABLE_SIZE 2000
    clock_t start, stop;
    float tmp = 0;
    double *angle_table_d = (double *)malloc(sizeof(double) * ANGLES_TABLE_SIZE);
    for (size_t i = 0; i < 2000; ++i) {
        esp_fill_random(&angle_table_d[i], 8);
    }
    start = clock();
    for (size_t i = 0; i < 20000; ++i) {
        tmp += sin(angle_table_f[i%2000]);
    }
    stop = clock();
    printf("test_sin_float:%ld\n", (stop - start));
```
And I know the FPU in stm32 need data alignment, And I try to use not data alignment code to run, but the time is same with last two code snippets.

I don't know why.

I need confirm those question:
> how to check FPU is already used?
> Does FPU need data alignment?
> Why float running time is same with double?

WangShuoran, 2022-12-14:
@Dazza0 Thank you for quick reply!
I am test all of your suggest:
1. When I printf float, the `stop-start` clock is not zero, so I think as your say that compile question.
2. When I objdump the app_main.c file with `xtensa-esp32s3-elf\bin\xtensa-esp32s3-elf-gcc.exe -S -o main.s main/sntp_example_main.c`, the output show that: `add.s	f0, f1, f0` but if I try to use double the same one is not exist, so I think as your say FPU is already running.
3. When I use esp_fill_random() is to fill the float content, I don't consider IEEE 754 compatibility issues, thank you.

I have any other question:
1. When I use STM32 with have FPU, if I don't let data alignment, the program is show "UNALIGNED" fault, example is https://www.armbbs.cn/forum.php?mod=viewthread&tid=109400, but ESP32S3 is ok, Can you give any introduce in the document? Why ESP32S3 don't show error, Internal principles? or more advanced?
2. When I see IDF5.0 https://docs.espressif.com/projects/esp-idf/en/v5.0/esp32c3/migration-guides/release-5.x/gcc.html show no extra penalty, but almost network resources show non data alignment will have any performance waste. Why ESP32 series have no extra penalty?

WangShuoran, 2022-12-15:
@Dazza0 Thank you for quick reply!
But the relevant excerpt show non-aligned data access will cause a CPU exception. This means that if the access is not aligned, it will cause an exception! Is the document wrong, or do I understand it wrong?
这个意思是如果不对齐访问会导致异常呀！
[image]

jcmvbkbc, 2022-12-15:
> Likewise, all of the ESP32 chips support unaligned access at the hardware level, so there is no performance overhead thus why the warning can be ignored.

This is not what I observe. I run the following simple program on ESP32:

```
static void time_int(void *addr)
{       
        uint32_t cc0, cc1, tmp;
        uint32_t v;
        asm volatile ("rsil %[flags], 15\n\t"
                      "l32i %[v], %[addr], 0\t\n"
                      "l32i %[v], %[addr], 4\t\n"
                      "l32i %[v], %[addr], 8\t\n"
                      "l32i %[v], %[addr], 12\t\n"
                      "rsr %[ccount0], ccount\t\n"
                      "l32i %[v], %[addr], 0\t\n"
                      "l32i %[v], %[addr], 4\t\n"
                      "l32i %[v], %[addr], 8\t\n"
                      "l32i %[v], %[addr], 12\t\n"
                      "rsr %[ccount1], ccount\t\n"
                      "wsr %[flags], ps\n\t"
                      : [ccount0] "=&a" (cc0),
                      [ccount1] "=&a" (cc1),
                      [v] "=&a" (v),
                      [flags] "=&a" (tmp)
                      : [addr] "r" (addr)); 
        printf("integer: p = %p, t = %d\n", addr, cc1 - cc0);
}

static void time_float(void *addr)
{
        uint32_t cc0, cc1, tmp;
        float v;
        asm volatile ("rsil %[flags], 15\n\t"
                      "lsi %[v], %[addr], 0\t\n"
                      "lsi %[v], %[addr], 4\t\n"
                      "lsi %[v], %[addr], 8\t\n"
                      "lsi %[v], %[addr], 12\t\n"
                      "rsr %[ccount0], ccount\t\n"
                      "lsi %[v], %[addr], 0\t\n"
                      "lsi %[v], %[addr], 4\t\n"
                      "lsi %[v], %[addr], 8\t\n"
                      "lsi %[v], %[addr], 12\t\n"
                      "rsr %[ccount1], ccount\t\n"
                      "wsr %[flags], ps\n\t"
                      : [ccount0] "=&a" (cc0),
                      [ccount1] "=&a" (cc1),
                      [v] "=f" (v),
                      [flags] "=&a" (tmp)
                      : [addr] "r" (addr));
        printf("float: p = %p, t = %d\n", addr, cc1 - cc0);
}

void app_main(void)
{
        uint32_t iv[5] = {0};
        float fv[5] = {0};

        time_int(iv);
        time_int(1 + (void *)iv);
        time_float(fv);
        time_float(1 + (void *)fv);
}
```
and its output is the following:
```
integer: p = 0x3ffb5970, t = 5
integer: p = 0x3ffb5971, t = 33
float: p = 0x3ffb5960, t = 5
float: p = 0x3ffb5961, t = 35
```

which indicates that unaligned access in hardware is ~8 times slower than aligned, both for integer and FPU registers.

jcmvbkbc, 2022-12-15:
> The FPU does not read/write memory directly.

@Dazza0 this is not correct, FPU has the following instructions that directly read FP registers from memory: lsi, lsip, lsx, lsxp, and the following instructions that directly write FP registers into memory: ssi, ssip, ssx, ssxp.

WangShuoran, 2022-12-20:
@Dazza0 When I look menuconfig, I find Component config -> ESP System Settings -> Cache config:
[image]
Does those menu config the data alignment?

Dazza0, 2023-01-04:
@WangShuoran

> Does those menu config the data alignment?

Those configuration options configure the cache for externally connected SPI flash/psram. However, from the CPU's perspective, the cache is simply a region with it's virtual address range that should allow for unaligned access (though the unaligned access may end up causing multiple cache misses).

Alvin1Zhang, 2023-02-16:
@WangShuoran Thanks for reporting, would you please help share if any further updates for the issue? Thanks.

WangShuoran, 2023-02-23:
@Alvin1Zhang

I have several verified results, such as

@Alvin1Zhang

As a result of verification, the non-aligned data will certainly be slower than the aligned data.

Therefore, IDF documents https://docs.espressif.com/projects/esp-idf/en/v5.0/esp32c3/migration-guides/release-5.x/gcc.html There must be some problems. Please confirm?

---That is chinese:
@Alvin1Zhang 
我有几个验证后的结果，正如
@Alvin1Zhang 
验证后的结果，非对齐的数据在执行时肯定会比对齐的慢。
所以，IDF文档https://docs.espressif.com/projects/esp-idf/en/v5.0/esp32c3/migration-guides/release-5.x/gcc.html肯定有点问题。请确认？

WangShuoran, 2023-02-23:
@jcmvbkbc and @Dazza0 
关于下面问题还没有一个确定的回答：

> > The FPU does not read/write memory directly.FPU 不直接读写内存。
> 
> @Dazza0 this is not correct, FPU has the following instructions that directly read FP registers from memory: lsi, lsip, lsx, lsxp, and the following instructions that directly write FP registers into memory: ssi, ssip, ssx, ssxp.
> 
> @ Dazza0这是不正确的，FPU 有以下指令直接从内存读取 FP 寄存器: lsi，lsip，lsx，lsxp，以及以下指令直接将 FP 寄存器写入内存: ssi，ssip，ssx，ssxp。

WangShuoran, 2023-02-23:
@Alvin1Zhang 你好，下面的这个回答

> @Dazza0 Thank you for quick reply! But the relevant excerpt show non-aligned data access will cause a CPU exception. This means that if the access is not aligned, it will cause an exception! Is the document wrong, or do I understand it wrong? 这个意思是如果不对齐访问会导致异常呀！ [image]
> 
> @ Dazza0谢谢你的快速回复！但相关摘录显示，不对齐的数据访问将导致 CPU 异常。这意味着如果访问没有对齐，将导致异常！是文档错了，还是我理解错了？这个意思是如果不对齐访问会导致异常呀！

和

> @WangShuoran
> 
> @ 王朔然
> 
> > 1. When I use STM32 with have FPU, if I don't let data alignment, the program is show "UNALIGNED" fault, example is  当我使用具有 FPU 的 STM32时，如果我不让数据对齐，程序就显示“ UNALIGNED”故障，例如https://www.armbbs.cn/forum.php?mod=viewthread&tid=109400, but ESP32S3 is ok, Can you give any introduce in the document? Why ESP32S3 don't show error, Internal principles? or more advanced? ，但 ESP32S3是确定的，你能给任何介绍在文件中？为什么 ESP32S3不显示错误，内部原则？还是更先进？
> 
> That's because the ESP32-S3 supports unaligned access on the CPU's data bus. The CPU buses and their alignments is covered in Chapter 4 of the ESP32-S3 TRM ([EN](https://www.espressif.com/sites/default/files/documentation/esp32-s3_technical_reference_manual_en.pdf#section.4), [中](https://www.espressif.com/sites/default/files/documentation/esp32-s3_technical_reference_manual_cn.pdf#section.4)). Here is the relevant excerpt:
> 
> 这是因为 ESP32-S3支持 CPU 数据总线上的未对齐访问。ESP32-S3 TRM (EN，中)的第4章介绍了 CPU 总线及其校准。以下是相关摘录:
> 
> > Both data bus and instruction bus are little-endian. The CPU can access data via the data bus using single-byte, double-byte, 4-byte and 16-byte alignment. The CPU can also access data via the instruction bus, but only in 4-byte aligned manner; non-aligned data access will cause a CPU exception.数据总线和指令总线都是小端点的。CPU 可以使用单字节、双字节、4字节和16字节对齐方式通过数据总线访问数据。CPU 也可以通过指令总线访问数据，但只能以4字节对齐的方式; 不对齐的数据访问将导致 CPU 异常。
> 
> > 2. When I see IDF5.0  当我看到 IDF5.0https://docs.espressif.com/projects/esp-idf/en/v5.0/esp32c3/migration-guides/release-5.x/gcc.html show no extra penalty, but almost network resources show non data alignment will have any performance waste. Why ESP32 series have no extra penalty? 显示没有额外的惩罚，但几乎网络资源显示非数据对齐将有任何性能浪费。为什么 ESP32系列没有额外的惩罚？
> 
> Likewise, all of the ESP32 chips support unaligned access at the hardware level, so there is no performance overhead thus why the warning can be ignored.
> 
> 同样，所有的 ESP32芯片都支持硬件级别的未对齐访问，因此没有性能开销，因此可以忽略警告。

这个回答存在冲突，请确认？

WangShuoran, 2023-02-23:
> @Dazza0 When I look menuconfig, I find Component config -> ESP System Settings -> Cache config: [image] Does those menu config the data alignment?
> 
> @ Dazza0当我查看 menuconfig 时，我发现 Component config-> ESP System Settings-> Cache config: 这些菜单是否配置了数据对齐？

@Alvin1Zhang ，因为这个issue中的多个相关问题的回答没有一致，甚至完全相反。
所以，这个问题我要确认@Dazza0 的回答的正确性？
并且，请问留个我们这些配置有什么用(抱歉我没有读懂相关文档的意思)？

Dazza0, 2023-02-28:
@jcmvbkbc 

> This is not what I observe. I run the following simple program on ESP32:

On the ESP32, there should be some hardware overhead with unaligned access, but not any software overhead. When accessing a unaligned address (e.g., read), the memory bus will transparently split the access into two aligned accesses, and then combine them into the unaligned result. However, from the CPU's perspective, it is still a single `l32`/`s32` instruction. Thus the compiler will not add any software overhead (i.e., .multiple `l32`/s32` calls with bit shifting/masking).

Dazza0, 2023-02-28:
@WangShuoran 

> 请问留个我们这些配置有什么用(抱歉我没有读懂相关文档的意思)？

这些 config 跟 aligned/unaligned access 没有关系。这些 config 是用来配置 ESP32 的 cache （[缓存](https://baike.baidu.com/item/%E7%BC%93%E5%AD%98/100710?fr=aladdin)）。

jcmvbkbc, 2023-02-28:
> However, from the CPU's perspective, it is still a single l32/s32 instruction.
> Thus the compiler will not add any software overhead (i.e., .multiple l32/s32` calls with bit shifting/masking).

@Dazza0 In this statement 'thus' does not follow from 'however'. In my example above the compiler will not add anything because unaligned access is written in inline assembly. But if you let the compiler itself generate the code that access data that the compiler knows is not properly aligned it will happily do that. E.g. for the following program
```
struct S {
        char c;
        float f;
} __attribute__((packed));

struct S s;

void foo(float v)
{
        s.f = v;
}

float bar(void)
{
        return s.f;
}
```
the ESP32 compiler from 2021r2-patch3 invoked with `-O2` generates the following code, where one can clearly see multiple load and store instructions for a single unaligned piece of data:

```
        .file   "packed-unaligned.c"
        .text
        .literal_position
        .literal .LC0, s
        .align  4
        .global foo
        .type   foo, @function
foo:
        entry   sp, 32
        l32r    a9, .LC0
        extui   a12, a2, 8, 8
        extui   a11, a2, 16, 8
        extui   a10, a2, 24, 8
        s8i     a2, a9, 1
        s8i     a12, a9, 2
        s8i     a11, a9, 3
        s8i     a10, a9, 4
        retw.n
        .size   foo, .-foo
        .literal_position
        .literal .LC1, s
        .align  4
        .global bar
        .type   bar, @function
bar:
        entry   sp, 32
        l32r    a10, .LC1
        l8ui    a9, a10, 2
        l8ui    a11, a10, 1
        l8ui    a8, a10, 3
        l8ui    a2, a10, 4
        slli    a9, a9, 8
        or      a9, a9, a11
        slli    a8, a8, 16
        or      a8, a8, a9
        slli    a2, a2, 24
        or      a2, a2, a8
        retw.n
        .size   bar, .-bar
        .comm   s,5,4
        .ident  "GCC: (crosstool-NG esp-2021r2-patch3) 8.4.0"
```

jcmvbkbc, 2023-03-01:
> the ESP32 compiler from 2021r2-patch3 invoked with -O2 generates the following code

I've posted a possible fix for that: https://gcc.gnu.org/pipermail/gcc-patches/2023-February/612987.html

Dazza0, 2023-03-02:
@jcmvbkbc I just had another look at the Xtensa ISA reference manual and found this line regarding the `s32i` instruction:

> Without the Unaligned Exception Option (page 116), the two least significant bits of the address are ignored. A reference to an address that is not 0 mod 4 produces the same result as a reference to the address with the least significant bits cleared.

Could this be why the compiler is using four `s8i` instead?

jcmvbkbc, 2023-03-02:
> Could this be why the compiler is using four s8i instead?

No. It is using s8i because it has a built-in assumption that the code generated for xtensa requires strict alignment, [see gcc/config/xtensa/xtensa.h](https://github.com/gcc-mirror/gcc/blob/releases/gcc-8.4.0/gcc/config/xtensa/xtensa.h#L151).
The fix that I mentioned adjusts this definition to account for the capability to access unaligned data in hardware.

igrr, 2023-03-02:
One thing to note about the "Hardware Alignment Option", quoting the ISA reference manual:

A LoadStoreAlignmentCause exception may still be raised in some implementations with the Hardware Alignment Option if the address of a load or store instruction is not a multiple of its size and any of the following conditions is also true:
...
2. The memory type for either portion is XLMI, IRAM, or IROM.
...

I think this point could be an issue. It's probably not possible to determine whether the address is going to point to XLMI, IRAM/IROM or DRAM/DROM at compilation time. The compiler might consider that an unaligned access to some variable is allowed but the variable could be placed into, say, IRAM, by the linker script. Another case is the ESP32-S2, where due to the lack of data address space some portions of PSRAM would be accessed over XLMI (https://github.com/espressif/esp-idf/issues/8295).

jcmvbkbc, 2023-03-02:
> The compiler might consider that an unaligned access to some variable is allowed but the variable could be placed into, say, IRAM, by the linker script.

Then s8i wouldn't work either, because IRAM only supports 32-bit access, right?

> Another case is the ESP32-S2, where due to the lack of data address space some portions of PSRAM would be accessed over XLMI 

That's unfortunate.

I guess a compiler switch `m[no-]strict-align` that controls whether unaligned access is allowed might be useful, and the question is what it should default to. Defaulting to `mno-strict-align` when there's HW support for unaligned access and to `mstrict-align` otherwise looks reasonable to me. WDYT?

igrr, 2023-03-02:
> Then s8i wouldn't work either, because IRAM only supports 32-bit access, right?

Yes, that's correct! So IRAM is indeed not relevant to the discussion, and we do already have an unaligned load/store exception handler for that, IIRC.

> Defaulting to mno-strict-align when there's HW support for unaligned access and to mstrict-align otherwise looks reasonable to me.

Looks reasonable to me as well!

We can use this switch in the SDK depending on the chip and on the hardware features enabled. (E.g. ESP32-S2 with PSRAM Kconfig option enabled => -mstrict-align. ESP32-S3 => -mno-strict-align.)

jcmvbkbc, 2023-03-02:
> ESP32-S3 => -mno-strict-align

or nothing at all, and the compiler would choose `-mno-strict-align` from the core configuration data.
Ok, I'll make v2.

jcmvbkbc, 2023-04-19:
I've committed the following version for the gcc-14: https://gcc.gnu.org/pipermail/gcc-patches/2023-April/616238.html