espressif/esp-idf #10354: mbedtls_ssl stuck issue during weak WiFi internet availability

hardik-dozee, 2022-12-12:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.3

### Operating System used.

Linux

### How did you build your project?

Command line with Make

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32 Board

### Power Supply used.

External 5V

### What is the expected behavior?

Facing issue with wifi no internet whenever is available during file upload operation the mbedtls stucks for more than 30 minutes.
Can someone help to fix this issue or let me know how can I make the mbedlts timeout shorter than 2 minutes. 

I implemented the watchdog but I donot want to restart the esp due to data loss issues with other task. 

`
int server_init(void)
{
	printf("Session InIT\n");
	int ret = 0;
	mbedtls_net_init(&server_fd);
	mbedtls_ssl_init(&ssl);
	mbedtls_ssl_config_init(&conf);
	mbedtls_x509_crt_init(&cacert);
	mbedtls_x509_crt_init(&clcert);
	mbedtls_pk_init(&clkey);
	mbedtls_ctr_drbg_init(&ctr_drbg);
	mbedtls_entropy_init(&entropy);

	if((ret = mbedtls_ctr_drbg_seed(&ctr_drbg, mbedtls_entropy_func, &entropy, NULL, 0)) != 0)
	{
		if(enable_debug) printf("mbedtls_ctr_drbg_seed returned %d", ret);
		return ret;
	}
	if(enable_debug) printf("  . Parsing client certificate ...");

	ret = mbedtls_x509_crt_parse(&clcert, (const unsigned char *)client_cert, strlen(client_cert) + 1);
	if(ret < 0)
	{
		if(enable_debug) printf(" failed\n ! mbedtls_x509_crt_parse returned -0x%x\n\n", -ret);
		return ret;
	}
	if(enable_debug){
		printf(" ok\n");
		printf("  . Parsing client private key ...");
	}

	ret = mbedtls_pk_parse_key(&clkey, (const unsigned char *)client_key, strlen(client_key) + 1, (const unsigned char *) "dozee", 5);
	if(ret != 0)
    {
        if(enable_debug) printf(" failed\n ! mbedtls_pk_parse_key returned -0x%x\n\n", -ret);
        return ret;
    }

	if(enable_debug) printf(" ok\n");

    if(enable_debug) printf("  . SSL/TLS config setup ...");
    if((ret = mbedtls_ssl_config_defaults(&conf,
                    MBEDTLS_SSL_IS_CLIENT,
                    MBEDTLS_SSL_TRANSPORT_STREAM,
                    MBEDTLS_SSL_PRESET_DEFAULT)) != 0)
    {
        if(enable_debug) printf(" failed\n  ! mbedtls_ssl_config_defaults returned %d\n\n", ret);
        return ret;
    }
    if(enable_debug) printf(" ok\n");
	
	mbedtls_ssl_conf_authmode(&conf, MBEDTLS_SSL_VERIFY_OPTIONAL);
	mbedtls_ssl_conf_ca_chain(&conf, &cacert, NULL);
	mbedtls_ssl_conf_rng(&conf, mbedtls_ctr_drbg_random, &ctr_drbg);

	if(enable_debug) printf("  . Binding client certificate and key ...");

	ret = mbedtls_ssl_conf_own_cert(&conf, &clcert, &clkey);
	if (ret != 0) 
	{
		if(enable_debug) printf(" failed\n  ! mbedtls_ssl_conf_own_cert returned %d\n\n", ret);
		return ret;
	}
	if(enable_debug) printf(" ok\n");

	if(enable_debug) printf("  . SSL setup ...");
	ret = mbedtls_ssl_setup(&ssl, &conf);
	if (ret != 0)
	{
		if(enable_debug) printf(" failed\n  ! mbedtls_ssl_setup returned %x\n\n", ret);
		return ret;
	}
	if(enable_debug) printf(" Server Init ok %x\n",ret);
	//end of mbedtls ssl setup
	return ret;
}

int server_deinit(void)
{
	int ret = 0;
	if(enable_debug) printf(". closing ssl session...");
	ret = mbedtls_ssl_close_notify(&ssl);
	if(enable_debug) printf("ok . %d\n", ret);

	if(enable_debug) printf(". resetting ssl session...");
	ret = mbedtls_ssl_session_reset(&ssl);
	if(enable_debug) printf("ok . %d\n", ret);

	//if(enable_debug) printf(". mbedtls ssl free...");
    //mbedtls_ssl_free( &ssl );
    //if(enable_debug) printf("ok . %d\n", ret);

	if(enable_debug) printf(". mbedtls net free...");
	mbedtls_net_free(&server_fd);
	//mbedtls_ssl_free(&ssl);
	if(enable_debug) printf("ok . %d\n", ret);
	
   // mbedtls_x509_crt_free( &cacert );
    //mbedtls_ssl_free( &ssl );
    //mbedtls_ssl_config_free( &conf );
    //mbedtls_ctr_drbg_free( &ctr_drbg );
    //mbedtls_entropy_free( &entropy );

	//if(enable_debug)	
	//	mbedtls_cleanup(ssl);	
	printf("Last one..\n");
	return ret;
}

publish_data_to_server()
{
	if(enable_debug) printf("  . Setting hostname ...");
	if(((ret = mbedtls_ssl_set_hostname(&ssl, hostname )) != 0))
	{
		if(enable_debug) printf(" failed\n  ! mbedtls_ssl_set_hostname returned %x\n\n", ret);
		retFn = FAILED_TO_SET_HOSTNAME_PUB_DATA_SERVER;
		server_deinit();
		return retFn;
	}

	if(enable_debug)
	{
		printf(" ok\n");
		printf("  . Connecting to %s:%s...", hostname, "443\n");
	}

	if((ret = mbedtls_net_connect(&server_fd, hostname,"443", MBEDTLS_NET_PROTO_TCP)) != 0)
	{
		if(enable_debug) printf(" failed\n  ! mbedtls_net_connect returned %x\n\n", ret);
		retFn = FAILED_TO_CONNECT_TO_HOST_PUB_DATA_SERVER;
		server_deinit();
		return retFn;
	}

	if(enable_debug) printf(" ok\n");

	mbedtls_ssl_set_bio(&ssl, &server_fd, mbedtls_net_send, mbedtls_net_recv, NULL);
	if(enable_debug)
	{
		printf(" ok\n");
		printf("  . Performing the SSL/TLS handshake...\n");
	}

	while((ret = mbedtls_ssl_handshake(&ssl)) != 0)
	{
		printf("I am coming here\n");
		if(ret != MBEDTLS_ERR_SSL_WANT_READ && ret != MBEDTLS_ERR_SSL_WANT_WRITE)
		{
			if(enable_debug) printf(" failed\n  ! mbedtls_ssl_handshake returned -0x%x\n\n", -ret);
			retFn = FAILED_TO_FETCH_FW_API_PUB_DATA_SERVER;
			server_deinit();
			return retFn;
		}
	}

	if(enable_debug) printf("  . Writing HTTP header to SSL socket ... %s\n", payload);

	while((ret = mbedtls_ssl_write(&ssl, (unsigned char *) payload, strlen(payload))) <= 0)
	{
		if(ret != MBEDTLS_ERR_SSL_WANT_READ && ret != MBEDTLS_ERR_SSL_WANT_WRITE)
		{
			if(enable_debug) printf(" failed\n  ! mbedtls_ssl_write returned %d\n\n", ret);
			retFn = FAILED_TO_POST_INFO_PUB_DATA_SERVER;
			server_deinit();
			return retFn;
		}
	}

			while((ret = mbedtls_ssl_write(&ssl, (unsigned char *) payload, num_bytes1)) <= 0)
			{
				printf("Stuck\n");
				if(ret != MBEDTLS_ERR_SSL_WANT_READ && ret != MBEDTLS_ERR_SSL_WANT_WRITE)
				{
					if(enable_debug) printf(" failed\n  ! mbedtls_ssl_write returned %d\n\n", ret);
					failFlag = true;
					retFn = FAILED_TO_READ_RESPONSE_FROM_SERVER_PUB_DATA_SERVER;
					server_deinit();
					return retFn;
				}
			}

	do
	{
		memset(recv_buf, 0, 512);
		vTaskDelay(1000);
		ret = mbedtls_ssl_read(&ssl, (unsigned char *) recv_buf, 512);
		vTaskDelay(1000);
		//recv_buf[511] = '\0';
		if(enable_debug) printf("\nret: %d data: %s\n", ret, recv_buf);

		if(ret == MBEDTLS_ERR_SSL_WANT_READ || ret == MBEDTLS_ERR_SSL_WANT_WRITE)
		{
			printf("\ncontinue publish to server ... %d\n", ret);
			continue;
		}

		if(ret == MBEDTLS_ERR_SSL_PEER_CLOSE_NOTIFY)
		{
			printf("\nMBEDTLS_ERR_SSL_PEER_CLOSE_NOTIFY exiting publish to server ... %d\n", ret);
			server_deinit();
			retFn = FAILED_PORT_CLOSED_BY_PEER_PUB_DATA_SERVER;
			return retFn;
		}

		if(ret < 0)
		{
			if(enable_debug) printf("failed\n  ! mbedtls_ssl_read returned %d\n\n", ret);
			retFn = FAILED_TO_READ_RESPONSE_FROM_SERVER_PUB_DATA_SERVER;
			server_deinit();
			return retFn;
		}

				if((strstr(recv_buf, "200 OK") != NULL) || (strstr(recv_buf, "201 Created") != NULL))
				{
					printf("I am coming here and checking 200 ok\n");
					retFn = NO_ERROR_PUB_DATA_SERVER;
				}
				else if((strstr(recv_buf, "400 BAD REQUEST") != NULL))
				{
					printf("I am coming here and checking 400 BAD Request ok\n");
					retFn = FAILED_TO_READ_RESPONSE_BAD_REQUEST_DATA_SERVER;
				}
				else
				{
					printf("It should never come here\n");
				}
	}while(!exitFlag);
	server_deinit();
	printf("\nexiting publish to server ... %d\n", retFn);
	return retFn;				
}

`

### What is the actual behavior?

Actual behaviour is during normal WiFi available with good internet connection upload is successful but when there is weak internet the file upload cause the device to stuck for more than 30 minutes

### Steps to reproduce.

1. During the file Upload to server turn off the mobile internet
2. The mbedtls api get stucks for more than 30 minutes

nachiketkukade, 2022-12-14:
@hardik-dozee , can you please provide console logs for the issue?

laukik-hase, 2022-12-15:
Hello, @hardik-dozee!

Could you try setting the connection read timeout before the `mbedtls_ssl_handshake()` call?
Example snippet - 
```c
...
/* Setting the recv_timeout callback function */
mbedtls_ssl_set_bio(&ssl, &server_fd, mbedtls_net_send, mbedtls_net_recv, mbedtls_net_recv_timeout); 
/* Setting the timeout to 10000ms = 10s */
mbedtls_ssl_conf_read_timeout(&conf, 10000); 

while ((ret = mbedtls_ssl_handshake(&ssl)) != 0)
{
...
```

**Note:** While working with HTTPS in ESP-IDF, refer to the [`https_server/simple`](https://github.com/espressif/esp-idf/tree/v4.3.4/examples/protocols/https_server/simple) example and the [`https_request`](https://github.com/espressif/esp-idf/tree/v4.3.4/examples/protocols/https_request) examples. In ESP-IDF, it is recommended to use the `ESP-TLS` APIs for TLS functionality rather than the Mbed TLS APIs directly.

laukik-hase, 2023-01-16:
Hello, @hardik-dozee!

Closing this issue due to lack of feedback, feel free to reopen with more updates. Thanks.

closed 2023-01-16 by laukik-hase