espressif/esp-idf #10359: mDNS and W5500 Ethernet chip for ESP32-S3

mgoeller, 2022-12-13:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hi

We have so far used ESP32 (with LAN8720), but now upgraded to ESP32-S3 using W5500 for a wired Ethernet connection. W5500 is officially supported by ESP-IDF and the connection works perfectly for all tasks **except** mDNS multicast packets. Using Wireshark we can clearly see that the packets are transmitted using the WiFi interface, but not when using a wired connection.

See also the same problem described here: [Link](https://www.reddit.com/r/esp32/comments/vwegz5/w5500_and_mdns/).

Can you please help?

ESP-YJM, 2022-12-14:
@mgoeller Could you please offer your IDF version, commit you used.

kostaond, 2022-12-16:
@mgoeller could you please share what ESP-IDF version do you use? I tried master version of [mDNS example](https://github.com/espressif/esp-protocols/tree/master/components/mdns/examples)  @c7f20ee with master version of ESP-IDF @a3dd334081 and it seems it works as expected with W5500. It was able to resolve IP address of my network printer, see below:

```hash
I (408) main_task: Calling app_main()                                           
I (428) mdns-test: mdns hostname set to: [esp32-mdns]                           
I (458) w5500.mac: version=4                                                    
I (468) esp_eth.netif.netif_glue: 02:00:00:12:34:56                             
I (468) esp_eth.netif.netif_glue: ethernet attached to netif                    
I (478) example_connect: Waiting for IP(s)                                      
I (2478) example_connect: Ethernet Link Up                                      
I (3478) esp_netif_handlers: example_connect: eth ip: 10.0.0.32, mask: 255.255.8
I (3478) example_connect: Got IPv4 event: Interface "example_connect: eth" addr2
I (4428) example_connect: Got IPv6 event: Interface "example_connect: eth" addrL
I (4428) example_connect: Connected to example_connect: eth                     
I (4438) example_connect: - IPv4 address: 10.0.0.32                             
I (4438) example_connect: - IPv6 address: fe80:0000:0000:0000:0000:00ff:fe12:34L
I (4448) gpio: GPIO[0]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulld 
I (4458) main_task: Returned from app_main()                                    
I (9458) mdns-test: Query both A and AAA: esp32-mdns.local                      
I (10558) mdns-test: Query A esp32-mdns.local finished                          
I (10558) mdns-test: Query AAAA esp32-mdns.local finished                       
I (10608) mdns-test: Query A: esp32.local                                       
W (12628) mdns-test: ESP_ERR_NOT_FOUND: Host was not found!                     
I (12628) mdns-test: Query PTR: _arduino._tcp.local                             
W (15728) mdns-test: No results found!                                          
I (15728) mdns-test: Query PTR: _http._tcp.local                                
W (18828) mdns-test: No results found!                                          
I (18828) mdns-test: Query PTR: _printer._tcp.local                             
1: Interface: ETH_DEF, Type: V4, TTL: 120                                       
  PTR : HP LaserJet P2015 Series @ raspberrypi._printer._tcp                    
  SRV : raspberrypi.local:0                                                     
  A   : 10.0.0.35                                                               
  AAAA: fe80:0000:0000:0000:b67b:d3b2:8037:66a8                                 
2: Interface: ETH_DEF, Type: V4, TTL: 120                                       
  PTR : HP LaserJet P2015 Device (Automatically setup by HPLIP) @ raspb._printep
  SRV : raspberrypi.local:0                                                     
  A   : 10.0.0.35                                                               
  AAAA: fe80:0000:0000:0000:b67b:d3b2:8037:66a8                                 
3: Interface: ETH_DEF, Type: V6, TTL: 120                                       
  PTR : HP LaserJet P2015 Series @ raspberrypi._printer._tcp                    
  SRV : raspberrypi.local:0                                                     
  AAAA: fe80:0000:0000:0000:b67b:d3b2:8037:66a8      
```

reinismu, 2023-01-23:
@mgoeller Seems like I have the same issue. Using WIFI it works fine, but with W5500 ETH it doesn't.

Have you managed to solve this issue?

@kostaond Did you manage to register service in local network with it?

kostaond, 2023-01-23:
@reinismu could you please share what version of ESP-IDF do you use?

reinismu, 2023-01-23:
@kostaond v4.4

I use https://github.com/esp-rs/esp-idf-svc project. Tho looking on calls it does the same

kostaond, 2023-01-23:
@reinismu very nice finding!!

There is also alternative way of how to fix your issue introduced by RUST wrapper lib. However, it will work only with latest version of mDNS. You could register the ESP NETIF with `ETH_CL_DEF` key by using [`mdns_register_netif`](https://espressif.github.io/esp-protocols/mdns/en/index.html#_CPPv419mdns_register_netifP11esp_netif_t). [Here is an example.](https://github.com/espressif/esp-protocols/blob/master/components/mdns/examples/main/mdns_example_main.c#L274)

kostaond, 2023-01-24:
@reinismu do you consider your issue resolved?

@mgoeller could you please share the IDF version you use? Could you please check if the issue is has the same root cause as @reinismu?

reinismu, 2023-01-24:
@kostaond ye I solved it, but someone should patch so mdns would fail if doesn't find any IP address to broadcast to

kostaond, 2023-01-25:
@reinismu mNDS is now maintained at [esp-protocols repo](https://github.com/espressif/esp-protocols), feel free to raise the issue there.

david-cermak, 2023-01-25:
@reinismu Would this fix the crash in mDNS?

```patch
diff --git a/components/mdns/mdns.c b/components/mdns/mdns.c
index 6a5c7b1097..876ead9d7c 100644
--- a/components/mdns/mdns.c
+++ b/components/mdns/mdns.c
@@ -1819,6 +1819,10 @@ static void _mdns_pcb_send_bye(mdns_if_t tcpip_if, mdns_ip_protocol_t ip_protoco
 static void _mdns_init_pcb_probe_new_service(mdns_if_t tcpip_if, mdns_ip_protocol_t ip_protocol, mdns_srv_item_t ** services, size_t len, bool probe_ip)
 {
     mdns_pcb_t * pcb = &_mdns_server->interfaces[tcpip_if].pcbs[ip_protocol];
+    if (pcb == NULL) {
+        return;
+    }
+
     size_t services_final_len = len;
 
     if (PCB_STATE_IS_PROBING(pcb)) {
```

If I understand correctly,  mDNS initializes the pcb's on startup when the interface is not ready yet and, when it appears it tries to send probes and dereferences its pcb.

kostaond, 2023-01-31:
@mgoeller could you please provide more information as already requested so we could try to reproduce the issue? Otherwise, I will have to close it...

mgoeller, 2023-02-14:
Using ESP-IDF 4.4.3, with a slightly adapted w5500 component, and as @reinismu mentioned the problem was the key. When changing it from ETH_SPI to ETH_DEF it worked again. That needs to be corrected and not hardcoded, right?

kostaond, 2023-04-03:
@mgoeller, you resolved your issue. It seems there is no issue in the ESP-IDF based on the above discussion. Therefore I am closing the GH issue. Please feel free to re-open if additional actions were needed.

closed 2023-04-03 by kostaond