espressif/esp-idf #10360: ESP IDF - Build results in undefined reference to a function

PadmaKannan10, 2022-12-13:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

esp-idf-tools-setup-offline-4.1.2.exe

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

CMD

### What is the expected behavior?

I wanted to add a new functionality to an existing code.
Building the existing code is successful, however after adding new functionality the build results in linking issue and incomplete. 
Expected the code to build without any linking issue.

### What is the actual behavior?

Adding new functionality to the existing project throws linking issue mentioning 'undefined reference to a function'.

**c:\project_workspace\hw_wdg\hoppecke\build/../SysTimeManager/SysTimer.c:205: undefined reference to `Wdgtimer_Reset'
collect2.exe: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
ninja failed with exit code 1**

### Steps to reproduce.

Task: To add a new functionality to an existing code

1. Executed build for existing code - Build successful
2. Added watchdog functionality to the existing code and did a build again
3. Project directory - structure
	- Utils
	  |_ Wdgtimer.c
	  |_ Wdgtimer.h
	  |_ CMakeLists.txt
	- SysTimeManager
	  |_ SysTimer.c
	  |_ SysTimer.h
	  |_ CMakeLists.txt
4. Defined watchdog timer function (function name: Wdgtimer_Reset) in Utils\Wdgtimer.c and declared the same in Utils\Wdgtimer.h
   Called Wdgtimer_Reset from SysTimeManager\SysTimer.c file
   Added Utils in INCLUDE_DIR in SysTimeManager\CMakeLists.txt file
   Included Wdgtimer.h in both Wdgtimer.c and SysTimer.c files
5. Also, tried adding LINK_INTERFACE_MULTIPLICITY as mentioned in ESP - Build guide, in Utils\CMakeLists.txt file. No changes were seen.
Reference: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/build-system.html

### Build or installation Logs.

```plain
Build issue - logs:

Partition table built but not signed. Sign partition data before flashing:
        C:/Espressif/python_env/idf4.1_py3.8_env/Scripts/python.exe C:/Espressif/frameworks/esp-idf-v4.1.2/components/esptool_py/esptool/espsecure.py sign_data --keyfile KEYFILE C:/Project_Workspace/HW_wdg/hoppecke/build/partition_table/partition-table.bin
[3/7] Performing build step for 'bootloader'
ninja: no work to do.
[6/7] Linking CXX executable CBv2FwApp_v1_00_06_2mRst_io0Toggle.elf
FAILED: CBv2FwApp_v1_00_06_2mRst_io0Toggle.elf
cmd.exe /C "cd . && C:\Espressif\tools\xtensa-esp32-elf\esp-2020r3-8.4.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-g++.exe  -mlongcalls -Wno-frame-address   @CMakeFiles\CBv2FwApp_v1_00_06_2mRst_io0Toggle.elf.rsp  -o CBv2FwApp_v1_00_06_2mRst_io0Toggle.elf  && cd ."
c:/espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: esp-idf/SysTimeManager/libSysTimeManager.a(SysTimer.c.obj):(.literal.periodic2msTimer+0x2c): undefined reference to `Wdgtimer_Reset'
c:/espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: esp-idf/SysTimeManager/libSysTimeManager.a(SysTimer.c.obj): in function `periodic2msTimer':
c:\project_workspace\hw_wdg\hoppecke\build/../SysTimeManager/SysTimer.c:205: undefined reference to `Wdgtimer_Reset'
collect2.exe: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
```

igrr, 2022-12-13:
@PadmaKannan10 Could you please show the contents of CMakeLists.txt files in `Utils` and `SysTimerManager` directories, as well as the top-level project CMakeLists.txt file?

PadmaKannan10, 2022-12-14:
@igrr thankyou for the response. Please find the attached files for your reference:

Utils\CMakeLists: [CMakeLists.txt](https://github.com/espressif/esp-idf/files/10225122/CMakeLists.txt)
SysTimerManager\CMakeLists: [CMakeLists.txt](https://github.com/espressif/esp-idf/files/10225123/CMakeLists.txt)
CMakeLists: [CMakeLists.txt](https://github.com/espressif/esp-idf/files/10225252/CMakeLists.txt)

igrr, 2022-12-14:
@PadmaKannan10 Unrelated to the issue report, I have noticed that you are using ESP-IDF v4.1.x now. I recommend upgrading to a more recent release, since v4.1.x will be EoL soon. More information [here](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/versions.html#support-periods).

closed 2022-12-14 by igrr

PadmaKannan10, 2022-12-15:
@igrr sure, thankyou for the information.