espressif/esp-idf #10361: espcoredump.py info_corefile not working on file

will-emmerson, 2022-12-13:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4.2-55-g9fe854c4ca

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

N/A

### Power Supply used.

USB

### What is the expected behavior?

I ran the following command to view a coredump elf file but it seems to be ignoring the file and trying to read it over flash:

```bash
espcoredump.py info_corefile --core coredump/coredump_602C_20221213_1151.elf build/8-channel.elf
espcoredump.py v0.4-dev
===============================================================
==================== ESP32 CORE DUMP START ====================
Connecting..................^C
```

### What is the actual behavior?

It tries to read from flash

### Steps to reproduce.

Above

alekseiapa, 2022-12-14:
Hello @will-emmerson ! We appreciate you reporting this bug. I'll investigate the issue.

dobairoland, 2023-02-17:
For future reference, d3eb5b234686be0e8c0432e929a5cdd594550dc7 fixes this issue.

Alvin1Zhang, 2023-06-20:
Thanks for reporting, fix is available, feel free to reopen. Thanks.

closed 2023-06-20 by Alvin1Zhang