espressif/esp-idf #10364: WPA3 connect endless loop without success

0xFEEDC0DE64, 2022-12-13:
Hi,

I just upgraded the esp-idf in our firmware project and I experience a lot WPA3 wifi connect problems now.
The wifi blob seems to go in some sort of endless loop? While rebasing the esp-idf repo I read some commits regarding connect retry count, should we enable some limit now?

```
I (7407) wpa bss.c:207 wpa_bss_add(): BSS: Add new id 28 BSSID 18:d6:c7:fc:f1:fb SSID 'Channel13' chan 13
E (9637) WIFI_STACK espwifistack.cpp:530 update(): scan failed with Failed
E (9747) WIFI_STACK espwifistack.cpp:530 update(): scan failed with Failed
E (9867) WIFI_STACK espwifistack.cpp:530 update(): scan failed with Failed
E (9967) WIFI_STACK espwifistack.cpp:530 update(): scan failed with Failed
E (10077) WIFI_STACK espwifistack.cpp:530 update(): scan failed with Failed
E (10187) WIFI_STACK espwifistack.cpp:530 update(): scan failed with Failed
I (10277) WIFI_STACK espwifistack.cpp:1584 wifi_event_cb(): event_base=WIFI_EVENT event_id=WIFI_EVENT_SCAN_DONE
I (10277) WIFI_STACK espwifistack.cpp:1587 wifi_event_cb(): SCAN Done: ID: 128, Status: 0, Results: 29
I (10287) WIFI_STACK espwifistack.cpp:513 update(): Finished scan with 29 results
I (10297) WIFI_STACK espwifistack.cpp:517 update(): Not connected after scan, building connect plan...
I (10307) WIFI_STACK espwifistack.cpp:2634 nextConnectPlanItem(): connecting to goe-iot (auth=OPEN, key=************, channel=1, rssi=-56, bssid=24:5A:4C:63:EE:14)
I (10317) WIFI_STACK espwifistack.cpp:2644 nextConnectPlanItem(): resetting wifi connect fail counter
I (10327) WIFI_STACK espwifistack.cpp:1049 wifi_set_esp_interface_ip(): STA set DYNAMIC
I (10337) WIFI_STACK espwifistack.cpp:1071 wifi_set_esp_interface_ip(): esp_netif_dhcpc_get_status() resulted in INIT
I (10347) WIFI_STACK espwifistack.cpp:1095 wifi_set_esp_interface_ip(): starting dhcpc
I (10357) WIFI_STACK espwifistack.cpp:2400 wifi_sta_begin(): clearing connect fail flag
I (10367) WIFI_STACK espwifistack.cpp:1258 set_sta_status(): CONNECTING (from IDLE_STATUS)
I (10377) WIFI_STACK espwifistack.cpp:2522 setWifiState(): state machine state changed from Scanning to Connecting
I (10387) WIFI_STACK espwifistack.cpp:629 update(): connecting: CONNECTING
I (10377) wpa bss.c:207 wpa_bss_add(): BSS: Add new id 29 BSSID 24:5a:4c:63:ee:14 SSID 'goe-iot' chan 1
W (10407) ESPCPPUTILS schedulertask.cpp:60 loop(): task wifi hang for 115ms (heap8=119096)
I (10807) wifi:new:<1,1>, old:<1,1>, ap:<1,1>, sta:<1,0>, prof:1
I (10807) wifi:state: init -> auth (b0)
W (11827) ESPCPPUTILS schedulertask.cpp:60 loop(): task cloud hang for 253ms (heap8=114020)
I (11837) wifi:state: auth -> assoc (0)
E (11857) wifi:Association refused temporarily, comeback time 1048 mSec
I (12907) wifi:state: assoc -> assoc (0)
I (13907) wifi:state: assoc -> init (200)
I (13907) wifi:new:<1,0>, old:<1,1>, ap:<1,1>, sta:<1,0>, prof:1
I (13907) wifi:new:<1,1>, old:<1,0>, ap:<1,1>, sta:<1,0>, prof:1
I (13907) wifi:state: init -> auth (b0)
I (13907) wpa esp_wpa3.c:35 wpa3_build_sae_commit(): wpa3: Skip SAE and use cached PMK instead
I (13927) wifi:state: auth -> assoc (0)
E (13987) wifi:Set status to INIT
I (13987) wifi:state: assoc -> init (3500)
I (13987) wifi:new:<1,0>, old:<1,1>, ap:<1,1>, sta:<1,0>, prof:1
I (13997) wifi:new:<1,1>, old:<1,0>, ap:<1,1>, sta:<1,0>, prof:1
I (13997) wifi:state: init -> auth (b0)
I (13997) wpa esp_wpa3.c:35 wpa3_build_sae_commit(): wpa3: Skip SAE and use cached PMK instead
I (14017) wifi:state: auth -> assoc (0)
I (15017) wifi:state: assoc -> init (400)
I (15017) wifi:new:<1,0>, old:<1,1>, ap:<1,1>, sta:<1,0>, prof:1
I (15017) wifi:new:<1,1>, old:<1,0>, ap:<1,1>, sta:<1,0>, prof:1
I (15027) wifi:state: init -> auth (b0)
I (15027) wpa esp_wpa3.c:35 wpa3_build_sae_commit(): wpa3: Skip SAE and use cached PMK instead
I (15037) wifi:state: auth -> assoc (0)
E (15097) wifi:Set status to INIT
I (15097) wifi:state: assoc -> init (3500)
I (15097) wifi:new:<1,0>, old:<1,1>, ap:<1,1>, sta:<1,0>, prof:1
I (15097) wifi:new:<1,1>, old:<1,0>, ap:<1,1>, sta:<1,0>, prof:1
I (15107) wifi:state: init -> auth (b0)
I (15107) wpa esp_wpa3.c:35 wpa3_build_sae_commit(): wpa3: Skip SAE and use cached PMK instead
I (15117) wifi:state: auth -> init (2c0)
I (15117) wifi:new:<1,0>, old:<1,1>, ap:<1,1>, sta:<1,0>, prof:1
I (15127) wifi:new:<1,1>, old:<1,0>, ap:<1,1>, sta:<1,0>, prof:1
I (15127) wifi:state: init -> auth (b0)
I (15137) wpa esp_wpa3.c:35 wpa3_build_sae_commit(): wpa3: Skip SAE and use cached PMK instead
I (15147) wifi:state: auth -> assoc (0)
I (16147) wifi:state: assoc -> init (400)
I (16147) wifi:new:<1,0>, old:<1,1>, ap:<1,1>, sta:<1,0>, prof:1
I (16147) wifi:new:<1,1>, old:<1,0>, ap:<1,1>, sta:<1,0>, prof:1
I (16157) wifi:state: init -> auth (b0)
I (16157) wpa esp_wpa3.c:35 wpa3_build_sae_commit(): wpa3: Skip SAE and use cached PMK instead
[... 134 lines cut ...]
E (26337) wifi:Set status to INIT
I (26337) wifi:state: assoc -> init (3500)
I (26337) wifi:new:<1,0>, old:<1,1>, ap:<1,1>, sta:<1,0>, prof:1
I (26337) wifi:new:<1,1>, old:<1,0>, ap:<1,1>, sta:<1,0>, prof:1
I (26347) wifi:state: init -> auth (b0)
I (26347) wpa esp_wpa3.c:35 wpa3_build_sae_commit(): wpa3: Skip SAE and use cached PMK instead
I (26357) wifi:state: auth -> init (2c0)
I (26367) wifi:new:<1,0>, old:<1,1>, ap:<1,1>, sta:<1,0>, prof:1
I (26367) wifi:new:<1,1>, old:<1,0>, ap:<1,1>, sta:<1,0>, prof:1
I (26377) wifi:state: init -> auth (b0)
I (26377) wpa esp_wpa3.c:35 wpa3_build_sae_commit(): wpa3: Skip SAE and use cached PMK instead
I (26387) wifi:state: auth -> assoc (0)
I (27387) wifi:state: assoc -> init (400)
I (27387) wifi:new:<1,0>, old:<1,1>, ap:<1,1>, sta:<1,0>, prof:1
I (27397) wifi:new:<1,1>, old:<1,0>, ap:<1,1>, sta:<1,0>, prof:1
I (27397) wifi:state: init -> auth (b0)
I (27397) wpa esp_wpa3.c:35 wpa3_build_sae_commit(): wpa3: Skip SAE and use cached PMK instead
I (27407) wifi:state: auth -> assoc (0)
E (27477) wifi:Set status to INIT
I (27477) wifi:state: assoc -> init (3500)
I (27477) wifi:new:<1,0>, old:<1,1>, ap:<1,1>, sta:<1,0>, prof:1
I (27487) wifi:new:<1,1>, old:<1,0>, ap:<1,1>, sta:<1,0>, prof:1
I (27487) wifi:state: init -> auth (b0)
I (27487) wpa esp_wpa3.c:35 wpa3_build_sae_commit(): wpa3: Skip SAE and use cached PMK instead
```

AxelLin, 2022-12-14:
Which esp-idf version (git describe --tags)?

0xFEEDC0DE64, 2022-12-20:
c92f2394f2a15d1d82f10cc10d2c70b4433d81a3

closed 2022-12-22 by espressif-bot

AxelLin, 2022-12-22:
@kapilkedawat Does v4.3 branch need this fix?

0xFEEDC0DE64, 2022-12-22:
i would like to see a v5.1-dev-2300 fix first maybe?

kapilkedawat, 2022-12-27:
Yes, it will be available in v4.3 as well.
 @0xFEEDC0DE64 https://github.com/espressif/esp-idf/commit/2e8ebdc0fc29e9d91a7567c04a9d963175c9d7a8 is fix on master, please try with that once.