espressif/esp-idf #10365: Read NVS with nvs_partition_gen?

someburner, 2022-12-13:
### Describe the solution you'd like.

I'd like to be able to read back values written by `nvs_partition_gen.py` in python as part of a manufacturing process. This is useful for handling customer returns, and would also be nice as a failsafe check to make sure NVS partition data is correct.

### Describe alternatives you've considered.

Appears there is a host test platform but it's in CPP and probably overkill for something that shouldnt be too hard to just reverse in python.

igrr, 2022-12-13:
There is also https://github.com/espressif/esp-idf/tree/master/components/nvs_flash/nvs_partition_parser in the `master` branch. (It should work with NVS binaries generated with older versions of IDF, as well.)

someburner, 2022-12-13:
Ah okay I missed that since I've been working off the release branch. Thanks! Would be good to document somewhere if it's not planning to be merged soon. Feel free to close.

igrr, 2022-12-14:
This utility is documented here: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/storage/nvs_partition_parse.html. Currently it is available in `master` branch only, will be part of the v5.1 release next year. Given that it's a new feature, I think we won't be backporting it to the existing release branches.