espressif/esp-idf #10370: [DFU Mode] Does not support Flash Encryption?

chipweinberger, 2022-12-14:
### Is your feature request related to a problem?

I don't see any mention of Flash Encryption for DFU Mode in the docs:

https://docs.espressif.com/projects/esp-idf/en/latest/esp32s2/api-guides/dfu.html

I am using Host Generated Keys. DFU should be possible I would think. But `idf.py dfu` does not support passing in a encryption key, or passing in pre-encrypted binaries. What are we supposed to do?

Thanks.

### Describe the solution you'd like.

`idf.py dfu --host-generated-key xxxxxxxxxx`

igrr, 2022-12-14:
Generally we recommend enabling Secure Download mode (`CONFIG_SECURE_ENABLE_SECURE_ROM_DL_MODE`) when enabling security features. USB related features are completely disabled in the secure download mode to reduce the potential attack surface of the bootloader. Given that, I'm not sure how likely it is that we will implement this feature request.

If you do want to use DFU together with Flash Encryption, you can add a custom CMake target for that in your project (`add_custom_command`):
* Call espsecure.py to encrypt the bootloader, partition table and app binaries
* Call $IDF_PATH/tools/mkdfu.py to create the DFU archive
* Call `dfu-util -d VID:PID -D dfu.bin` to send the DFU archive to the device.

chipweinberger, 2022-12-14:
Thanks for the answer.

Yes I suspected that might be the case, but I wanted to double check, since it was not mentioned in the docs.

> USB related features are completely disabled in the secure download mode

Does the include USB _Serial_ (as in USB_SERIAL_JTAG)? oh no, I was expecting to use that!

igrr, 2022-12-15:
USB_SERIAL_JTAG should work, sorry for saying that incorrectly. Only the software USB_CDC implementation based on USB_OTG peripheral is non-functional in the secure download mode. (Just because it's a software implementation => disabled to reduce the attack surface. USB_SERIAL_JTAG is fully implemented in hardware, so it is kept enabled.)

chipweinberger, 2022-12-16:
Thanks for the answers. 

The DFU docs really ought to mention this, IMO =)

aside: I really wish it was easier to edit docs in the github interface. Having to create a PR locally is much more work.

closed 2022-12-16 by chipweinberger