espressif/esp-idf #10371: CMake Error at run_serial_tool.cmake:50 ERROR MESSAGE WHEN FLASHING

olliec28, 2022-12-14:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

V3.3.2-dev

### Operating System used.

macOS

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

I am using the hello_world project from https://docs.espressif.com/projects/esp-idf/en/release-v4.3/esp32/get-started/index.html#get-started-get-esp-idf and I manage to get up to step 9 which is flashing it onto the esp32 device. The programme should be flashed onto the esp32 device.

### What is the actual behavior?

An error appears CMake Error at run_serial_tool.cmake:50 (message):
  /Users/olivercustance/.espressif/python_env/idf4.3_py3.8_env/bin/python
  /Users/olivercustance/esp/esp-idf/components/esptool_py/esptool/esptool.py
  --chip esp32 failed

### Steps to reproduce.

1. By following the step by step guide on - https://docs.espressif.com/projects/esp-idf/en/release-v4.3/esp32/get-started/index.html#get-started-get-esp-idf

### Build or installation Logs.

```plain
Olivers-MacBook-Pro:hello_world olivercustance$ idf.py -p /dev/cu.usbserial-0001 -b 460800 flash
Executing action: flash
Running ninja in directory /Users/olivercustance/esp/hello_world/build
Executing "ninja flash"...
[1/4] Performing build step for 'bootloader'
ninja: no work to do.
[3/4] cd /Users/olivercustance/esp/esp-idf/compon...p-idf/components/esptool_py/run_serial_tool.cmake
esptool.py esp32 -p /dev/cu.usbserial-0001 -b 460800 --before=default_reset --after=hard_reset write_flash --flash_mode dio --flash_freq 40m --flash_size 2MB 0x8000 partition_table/partition-table.bin 0x1000 bootloader/bootloader.bin 0x10000 hello-world.bin
esptool.py v3.3.2-dev
Serial port /dev/cu.usbserial-0001
Traceback (most recent call last):
  File "/Users/olivercustance/esp/esp-idf/components/esptool_py/esptool/esptool.py", line 5399, in <module>
    _main()
  File "/Users/olivercustance/esp/esp-idf/components/esptool_py/esptool/esptool.py", line 5392, in _main
    main()
  File "/Users/olivercustance/esp/esp-idf/components/esptool_py/esptool/esptool.py", line 4699, in main
    esp = esp or get_default_connected_device(ser_list, port=args.port, connect_attempts=args.connect_attempts,
  File "/Users/olivercustance/esp/esp-idf/components/esptool_py/esptool/esptool.py", line 114, in get_default_connected_device
    _esp = chip_class(each_port, initial_baud, trace)
  File "/Users/olivercustance/esp/esp-idf/components/esptool_py/esptool/esptool.py", line 320, in __init__
    self._port = serial.serial_for_url(port)
  File "/Users/olivercustance/.espressif/python_env/idf4.3_py3.8_env/lib/python3.8/site-packages/serial/__init__.py", line 90, in serial_for_url
    instance.open()
  File "/Users/olivercustance/.espressif/python_env/idf4.3_py3.8_env/lib/python3.8/site-packages/serial/serialposix.py", line 332, in open
    self._reconfigure_port(force_update=True)
  File "/Users/olivercustance/.espressif/python_env/idf4.3_py3.8_env/lib/python3.8/site-packages/serial/serialposix.py", line 517, in _reconfigure_port
    termios.tcsetattr(
termios.error: (22, 'Invalid argument')
CMake Error at run_serial_tool.cmake:50 (message):
  /Users/olivercustance/.espressif/python_env/idf4.3_py3.8_env/bin/python
  /Users/olivercustance/esp/esp-idf/components/esptool_py/esptool/esptool.py
  --chip esp32 failed

FAILED: CMakeFiles/flash /Users/olivercustance/esp/hello_world/build/CMakeFiles/flash 
cd /Users/olivercustance/esp/esp-idf/components/esptool_py && /usr/local/Cellar/cmake/3.25.1/bin/cmake -D IDF_PATH="/Users/olivercustance/esp/esp-idf" -D SERIAL_TOOL="/Users/olivercustance/.espressif/python_env/idf4.3_py3.8_env/bin/python /Users/olivercustance/esp/esp-idf/components/esptool_py/esptool/esptool.py --chip esp32" -D SERIAL_TOOL_ARGS="--before=default_reset --after=hard_reset write_flash @flash_args" -D WORKING_DIRECTORY="/Users/olivercustance/esp/hello_world/build" -P /Users/olivercustance/esp/esp-idf/components/esptool_py/run_serial_tool.cmake
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
```

### More Information.

The strange thing is that this worked the other day and I have come to do it again and this is the error I have. I have looked around and I haven't found a solution as of yet thanks.

Also, I am a beginner so solutions must be made clear to me so I can understand them. Thanks

radimkarnis, 2022-12-15:
Hello @olliec28,
this issue seems to be related to the hardware - your Mac is having trouble opening the serial port.

Although I don't know the exact reason, there are several things to check:
1) Are you selecting the correct port? (`-p /dev/cu.usbserial-0001`)
2) Are you using a USB hub? If yes, can it provide enough power? Please see [this thread](https://stackoverflow.com/a/42866736) where the same problem is discussed. Also, try connecting the devkit directly to the Mac.
3) Maybe the driver doesn't support this baud rate. Try lowering it from `-b 460800` to `-b 115200`

Please try these and let me know how it went!

KaeLL, 2022-12-15:
> V3.3.2-dev

> I am using the hello_world project from [https://docs.espressif.com/projects/esp-idf/en/release-v4.3](https://docs.espressif.com/projects/esp-idf/en/release-v4.3/esp32/get-started/index.html#get-started-get-esp-idf)

radimkarnis, 2022-12-15:
@KaeLL `V3.3.2-dev` is the esptool version. IDF is `v4.3`

olliec28, 2022-12-15:
> Hello @olliec28, this issue seems to be related to the hardware - your Mac is having trouble opening the serial port.
>
> Although I don't know the exact reason, there are several things to check:
>
> 1. [...]

So when I run the command ls /dev/cu.* there are 3 ports which appear. One port is called /dev/cu.SLAB_USBtoUART. The other is called /dev/cu.URT0 and the final port is called /dev/cu.usbserial-0001. I have tried all 3 port names with different baud rates as stated in your response and still no luck.

So I am not sure what a USB hub is but I am using a micro usb cable which goes from the devkit to a usb and then that goes into a usb to usb adapter into my Mac. So maybe I could buy a micros to usb cable and see if this has any effect. I just don't understand what's going as I stated, it worked a couple days ago and now it doesn't seem to work. 

No idea if this is maybe due to the fact that the hello_world project has already been flashed onto usb and this is the reason. But I have since erased all flash memory off the devkit so this shouldn't be the issue.

Thanks for your help.

olliec28, 2022-12-17:
> @olliec28, you can also try connecting to the chip with any [serial terminal program](https://docs.espressif.com/projects/esptool/en/latest/esp32/troubleshooting.html#serial-terminal-programs). Running miniterm with `python -m serial.tools.miniterm <PORT> 115200` is a good option. If this also doesn't work, there is something seriously wrong with the drivers/the USB-to-UART chip on your devkit/the physical connection.
> 
> The [esptool troubleshooting guide](https://docs.espressif.com/projects/esptool/en/latest/esp32/troubleshooting.html) mentions many fixes to common connection problems. Please try going through the guide.

Hi I have tried your above and this also don't work with the log put below. 

Olivers-MacBook-Pro:hello_world olivercustance$ python -m serial.tools.miniterm /dev/cu.usbserial-1420 115200
Traceback (most recent call last):
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/runpy.py", line 194, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "/Users/olivercustance/.espressif/python_env/idf4.3_py3.8_env/lib/python3.8/site-packages/serial/tools/miniterm.py", line 1042, in <module>
    main()
  File "/Users/olivercustance/.espressif/python_env/idf4.3_py3.8_env/lib/python3.8/site-packages/serial/tools/miniterm.py", line 998, in main
    serial_instance.open()
  File "/Users/olivercustance/.espressif/python_env/idf4.3_py3.8_env/lib/python3.8/site-packages/serial/serialposix.py", line 332, in open
    self._reconfigure_port(force_update=True)
  File "/Users/olivercustance/.espressif/python_env/idf4.3_py3.8_env/lib/python3.8/site-packages/serial/serialposix.py", line 517, in _reconfigure_port
    termios.tcsetattr(
termios.error: (22, 'Invalid argument')

Im thinking the issue may be that its not receiving enough power through my adapter, so I am going to try getting a cable which goes from the devout to my Mac in one cable rather than use an adapter. Any ideas why it worked couple days ago and not now though, that's the bit which im very confused about.

Another thing which may help you identify the problem. When I connect my devout to my Mac it flashes red quite rapidly. Not sure if you know if this is a sign of anything?

Thanks.

radimkarnis, 2022-12-19:
@olliec28 could you please tell me more about your hardware? Are you using a DevKit, such as [this one](https://www.espressif.com/en/products/devkits/esp32-devkitc), or do you have a [separate USB-to-Serial adapter](https://images-na.ssl-images-amazon.com/images/I/41n1ZNw4RaL._AC_.jpg)? This looks definitely like an issue with the USB-to-Serial chip used.

> Another thing which may help you identify the problem. When I connect my devout to my Mac it flashes red quite rapidly. Not sure if you know if this is a sign of anything?

If by "devout" you mean a "DevKit", the red flashing might signify the ESP is sending some data. There is either an application sending data running on it, or it's the ROM bootloader sending some info. This flashing should stop if you manually switch the board into download mode.

olliec28, 2022-12-19:
> @olliec28 could you please tell me more about your hardware? Are you using a DevKit, such as [this one](https://www.espressif.com/en/products/devkits/esp32-devkitc), or do you have a [separate [...]

Thanks for all your help. So I bought a new short cable which goes straight into my Mac now and it works, so I can confirm the issue was with the adapter I was using. The hello_world project works now.

Many thanks

radimkarnis, 2022-12-19:
@olliec28 great! Thanks for your cooperation, I am glad you could resolve the issue.

closed 2023-01-05 by dobairoland

sem-geologist, 2023-02-01:
Hello,
I am leaving my 2 cents, Just in case anyone fresh (to ESP32) like me would come over this problem again. I was getting exactly the same error on linux with hello_world trying it on ESP32-C3-M devKitM (manufactured by vcc-gnd studio) with dual usb... Actually it was so confusing, as one of USB goes through soldered on-board CH340 into uart tx 0 and tx1, while other USB goes directly to gpio 18 and 19. On my side I got confused as (hitherto) had no idea about differences  between ttyUSB* and ttyACM* . So initially I plugged my usb cable to "wrong" socket of dual usb devkit, and as I saw that `ttyACM0` had appeared under `/dev/` folder and it worked with `screen` (could connect and interact with preinstalled micropython) - I believed that my connection is OK. But that had errored at trying to flash exactly like in this reported issue simple hello_world. When using other usb socket, which on dev board goes to serial interface through USB-serial bridge, it appears to linux as `/dev/ttyUSB0` - using that the flashing was succesful.