espressif/esp-idf #10372: idf.py reconfigure doesn't create sdkconfig.defaults file

mbastida123, 2022-12-14:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

As per the IDF documentation: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/kconfig.html

> The format is the same as that of the sdkconfig file. sdkconfig.defaults can be created manually when one remembers all the changed configurations. Otherwise, the file can be generated automatically by running the ` idf.py save-defconfig ` command.

As per this issue https://github.com/espressif/esp-idf/issues/3965 I know that `idf.py save-defconfig` has been replaced by `idf.py reconfigure`

However, when I run `idf.py reconfigure` I can't find the sdkconfig.defaults file that should have been created as the documentation states.

dobairoland, 2022-12-14:
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/tools/idf-py.html#reconfigure-the-project-reconfigure

mbastida123, 2022-12-15:
Hi @dobairoland and thanks for the fast response.

It's clear that I was mistaken. However, going to mi primary question, which is the idf.py command to generate a sdkconfig.defaults file automatically?

dobairoland, 2022-12-15:
`idf.py save-defconfig` does it for you but not automatically. You need to run that command manually. It is available since ESP-IDF v5.0.

mbastida123, 2022-12-15:
> `idf.py save-defconfig` does it for you but not automatically. You need to run that command manually. It is available since ESP-IDF v5.0.

And what is the command if I'm in v4.4.3?

dobairoland, 2022-12-15:
There is no such command in earlier versions. One has to write manually `sdkconfig.defaults` in v4.4 or earlier versions.

dobairoland, 2022-12-15:
And you are reading the "latest" documentation which is for the upcoming v5.0.

This is the documentation for your version: https://docs.espressif.com/projects/esp-idf/en/release-v4.4/esp32/get-started/index.html

igrr, 2022-12-15:
One workaround I used in older versions (v4.3 and v4.4) is to open `idf.py menuconfig`, press `d`, then enter `../sdkconfig.defaults` and press Enter.
I am not 100% sure this has the same effect as `idf.py save-defconfig` but in the few cases where I used this approach it had worked correctly.

KaeLL, 2022-12-15:
Been using that method since v4.1, can confirm it works and it could only be better if it made me coffee.

mbastida123, 2022-12-15:
Hi @igrr @KaeLL 

When you say press d what do you exactly mean? 
I have tried to press d when in the menuconfig window but nothing happens

KaeLL, 2022-12-15:
`idf.py menuconfig` -> Among the keyboard shortcuts explained in the bottom section, there is "[D] Save minimal config (advanced)". Press it, and voilà.

mbastida123, 2022-12-15:
> `idf.py menuconfig` -> Among the keyboard shortcuts explained in the bottom section, there is "[D] Save minimal config (advanced)". Press it, and voilà.

explained in the bottom section? I think I'm missing on an interesting part XD
To be clear. I try pressing d (just the letter d) when I'm in this page:
[image]

igrr, 2022-12-15:
I'm sorry, I meant `idf.py menuconfig`, not the GUI Config editor from VS Code. It looks like this:
[image]