espressif/esp-idf #10373: Cannot esp_partition_mmap() on ESP32-S3 code from a partition, but can do it as data

vincentdupaquis, 2022-12-14:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4.3

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-S3-DevKitC-1 v1.0

### Power Supply used.

USB

### What is the expected behavior?

I have a partition containing code that I would like to map into the memory space. I use esp_partition_mmap() for doing so on an existing project using ESP32 (ESP-WROVER-KIT), and I wanted to port my code on a ESP32-S3 chip.
The expected behaviour is simply that I expect my code partition to be loaded.
Effectivelly, when I ask the same partition to be loaded as data (SPI_FLASH_MMAP_DATA instead of SPI_FLASH_MMAP_INST), the code works (but my code cannot be executed).``

### What is the actual behavior?

When exiting from esp_partition_mmap(), I get a ESP_ERR_NO_MEM error, and the code is not loaded.

### Steps to reproduce.

1. Create a partition on the chip (custom partition.csv, and then I create a partition  with IDs 0x40, 1 of any size (mine is 128K)
2. Call esp_partition_find_first() to get the partition handler
3. Call esp_partition_mmap(), having the 4th parameter to be SPI_FLASH_MMAP_INST
4. The return code of the last function is ESP_ERR_NO_MEM in my case

### Debug Logs.

```plain
I (149) esp_image: segment 4: paddr=00037b7c vaddr=403788ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x15 (USB_UART_CHIP_RESET),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40374283
0x40374283: _DebugExceptionVector at /home/vdupaquis/Dev/esp-idf/components/freertos/port/xtensa/xtensa_vectors.S:438

SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x1660
load:0x403c9700,len:0xbb8
load:0x403cc700,len:0x2e8c
entry 0x403c9954
I (24) boot: ESP-IDF v4.4.3 2nd stage bootloader
I (24) boot: compile time 16:33:46
I (25) boot: chip revision: 0
I (26) boot.esp32s3: Boot SPI Speed : 80MHz
I (31) boot.esp32s3: SPI Mode       : DIO
I (36) boot.esp32s3: SPI Flash Size : 2MB
I (40) boot: Enabling RNG early entropy source...
I (46) boot: Partition Table:
[... 158 lines cut ...]
This is esp32s3 chip with 2 CPU core(s), WiFi/BLE, silicon revision 0, 2MB external flash
Minimum free heap size: 345168 bytes
before esp_partition_find_first()
D (899) partition: Loading the partition table
V (909) calculated md5: 0x3fcf4158   35 a9 3f 61 03 dd cd bd  99 8b d8 e7 9c 48 71 00  |5.?a.........Hq.|
V (909) stored md5: 0x3c0380b0   35 a9 3f 61 03 dd cd bd  99 8b d8 e7 9c 48 71 00  |5.?a.........Hq.|
D (919) partition: Partition table MD5 verified
esp_partition_find_first() returned 0x3fcea000
before esp_partition_read()
esp_partition_find_first() returned ESP_OK
Data are 23 a0 01 00 13 f5 08 00 05 01 00 00 90 00 00 00
before esp_partition_read()
spi_flash_mmap() returned ESP_ERR_NO_MEM
before esp_partition_read()
esp_partition_mmap() returned ESP_ERR_NO_MEM
V (975) esp_apptrace: esp_apptrace_flush_nolock(): enter
V (991) esp_apptrace: esp_apptrace_flush_nolock(): enter
```

### More Information.

I tried to call esp_partition_read() to read and dump the partition to check if it can be accessed correctly. After verifying the first 16 bytes, it appears that the data I am expecting are there.
I also tried to call spi_flash_mmap() with the 3rd parameter being either SPI_FLASH_MMAP_INST or SPI_FLASH_MMAP_DATA. The partition is maped when using SPI_FLASH_MMAP_DATA, not when using SPI_FLASH_MMAP_INST.
Last, I tried to debug the function, and it appears that the error code is generated at the last steps of the debug when trying to find a place in the MMU pages or something like that.
Effectively the error comes from function spi_flash_mmap_pages() after the call to heap_caps_malloc()

Icarus113, 2022-12-15:
It's a bug in `flash_mmap.c`, the available vaddrs for instruction is only 2 pages. Could you please help do a 
```
uint32_t pages = spi_flash_mmap_get_free_pages(SPI_FLASH_MMAP_INST);
```
and let me know the `pages`? @vincentdupaquis

Icarus113, 2022-12-15:
And also modify this log level to print out it (just simply modify it to `ESP_EARLY_LOGI`), [this line](https://github.com/espressif/esp-idf/blob/master/components/bootloader_support/src/bootloader_utility.c#L878).

vincentdupaquis, 2022-12-15:
I've added the line you've requested, and the available pages are 0.
BTW, I had made some tests asking for the mapping of a single 4 bytes area, and it was failing too ...
for the other request you sent me, the line you pointed-out (line 878 in bootloader_utility.c) does not matches what I see in my v4.4.3 version.

Icarus113, 2022-12-16:
Thanks. The behaviour verifies this bug already. The returned `0 page` is because the project image takes around  2 pages (and then rounded up to 2 pages).

We'll fix this bug, and bring the bugfix to v4.4 later. Thanks for your report~ @vincentdupaquis

vincentdupaquis, 2022-12-16:
Thanks for being so reactive !

vincentdupaquis, 2022-12-16:
I'm eagerly waiting for this update to pursue

closed 2023-04-24 by Icarus113