espressif/esp-idf #10376: improving perfromance while using ble & wifi

manivalaguru, 2022-12-15:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

hello,
how do i improve ble & wifi perfromance much more in wrover 16mb.. right i have been advertising more than 200 beacons per second but only been getting 100 beacons on scan result when i use both wifi and ble.. is there any way to improve this perfromance so that i can get as much data as possibe?
Development Kit: custom board
Module or chip used: ESP32-WROVER
IDF version :4.3
Build System:Make
Compiler version :8.4
Operating System: Windows
(Windows only) environment type: MSYS2 mingw32.
Using an IDE?Yes eclipse IDE
thanks and regards
manilandan v

esp-lrh, 2024-02-23:
Hi, @manivalaguru 
This issue was created a long time ago, does the issue still exist and need to be solved?

KaeLL, 2024-02-23:
Weird question, given that it's a timeless issue, and it's also phrased that way.
How to improve BLE performance?
Seems like a cue for a new coex and/or BLE section under Performance in the docs.

esp-lrh, 2024-02-26:
> Weird question, given that it's a timeless issue, and it's also phrased that way. How to improve BLE performance? Seems like a cue for a new coex and/or BLE section under Performance in the docs.

Yes, this is a contradictory issue. WIFI and BLE is sharing a single RF module with time-division multiplexing method, it's normal for BLE performance to degrade when Wi-Fi is enabled.

We cannot adjust the time ratio arbitrarily between WIFI and BLE, because we need to ensure that both WIFI and BLE have enough time slices to maintain the connection (this is done in coexist module automatically). All we can do is prevent abnormal performance degradation in BLE.

esp-lrh, 2024-03-26:
Hi, @manivalaguru 

Due to no reply for a long time we will close this issue.

Thanks for you report, please feel free to reopen it if necessary.