espressif/esp-idf #10377: Web USB not Detecting for windows with ESP32S3 tiny USB stack.

nikhil-robinson, 2022-12-15:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I testing the tinyUSB device for esp32s3 in linux its ahowing corectly and am able to connect it using web usb but on windows i am a not able to claim the interface.
```

    tinyusb_config_t tusb_cfg = { 0 }; // the configuration uses default values
    ESP_ERROR_CHECK(tinyusb_driver_install(&tusb_cfg));

    tinyusb_config_cdcacm_t amc_cfg = {
        .usb_dev = TINYUSB_USBDEV_0,
        .cdc_port = TINYUSB_CDC_ACM_0,
        .rx_unread_buf_sz = 2048,
        .callback_rx = (tusb_cdcacm_callback_t)&tinyusb_cdc_rx_callback, // the first way to register a callback
        .callback_rx_wanted_char = NULL,
        .callback_line_state_changed = NULL,
        .callback_line_coding_changed = NULL
    };

    ESP_ERROR_CHECK(tusb_cdc_acm_init(&amc_cfg));
    /* the second way to register a callback */
    ESP_ERROR_CHECK(tinyusb_cdcacm_register_callback(
                        TINYUSB_CDC_ACM_0,
                        CDC_EVENT_LINE_STATE_CHANGED,
                        &tinyusb_cdc_line_state_changed_callback));

    ESP_LOGI(TAG, "USB initialization DONE");

```

this the despriport i am using.

In boot mode the webusb is able to detcct and connect to the esp32s3

chegewara, 2022-12-15:
Did you set USB version to 2.10?
```
tusb_desc_device_t {
.bcdUSB = 0x210
}
```
Its version when webUSB has been implemented and windows does not recognize it when its not set properly.

nikhil-robinson, 2022-12-27:
@chegewara  yes i tried changing this but still not able to claim the devce in web usb

roma-jam, 2022-12-28:
Hi @nikhil-robinson, 
Thanks for the information reporting. 

May I ask what error return the claimInterface(interfaceNumber) method, when invoked?

nikhil-robinson, 2022-12-28:
@roma-jam i am getting error "unable to claim the interface"

roma-jam, 2022-12-28:
@nikhil-robinson I need a few moments more to verify with what the problem is. 

Meanwhile, may I ask you the following: 
- What version of WebUSB do you use?
- How exactly did you try to set USB version to 2.10 (I mean with fully description of the device descriptor or in any other way)?

chegewara, 2022-12-29:
There is one more thing you could test, but im not sure now if its important, because i didnt test webUSB on windows for very long time.
I think i also had problems which i solved by adding USB-CDC together with webUSB, as a second interface.

roma-jam, 2022-12-29:
@chegewara Yes, thanks for the information, that makes sense. 

@nikhil-robinson Windows and Linux have different model of drivers with permission from the user space. 
May I ask you to verify this case (from the prospective that it is the same case you have): https://stackoverflow.com/questions/67910528/tried-to-connect-a-device-to-web-via-webusb-but-failed-to-claim-interface-acc

Windows highly unlikely gives an access to the USB devices from the web, except HID, printers and removable media devices (SmartCards and U2F FIDO also). 
For the connection to a CDC as a serial interface there could be a possible solution with doing one of the following steps: 
1. Using not a system driver for the device (use Zadig to change the driver from system to the winusb or libusb for direct connection, bypass the system driver in Windows)
2. Trying to use Web Serial API instead of Web USB API

Please, provide the update if it helps or you have any further questions.

roma-jam, 2023-01-02:
Hi everyone, 

We are closing this issue now. 
Feel free to reopen if necessary.

closed 2023-01-12 by Alvin1Zhang