espressif/esp-idf #10379: Regression on compilation error in 5.0 vs. 5.0-beta1,

eiffel31, 2022-12-15:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.0

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

The code compiles fine with 5.0-beta1

After installing 5.0, I have a compilation error (see below).

Switching back to 5.0-beta1: compilation does work.

Note that the faulty line in my code comes from example/system/ota/native_ota_example.
This example compiles without problem under 5.0
Maybe the new release is sensitive to include order? Require some more include?

### What is the actual behavior?

/home/philippe/.espressif/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -DUNITY_INCLUDE_CONFIG_H -Iconfig -I../main -I/users/philippe/dev/espressif/esp32/esp-idf_v5.0/components/newlib/platform_include -I/users/philippe/dev/espressif/esp32/esp-idf_v5.0/components/freertos/FreeRTOS-Kernel/include -I/users/philippe/dev/espressif/esp32/esp-idf_v5.0/components/freertos/esp_additions/include/freertos -I/users/philippe/dev/espressif/esp32/esp-idf_v5.0/components/freertos/FreeRTOS-Kernel/portable/xtensa/include -I/users/philippe/dev/espressif/esp32/esp-idf_v5.0/components/freertos/esp_additions/include -I/users/philippe/dev/espressif/esp32/esp-idf_v5.0/components/esp_hw_support/include -I/users/philippe/dev/espressif/esp32/esp-idf_v5.0/components/esp_hw_support/include/soc -I/users/philippe/dev/espressif/esp32/esp-idf_v5.0/components/esp_hw_support/include/soc/esp32 -I/users/philippe/dev/espressif/esp32/esp-idf_v5.0/components/esp_hw_support/port/esp32/. -I/users/philippe/dev/espressif/esp32/esp-idf_v5.0/components/esp_hw_support/port/esp32/private_include -I/users/philippe/dev/espressif/esp32/esp-idf_v5.0/components/heap/include -I/users/philippe/dev/espressif/esp32/esp-idf_v5.0/components/log/include -I/users/philippe/dev/espressif/esp32/esp-idf_v5.0/components/soc/include -I/users/philippe/dev/espressif/esp32/esp-idf_v5.0/components/soc/esp32/. -I/users/philippe/dev/espressif/esp32/esp-idf_v5.0/components/soc/esp32/include -I/users/philippe/dev/espressif/esp32/esp-idf_v5.0/components/hal/esp32/include -I/users/philippe/dev/espressif/esp32/esp-idf_v5.0/components/hal/include -I/users/philippe/dev/espressif/esp32/esp-idf_v5.0/components/hal/platform_port/include -I/users/philippe/dev/espressif/esp32/esp-idf_v5.0/components/esp_rom/include -I/users/philippe/dev/espressif/esp32/esp-idf_v5.0/components/esp_rom/include/esp32 -I/users/philippe/dev/espressif/esp32/esp-idf_v5.0/components/esp_rom/esp32 -I/users/philippe/dev/espressif/esp32/esp-idf_v5.0/components/esp_common/include -I/users/philippe/dev/espressif/esp32/esp-idf_v5.0/components/esp_system/include -I/users/philippe/dev/espressif/esp32/esp-idf_v5.0/components/esp_system/port/soc -I/users/philippe/dev/espressif/esp32/esp-idf_v5.0/components/esp_system/port/include/private -I/users/philippe/dev/espressif/esp32/esp-idf_v5.0/components/xtensa/include -I/users/philippe/dev/espressif/esp32/esp-idf_v5.0/components/xtensa/esp32/include -I/users/philippe/dev/espressif/esp32/esp-idf_v5.0/components/lwip/include -I/users/philippe/dev/espressif/esp32/esp-idf_v5.0/components/lwip/include/apps -I/users/philippe/dev/espressif/esp32/esp-idf_v5.0/components/lwip/include/apps/sntp -I/users/philippe/dev/espressif/esp32/esp-idf_v5.0/components/lwip/lwip/src/include -I/users/philippe/dev/espressif/esp32/esp-idf_v5.0/components/lwip/port/esp32/include -I/users/philippe/dev/espressif/esp32/esp-idf_v5.0/components/lwip/port/esp32/include/arch -I/users/philippe/dev/espressif/esp32/esp-idf_v5.0/components/esp_ringbuf/include -I/users/philippe/dev/espressif/esp32/esp-idf_v5.0/components/efuse/include -I/users/philippe/dev/espressif/esp32/esp-idf_v5.0/components/efuse/esp32/include -I/users/philippe/dev/espressif/esp32/esp-idf_v5.0/components/driver/include -I/users/philippe/dev/espressif/esp32/esp-idf_v5.0/components/driver/deprecated -I/users/philippe/dev/espressif/esp32/esp-idf_v5.0/components/driver/esp32/include -I/users/philippe/dev/espressif/esp32/esp-idf_v5.0/components/esp_pm/include -I/users/philippe/dev/espressif/esp32/esp-idf_v5.0/components/mbedtls/port/include -I/users/philippe/dev/espressif/esp32/esp-idf_v5.0/components/mbedtls/mbedtls/include -I/users/philippe/dev/espressif/esp32/esp-idf_v5.0/components/mbedtls/mbedtls/library -I/users/philippe/dev/espressif/esp32/esp-idf_v5.0/components/mbedtls/esp_crt_bundle/include -I/users/philippe/dev/espressif/esp32/esp-idf_v5.0/components/esp_app_format/include -I/users/philippe/dev/espressif/esp32/esp-idf_v5.0/components/bootloader_support/include -I/users/philippe/dev/espressif/esp32/esp-idf_v5.0/components/bootloader_support/bootloader_flash/include -I/users/philippe/dev/espressif/esp32/esp-idf_v5.0/components/esp_partition/include -I/users/philippe/dev/espressif/esp32/esp-idf_v5.0/components/app_update/include -I/users/philippe/dev/espressif/esp32/esp-idf_v5.0/components/spi_flash/include -I/users/philippe/dev/espressif/esp32/esp-idf_v5.0/components/pthread/include -I/users/philippe/dev/espressif/esp32/esp-idf_v5.0/components/esp_timer/include -I/users/philippe/dev/espressif/esp32/esp-idf_v5.0/components/app_trace/include -I/users/philippe/dev/espressif/esp32/esp-idf_v5.0/components/esp_event/include -I/users/philippe/dev/espressif/esp32/esp-idf_v5.0/components/nvs_flash/include -I/users/philippe/dev/espressif/esp32/esp-idf_v5.0/components/esp_phy/include -I/users/philippe/dev/espressif/esp32/esp-idf_v5.0/components/esp_phy/esp32/include -I/users/philippe/dev/espressif/esp32/esp-idf_v5.0/components/vfs/include -I/users/philippe/dev/espressif/esp32/esp-idf_v5.0/components/esp_netif/include -I/users/philippe/dev/espressif/esp32/esp-idf_v5.0/components/wpa_supplicant/include -I/users/philippe/dev/espressif/esp32/esp-idf_v5.0/components/wpa_supplicant/port/include -I/users/philippe/dev/espressif/esp32/esp-idf_v5.0/components/wpa_supplicant/esp_supplicant/include -I/users/philippe/dev/espressif/esp32/esp-idf_v5.0/components/esp_wifi/include -I/users/philippe/dev/espressif/esp32/esp-idf_v5.0/components/unity/include -I/users/philippe/dev/espressif/esp32/esp-idf_v5.0/components/unity/unity/src -I/users/philippe/dev/espressif/esp32/esp-idf_v5.0/components/cmock/CMock/src -I/users/philippe/dev/espressif/esp32/esp-idf_v5.0/components/console -I/users/philippe/dev/espressif/esp32/esp-idf_v5.0/components/http_parser -I/users/philippe/dev/espressif/esp32/esp-idf_v5.0/components/esp-tls -I/users/philippe/dev/espressif/esp32/esp-idf_v5.0/components/esp-tls/esp-tls-crypto -I/users/philippe/dev/espressif/esp32/esp-idf_v5.0/components/esp_adc/include -I/users/philippe/dev/espressif/esp32/esp-idf_v5.0/components/esp_adc/interface -I/users/philippe/dev/espressif/esp32/esp-idf_v5.0/components/esp_adc/esp32/include -I/users/philippe/dev/espressif/esp32/esp-idf_v5.0/components/esp_adc/deprecated/include -I/users/philippe/dev/espressif/esp32/esp-idf_v5.0/components/esp_eth/include -I/users/philippe/dev/espressif/esp32/esp-idf_v5.0/components/esp_gdbstub/include -I/users/philippe/dev/espressif/esp32/esp-idf_v5.0/components/esp_gdbstub/xtensa -I/users/philippe/dev/espressif/esp32/esp-idf_v5.0/components/esp_gdbstub/esp32 -I/users/philippe/dev/espressif/esp32/esp-idf_v5.0/components/esp_hid/include -I/users/philippe/dev/espressif/esp32/esp-idf_v5.0/components/tcp_transport/include -I/users/philippe/dev/espressif/esp32/esp-idf_v5.0/components/esp_http_client/include -I/users/philippe/dev/espressif/esp32/esp-idf_v5.0/components/esp_http_server/include -I/users/philippe/dev/espressif/esp32/esp-idf_v5.0/components/esp_https_ota/include -I/users/philippe/dev/espressif/esp32/esp-idf_v5.0/components/esp_lcd/include -I/users/philippe/dev/espressif/esp32/esp-idf_v5.0/components/esp_lcd/interface -I/users/philippe/dev/espressif/esp32/esp-idf_v5.0/components/protobuf-c/protobuf-c -I/users/philippe/dev/espressif/esp32/esp-idf_v5.0/components/protocomm/include/common -I/users/philippe/dev/espressif/esp32 [... cut ...]
  429 |         ESP_LOGW(TAG, "Configured OTA boot partition at offset 0x%08x, but running from offset 0x%08x",
      |         ^~~~~~~~
../main/rotation.c:430:1: error: format '%x' expects argument of type 'unsigned int', but argument 6 has type 'uint32_t' {aka 'long unsigned int'} [-Werror=format=]
  430 |                  configured->address, running->address);
      | ^          

### Steps to reproduce.

source $esp32/esp-idf_v5.0/export.sh

idf.py fullclean # In order to fully compile again with the new version
idf.py build

igrr, 2022-12-15:
This is one of the breaking changes in v5.0 described here: https://docs.espressif.com/projects/esp-idf/en/v5.0/esp32/migration-guides/release-5.x/gcc.html#int32-t-and-uint32-t-for-xtensa-compiler. You will need to adjust the format strings to use format specifiers from inttypes.h, for example `PRIu32` for `uint32_t`.
Example commit with such fixes: https://github.com/espressif/esp-idf/commit/55879e36ab3702eb950e0ace5088b2bee0b9d58e

eiffel31, 2022-12-15:
In native_ota_example.c, this line has not changed :
    ESP_LOGI(TAG, "Writing to partition subtype %d at offset 0x%x",
             update_partition->subtype, update_partition->address);

I do not understand why it is working fine in the example without applying the "fix" and it is not working in my project with the same line of code.

I suspect it depends on the includes.

It is strange to observe such a high impact change between beta1 and final release.

igrr, 2022-12-15:
> I do not understand why it is working fine in the example without applying the "fix" and it is not working in my project with the same line of code.

It is because the example CMakeLists.txt file had `target_compile_options(${COMPONENT_LIB} PRIVATE "-Wno-format")`, which silenced this warning. It was our mistake to leave these warnings silenced (but not fixed) in v5.0 release. We are fixing them now (e.g. in the commit I have linked to) and backporting the fixes to release/v5.0 branch.

> It is strange to observe such a high impact change between beta1 and final release.

Indeed, it's a rather large change which would have been better to merge before v5.0-beta1. Unfortunately this was done as part of the upgrade to the new toolchain release, which required a lot of changes in IDF. We haven't managed to merge it in time before v5.0-beta1. For the future toolchain upgrades, we will try to merge them earlier time before the beta release.

eiffel31, 2022-12-15:
OK, I missed the "-Wno-format" option because I checked the CMakeLists.txt from the project directory, not in the "main" directory.

Thanks for the support.

closed 2022-12-15 by eiffel31