espressif/esp-idf #10381: Debug on esp32s3 elf invalid

beckmx, 2022-12-15:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.0

### Operating System used.

macOS

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

I should be able to pick up a debug session on  VSCode

### What is the actual behavior?

There is a message appearing in VScode

```
Unable to start debugging. Program path '/Users/xx/idf-examples/hello_world/build/hello_world.elf' is missing or invalid.

GDB failed with message: I'm sorry, Dave, I can't do that.  Symbol format `elf32-xtensa-le' unknown.

This may occur if the process's executable was changed after the process was started, such as when installing an update. Try re-launching the application or restarting the machine.
```

### Steps to reproduce.

1. Compile on idf.py build
2. On VSCode configure the project to use gdb
3. Start the openocd command line
4. Connect VSCode to the already opened debug session
...

### Build or installation Logs.

```plain
My Launch.json file is:

{
    // Use IntelliSense to learn about possible attributes.
    // Hover to view descriptions of existing attributes.
    // For more information, visit: https://go.microsoft.com/fwlink/?linkid=830387
    "version": "0.2.0",
    "configurations": [
        {"name": "OpenOCD ESP32s3",
        "type": "cppdbg",
        "request": "launch",
        "cwd": "${workspaceFolder}/build",
        "program": "${workspaceFolder}/build/hello_world.elf",
        "miDebuggerPath": "/Users/xxx/.espressif/tools/xtensa-esp32s3-elf/esp-2022r1-11.2.0/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-gdb",
        "setupCommands": [
            {"text": "target extended-remote :3333"},
            {"text": "set remote hardware-watchpoint-limit 2"},
            {"text": "flushregs"}
        ]
    }
    ]
}

===========
My open ocd session logs this information:

openocd -f board/esp32s3-builtin.cfg
Open On-Chip Debugger v0.11.0-esp32-20221026 (2022-10-26-14:48)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : only one transport option; autoselect 'jtag'
Info : esp_usb_jtag: VID set to 0x303a and PID to 0x1001
Info : esp_usb_jtag: capabilities descriptor set to 0x2000
Warn : Transport "jtag" was already selected
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : esp_usb_jtag: serial (F4:12:FA:CB:8A:EC)
Info : esp_usb_jtag: Device found. Base speed 40000KHz, div range 1 to 255
Info : clock speed 40000 kHz
Info : JTAG tap: esp32s3.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32s3.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : starting gdb server for esp32s3.cpu0 on 3333
Info : Listening on port 3333 for gdb connections
Info : [esp32s3.cpu0] Debug controller was reset.
Info : [esp32s3.cpu0] Core was reset.
Info : [esp32s3.cpu1] Debug controller was reset.
Info : [esp32s3.cpu1] Core was reset.
Info : [esp32s3.cpu0] Target halted, PC=0x403788EA, debug_reason=00000000
Info : [esp32s3.cpu0] Reset cause (1) - (Power on reset)
Info : [esp32s3.cpu1] Target halted, PC=0x403788EA, debug_reason=00000000
Info : [esp32s3.cpu1] Reset cause (1) - (Power on reset)
Info : accepting 'gdb' connection on tcp/3333
Info : [esp32s3.cpu0] Target halted, PC=0x403788EA, debug_reason=00000000
Info : Set GDB target to 'esp32s3.cpu0'
Info : [esp32s3.cpu1] Target halted, PC=0x403788EA, debug_reason=00000000
Memory protection is enabled. Reset target to disable it...
Info : JTAG tap: esp32s3.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32s3.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : [esp32s3.cpu0] requesting target halt and executing a soft reset
Info : [esp32s3.cpu0] Debug controller was reset.
Info : [esp32s3.cpu0] Core was reset.
Info : [esp32s3.cpu0] Target halted, PC=0x500000EF, debug_reason=00000000
Info : [esp32s3.cpu0] Reset cause (3) - (Software core reset)
Info : [esp32s3.cpu1] requesting target halt and executing a soft reset
Info : [esp32s3.cpu0] Core was reset.
Info : [esp32s3.cpu0] Target halted, PC=0x40000400, debug_reason=00000000
Info : [esp32s3.cpu1] Debug controller was reset.
Info : [esp32s3.cpu1] Core was reset.
Info : [esp32s3.cpu1] Target halted, PC=0x40000400, debug_reason=00000000
Info : [esp32s3.cpu1] Reset cause (3) - (Software core reset)
Info : [esp32s3.cpu0] Reset cause (3) - (Software core reset)
Warn : No symbols for FreeRTOS!
Info : [esp32s3.cpu0] Target halted, PC=0x403B2482, debug_reason=00000001
Info : Flash mapping 0: 0x10020 -> 0x3c020020, 33 KB
Info : Flash mapping 1: 0x20020 -> 0x42000020, 100 KB
Info : [esp32s3.cpu0] Target halted, PC=0x403B2482, debug_reason=00000001
Info : Auto-detected flash bank 'esp32s3.cpu0.flash' size 16384 KB
Info : Using flash bank 'esp32s3.cpu0.flash' size 16384 KB
Info : [esp32s3.cpu0] Target halted, PC=0x403B2482, debug_reason=00000001
Info : Flash mapping 0: 0x10020 -> 0x3c020020, 33 KB
Info : Flash mapping 1: 0x20020 -> 0x42000020, 100 KB
Info : Using flash bank 'esp32s3.cpu0.irom' size 104 KB
Info : [esp32s3.cpu0] Target halted, PC=0x403B2482, debug_reason=00000001
Info : Flash mapping 0: 0x10020 -> 0x3c020020, 33 KB
Info : Flash mapping 1: 0x20020 -> 0x42000020, 100 KB
Info : Using flash bank 'esp32s3.cpu0.drom' size 36 KB
Info : New GDB Connection: 1, Target esp32s3.cpu0, state: halted
Info : dropped 'gdb' connection
```

### More Information.

I am using macOS only, still dont know if its because of that

igrr, 2022-12-16:
Please try to set GDB path as follows:
```
"miDebuggerPath": "/Users/xxx/.espressif/tools/xtensa-esp-elf-gdb/12.1_20221002/xtensa-esp-elf-gdb/bin/xtensa-esp32s3-elf-gdb"
```

While there is an xtensa-esp32-elf-gdb program in the xtensa-esp32s3-elf/esp-2022r1-11.2.0 directory as well, it's an older version that will eventually get removed. For example, it does not work on arm64 macOS.

(Normally IDF sets up the PATH variable so that `.espressif/tools/xtensa-esp-elf-gdb/12.1_20221002/xtensa-esp-elf-gdb/bin` is earlier in the paths list than `.espressif/tools/xtensa-esp32s3-elf/esp-2022r1-11.2.0/xtensa-esp32s3-elf/bin`, so this issue doesn't occur if you simply run `xtensa-esp32s3-elf-gdb` in console.)

beckmx, 2022-12-17:
@igrr got it, it's just that in this part I think the debug process seems a little bit confused when running the first time, but yeah it did work :) thanx

closed 2022-12-17 by beckmx