espressif/esp-idf #10384: FAILED: CMakeFiles/menuconfig /ot_br/build/CMakeFiles/menuconfig

app-z, 2022-12-16:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

idf5.1

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

I using Docker type of build. I successfully build sample
`sudo docker run --rm -v $PWD:/simple_sniffer -w /simple_sniffer espressif/idf idf.py build
`
Clean script also working well
`sudo docker run --rm -v $PWD:/simple_sniffer -w /simple_sniffer espressif/idf idf.py clean
`
But when I try
`sudo docker run --rm -v $PWD:/simple_sniffer -w /simple_sniffer espressif/idf idf.py menuconfig
`
there is error

WARNING: The TERM environment variable is not defined. The curses-based menuconfig will probably fail to run. Please consult the documentation of your terminal to set it up.

### What is the actual behavior?

See menuconfig screen. Before I already use idf toolchain but without Docker. Now I wanna move on Docker totally

I Also try mention another releases Ex. espressif/idf:release-v4.4 ... espressif/idf:release-v5.0 not sucesfuly build even
 How I can delete it from Docker now?

### Steps to reproduce.

1. Step
2. Step
3. Step
...

### Build or installation Logs.

```plain
dm@spbmap-ThinkPad:~/Documents/esp-idf/project/simple_sniffer$ sudo docker run --rm -v $PWD:/simple_sniffer -w /simple_sniffer espressif/idf idf.py menuconfig
Detecting the Python interpreter
Checking "python3" ...
Python 3.8.10
"python3" has been detected
Checking Python compatibility
Checking other ESP-IDF version.
Adding ESP-IDF tools to PATH...
Checking if Python packages are up to date...
Python requirements are satisfied.
Requirement files:
 - /opt/esp/idf/tools/requirements/requirements.core.txt
Python being checked: /opt/esp/python_env/idf5.1_py3.8_env/bin/python
Added the following directories to PATH:
  /opt/esp/idf/components/esptool_py/esptool
  /opt/esp/idf/components/espcoredump
  /opt/esp/idf/components/partition_table
  /opt/esp/idf/components/app_update
  /opt/esp/tools/xtensa-esp-elf-gdb/12.1_20221002/xtensa-esp-elf-gdb/bin
  /opt/esp/tools/riscv32-esp-elf-gdb/12.1_20221002/riscv32-esp-elf-gdb/bin
  /opt/esp/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin
  /opt/esp/tools/xtensa-esp32s2-elf/esp-2022r1-11.2.0/xtensa-esp32s2-elf/bin
  /opt/esp/tools/xtensa-esp32s3-elf/esp-2022r1-11.2.0/xtensa-esp32s3-elf/bin
  /opt/esp/tools/riscv32-esp-elf/esp-2022r1-11.2.0/riscv32-esp-elf/bin
  /opt/esp/tools/esp32ulp-elf/2.35_20220830/esp32ulp-elf/bin
  /opt/esp/tools/cmake/3.24.0/bin
  /opt/esp/tools/openocd-esp32/v0.11.0-esp32-20221026/openocd-esp32/bin
  /opt/esp/python_env/idf5.1_py3.8_env/bin
  /opt/esp/idf/tools
Done! You can now compile ESP-IDF projects.
Go to the project directory and run:

  idf.py build

[0/1] cd /simple_sniffer/build && /opt/esp/python_env/idf5.1_py3.8_env/bin/python /opt/esp/idf/tools/kconfig_new/prepare_kconfig_files.py --list-separator=semicolon --env-file /simple_sniffer/build/config.env && /opt/esp/python_env/idf5.1_py3.8_env/bin/python -m kconfgen --list-separator=semicolon --kconfig /opt/esp/idf/Kconfig --sdkconfig-rename /opt/esp/idf/sdkconfig.rename --config /simple_sniffer/sdkconfig --defaults /simple_sniffer/sdkconfig.defaults --env-file /simple_sniffer/build/config.env --env IDF_TARGET=esp32 --env IDF_ENV_FPGA= --dont-write-deprecated --output config /simple_sniffer/sdkconfig && /opt/esp/python_env/idf5.1_py3.8_env/bin/python /opt/esp/idf/tools/check_term.py && /opt/esp/tools/cmake/3.24.0/bin/cmake -E env COMPONENT_KCONFIGS_SOURCE_FILE=/simple_sniffer/build/kconfigs.in COMPONENT_KCONFIGS_PROJBUILD_SOURCE_FILE=/simple_sniffer/build/kconfigs_projbuild.in KCONFIG_CONFIG=/simple_sniffer/sdkconfig IDF_TARGET=esp32 IDF_ENV_FPGA= /opt/esp/python_env/idf5.1_py3.8_env/bin/python -m menuconfig /opt/esp/idf/Kconfig && /opt/esp/python_env/idf5.1_py3.8_env/bin/python -m kconfgen --list-separator=semicolon --kconfig /opt/esp/idf/Kconfig --sdkconfig-rename /opt/esp/idf/sdkconfig.rename --config /simple_sniffer/sdkconfig --defaults /simple_sniffer/sdkconfig.defaults --env-file /simple_sniffer/build/config.env --env IDF_TARGET=esp32 --env IDF_ENV_FPGA= --output config /simple_sniffer/sdkconfig
Loading defaults file /simple_sniffer/sdkconfig.defaults...
WARNING: The TERM environment variable is not defined. The curses-based menuconfig will probably fail to run. Please consult the documentation of your terminal to set it up.
Loaded configuration '/simple_sniffer/sdkconfig'
Traceback (most recent call last):
  File "/usr/lib/python3.8/runpy.py", line 194, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/usr/lib/python3.8/runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "/opt/esp/python_env/idf5.1_py3.8_env/lib/python3.8/site-packages/menuconfig.py", line 3278, in <module>
    _main()
  File "/opt/esp/python_env/idf5.1_py3.8_env/lib/python3.8/site-packages/menuconfig.py", line 663, in _main
    menuconfig(standard_kconfig(__doc__))
  File "/opt/esp/python_env/idf5.1_py3.8_env/lib/python3.8/site-packages/menuconfig.py", line 732, in menuconfig
    print(curses.wrapper(_menuconfig))
  File "/usr/lib/python3.8/curses/__init__.py", line 84, in wrapper
    stdscr = initscr()
  File "/usr/lib/python3.8/curses/__init__.py", line 29, in initscr
    setupterm(term=_os.environ.get("TERM", "unknown"),
_curses.error: setupterm: could not find terminal
FAILED: CMakeFiles/menuconfig /simple_sniffer/build/CMakeFiles/menuconfig 
cd /simple_sniffer/build && /opt/esp/python_env/idf5.1_py3.8_env/bin/python /opt/esp/idf/tools/kconfig_new/prepare_kconfig_files.py --list-separator=semicolon --env-file /simple_sniffer/build/config.env && /opt/esp/python_env/idf5.1_py3.8_env/bin/python -m kconfgen --list-separator=semicolon --kconfig /opt/esp/idf/Kconfig --sdkconfig-rename /opt/esp/idf/sdkconfig.rename --config /simple_sniffer/sdkconfig --defaults /simple_sniffer/sdkconfig.defaults --env-file /simple_sniffer/build/config.env --env IDF_TARGET=esp32 --env IDF_ENV_FPGA= --dont-write-deprecated --output config /simple_sniffer/sdkconfig && /opt/esp/python_env/idf5.1_py3.8_env/bin/python /opt/esp/idf/tools/check_term.py && /opt/esp/tools/cmake/3.24.0/bin/cmake -E env COMPONENT_KCONFIGS_SOURCE_FILE=/simple_sniffer/build/kconfigs.in COMPONENT_KCONFIGS_PROJBUILD_SOURCE_FILE=/simple_sniffer/build/kconfigs_projbuild.in KCONFIG_CONFIG=/simple_sniffer/sdkconfig IDF_TARGET=esp32 IDF_ENV_FPGA= /opt/esp/python_env/idf5.1_py3.8_env/bin/python -m menuconfig /opt/esp/idf/Kconfig && /opt/esp/python_env/idf5.1_py3.8_env/bin/python -m kconfgen --list-separator=semicolon --kconfig /opt/esp/idf/Kconfig --sdkconfig-rename /opt/esp/idf/sdkconfig.rename --config /simple_sniffer/sdkconfig --defaults /simple_sniffer/sdkconfig.defaults --env-file /simple_sniffer/build/config.env --env IDF_TARGET=esp32 --env IDF_ENV_FPGA= --output config /simple_sniffer/sdkconfig
ninja: build stopped: subcommand failed.
Executing action: menuconfig
Running ninja in directory /simple_sniffer/build
Executing "ninja menuconfig"...
ninja failed with exit code 1
```

### More Information.

Ubuntu 18.04.6 LTS 

echo $TERM
xterm-256color

igrr, 2022-12-16:
Menuconfig is an interactive program, so you need to launch Docker with `-i -t` flags. Without that, Docker will not allocate a TTY to the program running in the container.

The following should work:
```
docker run --rm -v $PWD:/simple_sniffer -w /simple_sniffer -it espressif/idf idf.py menuconfig
```

> I Also try mention another releases Ex. espressif/idf:release-v4.4 ... espressif/idf:release-v5.0 not sucesfuly build even

You may need to delete the build directory when switching between the release branches. If that doesn't help, please post the error output you get.

app-z, 2022-12-16:
Thanks, option -it allow me run menuconfig

build with mention release still problem

`sudo docker run --rm -v $PWD:/simple_sniffer -w /simple_sniffer espressif/idf:release-v4.4 idf.py build
`

Detecting the Python interpreter
Checking "python" ...
Python 3.8.10
"python" has been detected
Adding ESP-IDF tools to PATH...
Using Python interpreter in /opt/esp/python_env/idf4.4_py3.8_env/bin/python
Checking if Python packages are up to date...
Python requirements from /opt/esp/idf/requirements.txt are satisfied.
Added the following directories to PATH:
  /opt/esp/idf/components/esptool_py/esptool
  /opt/esp/idf/components/espcoredump
  /opt/esp/idf/components/partition_table
  /opt/esp/idf/components/app_update
  /opt/esp/tools/xtensa-esp-elf-gdb/11.2_20220823/xtensa-esp-elf-gdb/bin
  /opt/esp/tools/riscv32-esp-elf-gdb/11.2_20220823/riscv32-esp-elf-gdb/bin
  /opt/esp/tools/xtensa-esp32-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32-elf/bin
  /opt/esp/tools/xtensa-esp32s2-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32s2-elf/bin
  /opt/esp/tools/xtensa-esp32s3-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32s3-elf/bin
  /opt/esp/tools/riscv32-esp-elf/esp-2021r2-patch5-8.4.0/riscv32-esp-elf/bin
  /opt/esp/tools/esp32ulp-elf/2.35_20220830/esp32ulp-elf/bin
  /opt/esp/tools/cmake/3.23.1/bin
  /opt/esp/tools/openocd-esp32/v0.11.0-esp32-20221026/openocd-esp32/bin
  /opt/esp/python_env/idf4.4_py3.8_env/bin
  /opt/esp/idf/tools
Done! You can now compile ESP-IDF projects.
Go to the project directory and run:

  idf.py build

[0/1] Re-running CMake...
/bin/sh: 1: /opt/esp/tools/cmake/3.24.0/bin/cmake: not found
FAILED: build.ninja 
/opt/esp/tools/cmake/3.24.0/bin/cmake --regenerate-during-build -S/simple_sniffer -B/simple_sniffer/build
ninja: error: rebuilding 'build.ninja': subcommand failed
ninja failed with exit code 1
Executing action: all (aliases: build)
Running ninja in directory /simple_sniffer/build
Executing "ninja all"...

app-z, 2022-12-16:
`sudo docker run --rm -v $PWD:/simple_sniffer -w /simple_sniffer espressif/idf:release-v5.0 idf.py build`

Detecting the Python interpreter
Checking "python3" ...
Python 3.8.10
"python3" has been detected
Checking Python compatibility
Checking other ESP-IDF version.
Adding ESP-IDF tools to PATH...
Checking if Python packages are up to date...
Python requirements are satisfied.
Requirement files:
 - /opt/esp/idf/tools/requirements/requirements.core.txt
Python being checked: /opt/esp/python_env/idf5.0_py3.8_env/bin/python
Added the following directories to PATH:
  /opt/esp/idf/components/esptool_py/esptool
  /opt/esp/idf/components/espcoredump
  /opt/esp/idf/components/partition_table
  /opt/esp/idf/components/app_update
  /opt/esp/tools/xtensa-esp-elf-gdb/11.2_20220823/xtensa-esp-elf-gdb/bin
  /opt/esp/tools/riscv32-esp-elf-gdb/11.2_20220823/riscv32-esp-elf-gdb/bin
  /opt/esp/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin
  /opt/esp/tools/xtensa-esp32s2-elf/esp-2022r1-11.2.0/xtensa-esp32s2-elf/bin
  /opt/esp/tools/xtensa-esp32s3-elf/esp-2022r1-11.2.0/xtensa-esp32s3-elf/bin
  /opt/esp/tools/riscv32-esp-elf/esp-2022r1-11.2.0/riscv32-esp-elf/bin
  /opt/esp/tools/esp32ulp-elf/2.35_20220830/esp32ulp-elf/bin
  /opt/esp/tools/cmake/3.24.0/bin
  /opt/esp/tools/openocd-esp32/v0.11.0-esp32-20221026/openocd-esp32/bin
  /opt/esp/python_env/idf5.0_py3.8_env/bin
  /opt/esp/idf/tools
Done! You can now compile ESP-IDF projects.
Go to the project directory and run:

  idf.py build

Executing action: all (aliases: build)
Running ninja in directory /simple_sniffer/build
Executing "ninja all"...
[0/1] Re-running CMake...
CMake Error at /opt/esp/idf/tools/cmake/project.cmake:243 (message):
  Directory specified in EXTRA_COMPONENT_DIRS doesn't exist:
  /opt/esp/idf/examples/ethernet/basic/components/ethernet_init
Call Stack (most recent call first):
  /opt/esp/idf/tools/cmake/project.cmake:327 (__project_init)
  CMakeLists.txt:8 (project)

-- Configuring incomplete, errors occurred!
See also "/simple_sniffer/build/CMakeFiles/CMakeOutput.log".
FAILED: build.ninja 
/opt/esp/tools/cmake/3.24.0/bin/cmake --regenerate-during-build -S/simple_sniffer -B/simple_sniffer/build
ninja: error: rebuilding 'build.ninja': subcommand failed
ninja failed with exit code 1, output of the command is in the /simple_sniffer/build/log/idf_py_stderr_output_161 and /simple_sniffer/build/log/idf_py_stdout_output_161

igrr, 2022-12-16:
> Directory specified in EXTRA_COMPONENT_DIRS doesn't exist:
> /opt/esp/idf/examples/ethernet/basic/components/ethernet_init

It looks like you are trying to compile an example provided in IDF `master` branch with IDF v5.0. Since in the `master` branch the example relies on another component which is in a separate directory, this same example doesn't work in IDF v5.0 where that directory doesn't exist. If you need to build the example with IDF v5.0, you also need to pick the example project from IDF v5.0.

igrr, 2022-12-16:
> /bin/sh: 1: /opt/esp/tools/cmake/3.24.0/bin/cmake: not found

And this issue is related to the point I mentioned above: you need to delete the `build` directory before you start building with a different release. (This is related to the fact that CMakeCache.txt and build.ninja files contain paths to specific tools, which are different between releases.)

app-z, 2022-12-16:
Thanks
Will try one more time

`sudo docker run --rm -v $PWD:/ot_br -w /ot_br -it espressif/idf idf.py  -p /dev/ttyUSB0  flash`

Am I right try flash? But I see error. Try couple example

Detecting the Python interpreter
Checking "python3" ...
Python 3.8.10
"python3" has been detected
Checking Python compatibility
Checking other ESP-IDF version.
Adding ESP-IDF tools to PATH...
Checking if Python packages are up to date...
Requirement files:
 - /opt/esp/idf/tools/requirements/requirements.core.txt
Python being checked: /opt/esp/python_env/idf5.1_py3.8_env/bin/python
Python requirements are satisfied.
Added the following directories to PATH:
  /opt/esp/idf/components/esptool_py/esptool
  /opt/esp/idf/components/espcoredump
  /opt/esp/idf/components/partition_table
  /opt/esp/idf/components/app_update
  /opt/esp/tools/xtensa-esp-elf-gdb/12.1_20221002/xtensa-esp-elf-gdb/bin
  /opt/esp/tools/riscv32-esp-elf-gdb/12.1_20221002/riscv32-esp-elf-gdb/bin
  /opt/esp/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin
  /opt/esp/tools/xtensa-esp32s2-elf/esp-2022r1-11.2.0/xtensa-esp32s2-elf/bin
  /opt/esp/tools/xtensa-esp32s3-elf/esp-2022r1-11.2.0/xtensa-esp32s3-elf/bin
  /opt/esp/tools/riscv32-esp-elf/esp-2022r1-11.2.0/riscv32-esp-elf/bin
  /opt/esp/tools/esp32ulp-elf/2.35_20220830/esp32ulp-elf/bin
  /opt/esp/tools/cmake/3.24.0/bin
  /opt/esp/tools/openocd-esp32/v0.11.0-esp32-20221026/openocd-esp32/bin
  /opt/esp/python_env/idf5.1_py3.8_env/bin
  /opt/esp/idf/tools
Done! You can now compile ESP-IDF projects.
Go to the project directory and run:

  idf.py build

Executing action: flash
Running ninja in directory /ot_br/build
Executing "ninja flash"...
[1/5] cd /ot_br/build/esp-idf/esptool_py && /opt/esp/python_env/idf5.1_py3.8_env/bin/python /opt/esp/idf/components/partition_table/check_sizes.py --offset 0x8000 partition --type app /ot_br/build/partition_table/partition-table.bin /ot_br/build/esp_ot_br.bin
esp_ot_br.bin binary size 0x14d230 bytes. Smallest app partition is 0x177000 bytes. 0x29dd0 bytes (11%) free.
[2/5] Performing build step for 'bootloader'
[1/1] cd /ot_br/build/bootloader/esp-idf/esptool_py && /opt/esp/python_env/idf5.1_py3.8_env/bin/python /opt/esp/idf/components/partition_table/check_sizes.py --offset 0x8000 bootloader 0x1000 /ot_br/build/bootloader/bootloader.bin
Bootloader binary size 0x6720 bytes. 0x8e0 bytes (8%) free.
[2/3] cd /opt/esp/idf/components/esptool_py && /opt/esp/tools/cmake/3.24.0/bin/cmake -D IDF_PATH=/opt/esp/idf -D "SERIAL_TOOL=/opt/esp/python_env/idf5.1_py3.8_env/bin/python;;/opt/esp/idf/components/esptool_py/esptool/esptool.py;--chip;esp32" -D "SERIAL_TOOL_ARGS=--before=default_reset;--after=hard_reset;write_flash;@flash_args" -D WORKING_DIRECTORY=/ot_br/build -P /opt/esp/idf/components/esptool_py/run_serial_tool.cmake
esptool esp32 -p /dev/ttyUSB0 -b 460800 --before=default_reset --after=hard_reset write_flash --flash_mode dio --flash_freq 40m --flash_size 2MB 0x1000 bootloader/bootloader.bin 0x10000 esp_ot_br.bin 0x8000 partition_table/partition-table.bin
esptool.py v4.5.dev0
Serial port /dev/ttyUSB0

A fatal error occurred: Could not open /dev/ttyUSB0, the port doesn't exist
CMake Error at run_serial_tool.cmake:55 (message):
  
  /opt/esp/python_env/idf5.1_py3.8_env/bin/python;;/opt/esp/idf/components/esptool_py/esptool/esptool.py;--chip;esp32
  failed

FAILED: CMakeFiles/flash /ot_br/build/CMakeFiles/flash 
cd /opt/esp/idf/components/esptool_py && /opt/esp/tools/cmake/3.24.0/bin/cmake -D IDF_PATH=/opt/esp/idf -D "SERIAL_TOOL=/opt/esp/python_env/idf5.1_py3.8_env/bin/python;;/opt/esp/idf/components/esptool_py/esptool/esptool.py;--chip;esp32" -D "SERIAL_TOOL_ARGS=--before=default_reset;--after=hard_reset;write_flash;@flash_args" -D WORKING_DIRECTORY=/ot_br/build -P /opt/esp/idf/components/esptool_py/run_serial_tool.cmake
ninja: build stopped: subcommand failed.
ninja failed with exit code 1, output of the command is in the /ot_br/build/log/idf_py_stderr_output_161 and /ot_br/build/log/idf_py_stdout_output_161

app-z, 2022-12-16:
#ls /dev/ttyUSB0 
/dev/ttyUSB0

I see port is ok

#sudo usermod -a -G dialout $USER
Did it also

igrr, 2022-12-16:
You need to pass the serial device into Docker to make it available inside the container. Please refer to Docker documentation for this.

Note that passing serial devices into Docker containers only works on Linux hosts (see [note](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/tools/idf-docker-image.html#using-the-image-interactively) in the docs).

app-z, 2022-12-17:
Yes!

I totally forgotten about I under Docker. There is many way mention USB to Docker. I use --device=/dev/ttyUSB0

`sudo docker run --device=/dev/ttyUSB0 --rm -v $PWD:/simple_sniffer -w /simple_sniffer -it espressif/idf idf.py -p /dev/ttyUSB0 flash`

About option mention releases for ex. espressif/idf:release-v5.0. Look like necessary exactly sample from exactly release version. And make clean not enough. Necessary delete folder build

Many thanks!!

closed 2022-12-18 by app-z

ligantx, 2023-03-21:
> Menuconfig is an interactive program, so you need to launch Docker with `-i -t` flags. Without that, Docker will not allocate a TTY to the program running in the container.
>
> The following should [...]

Just an addition to this issue in case other people are facing:
If you want to pass menuconfig to a remote client, you should add:
`ssh **-t** user@remoteserver "docker exec/run ...."`