espressif/esp-idf #10385: wifi multiple antenna switch GPIO doesn't work on ESP32-S3

chndao77, 2022-12-16:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I am trying add a FEM to esp32s3, 
I found following code, works on ESP32 chip, but doesn't work on ESP32-S3, 
no error report, 
on ESP32 chip, tx/rx gpios work great,: when tx is high, rx is low, and when tx is low, rx is high
on ESP32-S3, tx/rx gpios keep low.
lastest esp-idf git version and lastest wifi lib.

`#include "driver/gpio.h"
#include "esp_rom_gpio.h"
void gpio_out_tx_status()
{
wifi_ant_gpio_config_t config_gpio;

//GPIO pull down initially for TX
 esp_rom_gpio_pad_select_gpio(15);
 gpio_set_direction(15, GPIO_MODE_OUTPUT);
 gpio_set_level(15, 0);
 //GPIO pull down initially for RX
 esp_rom_gpio_pad_select_gpio(16);
 gpio_set_direction(16, GPIO_MODE_OUTPUT);
 gpio_set_level(16, 0);

config_gpio.gpio_cfg[0].gpio_select = 1;
config_gpio.gpio_cfg[0].gpio_num = 15;
config_gpio.gpio_cfg[1].gpio_select = 1;
config_gpio.gpio_cfg[1].gpio_num = 16;
wifi_ant_config_t config_ant = {
.rx_ant_mode = WIFI_ANT_MODE_ANT1,
.rx_ant_default = WIFI_ANT_MODE_ANT1,
.tx_ant_mode = WIFI_ANT_MODE_ANT0,
.enabled_ant0 = 1,
.enabled_ant1 = 2
};
esp_wifi_set_ant_gpio(&config_gpio);
esp_wifi_set_ant(&config_ant);
}

static int wifi_cmd_ap(int argc, char **argv)
{
    int nerrors = arg_parse(argc, argv, (void **) &ap_args);

    if (nerrors != 0) {
        arg_print_errors(stderr, ap_args.end, argv[0]);
        return 1;
    }

    wifi_cmd_ap_set(ap_args.ssid->sval[0], ap_args.password->sval[0]);
    ESP_LOGI(TAG, "AP mode, %s %s", ap_args.ssid->sval[0], ap_args.password->sval[0]);

    gpio_out_tx_status();
    ESP_LOGI(TAG, "AP mode, load ATT config");    
    return 0;
}`

sbuto1, 2023-04-20:
Has there been any investigation into this issue? I can confirm it is present when using ESP-IDF v4.4.4 running on a ESP32-S3-DevKitC-1. The same code has no issue on a ESP32. I also tried a variety of GPIO pins on the board but all had the same behavior (net remains low at all times). Thanks

MaxwellAlan, 2023-09-08:
Hi @sbuto1 

Thanks for your report, we have fixed this bug internal. Plz wait for the fix to sync to Github.

Thanks.

sbuto1, 2023-09-14:
@MaxwellAlan, thanks for checking on & fixing this issue! Since submitting this I also discovered that the ESP32-C3 also has the issue, so it is not specific to ESP32-S3 line - hopefully the fix is applied to both/all chip families. 

FYI, there is also a small bug in the new wifi antenna example projects - I noted that in issue IDFGH-10851.

Xiehanxin, 2023-10-19:
hi @sbuto1 could I close this issue? if there are any update can reopen it.

sbuto1, 2023-10-25:
Hi , I'm not the original poster so I don't have the ability to close the issue. I did post a very similar issue, https://github.com/espressif/esp-idf/issues/12055 (which has not been marked as done).  In both cases I still see the issue in the most recent public ESP-IDF release, v5.1.1. have these issues been fixed in any public commit? If not I would suggest keeping them open. Thanks!

Xiehanxin, 2023-10-31:
sorry @sbuto1 ，we will give feedback to https://github.com/espressif/esp-idf/issues/12055, and hi @chndao77 ,could I close this issue? if there are any update can reopen it.

mhdong, 2023-11-01:
Hi @chndao77 fix on master branch is available at https://github.com/espressif/esp-idf/commit/16ec59907e9abf95981e87ca3c9154be2325e245, we are also back porting to other release branches.
Will update further if any progress. Thanks.

closed 2023-11-01 by Alvin1Zhang