espressif/esp-idf #10387: [NVS] Is it possible to have multiple encrypted NVS Partitions?

chipweinberger, 2022-12-16:
### Is your feature request related to a problem?

ESP-IDF v4.4.3

I want 2 encrypted NVS partitions.

- a typical one, for holding device settings, wifi passwords, cleared on factory reset.
- a factory one, for holding certificates, keys, etc, `not` cleared on factory reset.

the nvs_keys subtype, will it always apply to both partitions? or just the partition named "nvs"?

chipweinberger, 2022-12-16:
The docs have this to say, but it's pretty vague. 

> It is possible for an application to use different keys for different NVS partitions and thereby have multiple key-partitions. However, it is a responsibility of the application to provide correct key-partition/keys for the purpose of encryption/decryption.

How do I specify which partition uses which key?

also - are multiple nvs partitions *well* supported? I don't want to hit bugs!

chipweinberger, 2022-12-16:
Thank you for the example. It is very useful. 

Is my understanding correct:
- if a partition is named "nvs" it will be encrypted by default, no code changes necessary, when flash encryption is enabled
- other partitions named xxxx, will only be encrypted if you use the code from your example, and will otherwise not be encrypted.

mahavirj, 2022-12-16:
> if a partition is named "nvs" it will be encrypted by default, no code changes necessary, when flash encryption is enabled

Yes. This has been the behavior from IDF v4.3 on-wards. 

> for other partitions named xxxx, it will only be encrypted if you use the code from your example, and will otherwise not be encrypted.

Yes, for custom NVS partitions one must use `nvs_flash_secure_*` APIs for the initialization

This is documented here: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/storage/nvs_flash.html#encrypted-read-write

closed 2022-12-16 by chipweinberger