espressif/esp-idf #10388: I can compile but I cannot invoke idf.py

mbastida123, 2022-12-16:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hi, I'm on a Windows machine using IDF v4.4.3 on vscode. Through VSCode I can compile with no issues using the build buttons.

However, today I tried to invoke idf.py through the command line and this window appeared:
[image]
If I run install.bat nothing changes, everything seems to be installed correctly.

As for the enviorment variables this is what I have configured:
[image]
And on the Path:
[image]

dobairoland, 2022-12-16:
@mbastida123 Please use one of the command line environments which were installed by the installer. Everything is set up there to run `idf.py` commands.

You can find them in the Start menu. "ESP-IDF" is the name.

mbastida123, 2022-12-16:
Hi @dobairoland 

With command line enviorment do you mean the python env?
In the start menu I have nothing:
[image]

If I activate the` idf4.4_py3.8_env` enviorment located in `c:\Users\Marti\esp\.espressif\python_env\idf4.4_py3.8_env\Scripts\activate.bat` I have the same result. But I don't know if that's what you are refering to.

dobairoland, 2022-12-16:
None of the above.

ESP-IDF command prompt or PowerShell.

https://youtu.be/byVPAfodTyY?t=159

mbastida123, 2022-12-16:
> None of the above.
> 
> ESP-IDF command prompt or PowerShell.
> 
> https://youtu.be/byVPAfodTyY?t=159

I tried searching for idf_cmd_init.bat in the esp folder and I didn't find any file. Is there any chance that the tool was not downloaded?

This is my .espressif folder contents by the way: 
[image]

dobairoland, 2022-12-16:
Please take a look at the whole video how to install ESP-IDF tools. The installers can be found here: https://dl.espressif.com/dl/esp-idf/

You can find many helpful articles in the Vscode plugin repository: https://github.com/espressif/vscode-esp-idf-extension

In case you want to set-up environment separately from Vscode then here is the official documentation: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/get-started/windows-setup.html

closed 2022-12-16 by dobairoland