espressif/esp-idf #10389: Cant connect via PowerShell console using "idf.py build"

Coffeeye, 2022-12-16:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4-dev-3978-gf0628ea2db

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

PowerShell

### What is the expected behavior?

Build via PowerShell console using "idf.py build -p COM3 -b 921600 flash monitor"

### What is the actual behavior?

> Serial port COM3
> Connecting......................................
> 
> A fatal error occurred: Failed to connect to ESP32: No serial data received.
> For troubleshooting steps visit: https://github.com/espressif/esptool#troubleshooting
> CMake Error at run_serial_tool.cmake:56 (message):
>   python
>   C:/Users/.../esp-idf/components/esptool_py/esptool/esptool.py
>   --chip esp32 failed
> 
> FAILED: CMakeFiles/flash

### Steps to reproduce.

1. My PowerShell script file contains this line : "cmd /c "$env:IDF_PATH\export.bat && PowerShell""
2. Iam running script and preparing console with link "C:\Windows\System32\WindowsPowerShell\v1.0\powershell.exe -ExecutionPolicy Bypass -File "%PROJ_DIR%\console.ps1"
3. type -> "idf.py -p COM3 -b 921600 flash monitor"

### Build or installation Logs.

```plain
PS C:\Users\...> idf.py -p COM3 -b 921600 flash monitor
Executing action: flash
Running ninja in directory c:\...\build
Executing "ninja flash"...
[1/5] cmd.exe /C "cd /D C:\.../build/PROJ.bin"
PROJ.bin binary size 0x26cfa0 bytes. Smallest app partition is 0x300000 bytes. 0x93060 bytes (19%) free.
[2/5] Performing build step for 'bootloader'
[1/1] cmd.exe /C "cd /D C:\...\build\bootloader\esp-idf\esptool_py && python C:/.../esp-idf/components/partition_table/check_sizes.py --offset 0x8000 bootloader 0x1000 C:/.../build/bootloader/bootloader.bin"
Bootloader binary size 0x5cb0 bytes. 0x1350 bytes (17%) free.
[2/3] cmd.exe /C "cd /D C:\.../esp-idf/components/esptool_py/run_serial_tool.cmake"
esptool.py esp32 -p COM3 -b 921600 --before=default_reset --after=hard_reset write_flash --flash_mode dio --flash_freq 40m --flash_size 16MB 0x1000 bootloader/bootloader.bin 0x10000 PROJ.bin 0x8000 partition_table/partition-table.bin 0xd000 ota_data_initial.bin
esptool.py v3.2-dev
Serial port COM3
Connecting......................................

A fatal error occurred: Failed to connect to ESP32: No serial data received.
For troubleshooting steps visit: https://github.com/espressif/esptool#troubleshooting
CMake Error at run_serial_tool.cmake:56 (message):
  python
  C:/.../esp-idf/components/esptool_py/esptool/esptool.py
  --chip esp32 failed

FAILED: CMakeFiles/flash
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
```

### More Information.

Sometimes it works, sometimes it does not. I have no idea whats the case. Building with Eclipse passes everytime but not with console command.

radimkarnis, 2022-12-16:
Hello @Coffeeye,
the issue happens during the flashing phase when IDF uses [`esptool.py`](https://github.com/espressif/esptool) to connect to the chip and flash the compiled binaries.

It's this command:

```
esptool.py esp32 -p COM3 -b 921600 --before=default_reset --after=hard_reset write_flash --flash_mode dio --flash_freq 40m --flash_size 16MB 0x1000 bootloader/bootloader.bin 0x10000 PROJ.bin 0x8000 partition_table/partition-table.bin 0xd000 ota_data_initial.bin
```

IDF `v4.4-dev-3978-gf0628ea2db` actually uses a quite old version of `esptool.py`, that's why it's not clear what's happening.

My best guess is that your chip is not [entering the bootloader (switching into a download mode)](https://docs.espressif.com/projects/esptool/en/latest/esp32/esptool/entering-bootloader.html#entering-the-bootloader) automatically. If your devkit has buttons, you can try holding the `BOOT/IO0` button and running the operation again. If this fails, try holding the `BOOT/IO0` button and pressing the `RST/EN` button before running the operation.

You can also try using a newer `esptool.py` version. You can [install it](https://docs.espressif.com/projects/esptool/en/latest/esp32/installation.html) and run the above-mentioned command manually. This might help or provide more information about the problem.

Coffeeye, 2022-12-19:
Hello @radimkarnis thanks for your answear!, 
unfortunatly even after runing **pip** to install **esptool** I didnt manage to use the command you mentioned with success. 

I've read the idea of download mode to flash the program, point is that I've prepared PCB to work with those condition. 
The program running on my ESP restarts every time I press enter to start the **idf.py** procedure, but it rarely succeeds to start flashing it, _since I changed my environment_.

radimkarnis, 2022-12-19:
When using the new version of esptool, did you get any messages, such as `Wrong boot mode detected (0xXX)! The chip needs to be in download mode.`? Could you please try running with the `--trace` argument and post the resulting log here?

> Building with Eclipse passes everytime but not with console command.

So you are able to **flash** the board every time when using eclipse? Maybe you could try to see which version of esptool is used when you use eclipse and try to install that in your command line environment.

Coffeeye, 2022-12-19:
`A fatal error occurred: Failed to connect to ESP32: Wrong boot mode detected (0x17)! The chip needs to be in download mode.`

This is the message I'am getting after updating **esptool**

radimkarnis, 2022-12-19:
@Coffeeye your board needs auto-reset circuitry in order for the automatic reset to download mode to work - this is described in detail [here](https://docs.espressif.com/projects/esptool/en/latest/esp32/advanced-topics/boot-mode-selection.html#automatic-bootloader).

If your board doesn't support this, the only option is to do it [manually](https://docs.espressif.com/projects/esptool/en/latest/esp32/advanced-topics/boot-mode-selection.html#manual-bootloader). This can be done either by pressing the buttons or bridging GPIO0 to GND and resetting the chip.

Coffeeye, 2022-12-20:
[This](https://randomnerdtutorials.com/solved-failed-to-connect-to-esp32-timed-out-waiting-for-packet-header/)
solved the problem.

Wierd part is that before I've changed the env it was working without it. 

Custom Board.

closed 2022-12-20 by Coffeeye