espressif/esp-idf #10390: esp-camera causes ESP32S3 WiFi hang in *some* AP

wuyuanyi135, 2022-12-16:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1-dev-2355-gedd815af2e

### Operating System used.

Windows

### How did you build your project?

CLion IDE

### If you are using Windows, please specify command line type.

PowerShell

### Development Kit.

ESP32-S3-WROOM-1-N16R2

### Power Supply used.

External 3.3V

### What is the expected behavior?

The ESP32S3 (N16R2) allows WiFi and esp-camera module to run on *any* hotspot.

### What is the actual behavior?

When `esp-camera` and WiFi are enabled simultaneously, the module cannot connect to the hotspot generated by Windows Mobile Hotspot but **can connect to my home router.**

If I initialize esp-camera library after WiFi has been connected, the WiFi becomes nonresponsive, with kernel still running (I used a timer to print periodically). However, pinging, http access, etc. all stop working on the ESP32. Also, there is no more log output from wifi task even in verbose level.

### Steps to reproduce.

1. Create Windows Mobile Hotspot
2. Let the esp to connect to it
3. initialize the camera library and it will no longer respond to the ping.
(See the details below)

### More Information.

I am developing a product with ESP32S3 module (N16R2) that connects to a Windows host computer using Windows Mobile Hotspot feature. The ESP32S3 runs fine until I recently introduced the `esp-camera` library (version 2.0.3) into my project. 

When esp-camera library and WiFi are used together, the esp cannot connect to the WiFi spawned by Windows Mobile Hotspot, with flooding WIFI_EVENT_STA_DISCONNECTED and `reason=2`. **It can still connect to my router normally.**

I bisected where the problem occurs, it turns out if I comment out the following lines in the `esp-camera` source code, both Windows Mobile Hotspot and home router can be connected successfully. However, I haven't test if the camera still works if those lines are removed. These lines involves some low level register operation that's why I believe this issue should be posted here.

https://github.com/espressif/esp32-camera/blob/402b811b835cd348343b567a97fdf984c9d16fb9/target/esp32s3/ll_cam.c#L238-L243

I also attached the logging output from 
1. [failed connecting to windows mobile hotspot](https://github.com/espressif/esp-idf/files/10249155/failed.connecting.to.hotspot.txt)
2. [success connected to windows mobile hotspot by comment out the lines in esp-camera source](https://github.com/espressif/esp-idf/files/10249159/success.by.comment.out.txt)
3. [successful connecting my home wifi regardless the lines are commented out or not](https://github.com/espressif/esp-idf/files/10249165/home.wifi.-.success.txt)

I only see one difference between my home wifi and Windows Mobile Hotspot one: the home wifi has `40U` subchannel but the Hotspot has `BW20`. 

Please let me know if any more information is needed.

wuyuanyi135, 2022-12-17:
I further bisected the issue and found that it was the reset that caused the issue.
```
        REG_SET_BIT(SYSTEM_PERIP_RST_EN1_REG, SYSTEM_LCD_CAM_RST);
        REG_CLR_BIT(SYSTEM_PERIP_RST_EN1_REG, SYSTEM_LCD_CAM_RST);
```

After commenting out those two lines, WiFi is no longer problematic but the camera is not probed by SCCB.

wuyuanyi135, 2022-12-17:
I tried to execute those four lines but it did not affect WiFi. I found it only affected WiFi if this line is first executed,
https://github.com/espressif/esp32-camera/blob/402b811b835cd348343b567a97fdf984c9d16fb9/target/esp32s3/ll_cam.c#L330 , then the LCD_CAM clock gets enabled and reset will trigger the WiFi Problem.

wuyuanyi135, 2022-12-17:
I have just tried connecting to my Android Hotspot and it seems the connection is fine. Here is the log. 

So this is likely something wrong with the Windows Mobile Hotspot. However, without camera running WiFi works just fine. What else could be different here? Maybe the Windows Mobile Hotspot has lower tolerance of response time-out and the ESP runs slower when camera is used so WiFi did not catch up in time?

```
D (8751) wifi:profile match: ss_state=0x7
D (8751) wifi:scan end: arg=0, status=0, ss_state=0x7
D (8751) wifi:find first mathched ssid, scan done
D (8751) wifi:filter: set rx policy=4
D (8751) wifi:first chan=1
D (8751) wifi:handoff_cb: status=0
D (8751) wifi:ap found, mac=[REMOVED]
D (8751) wifi:new_bss=0x3c130fc0, cur_bss=0, new_chan=<6,0>, cur_chan=1
D (8751) wifi:filter: set rx policy=5
I (8751) wifi:new:<6,0>, old:<1,0>, ap:<255,255>, sta:<6,0>, prof:1
D (8751) wifi:connect_op: status=0, auth=5, cipher=3
I (9291) cam:2500: Got FB: 320x240
D (9491) wifi:auth mode is not none
D (9491) wifi:connect_bss: auth=1, reconnect=0
I (9491) wifi:state: init -> auth (b0)
D (9491) wifi:start 1s AUTH timer
D (9491) wifi:clear scan ap list
D (9561) wifi:recv auth: seq=2, status=0
I (9561) wifi:state: auth -> assoc (0)
D (9561) wifi:restart connect 1s timer for assoc
D (9571) wifi:recv assoc: type=0x10
D (9581) wifi:filter: set rx policy=6
I (9581) wifi:state: assoc -> run (10)
D (9581) wifi:start 10s connect timer for 4 way handshake
I (9661) wifi:connected with W, aid = 1, channel 6, BW20, bssid = [REMOVED]
I (9661) wifi:security: WPA2-PSK, phy: bgn, rssi: -29
D (9661) wifi:remove all except [REMOVED] from rc list
D (9661) wifi:clear blacklist
D (9671) wifi:filter: set rx policy=7
I (9671) wifi:pm start, type: 1

I (9671) wifi:set rx beacon pti, rx_bcn_pti: 0, bcn_timeout: 0, mt_pti: 25000, mt_time: 10000
D (9671) wifi:Send sta connected event
D (9671) wifi:connect status 1 -> 5
D (9671) wifi:obss scan is disabled
D (9671) wifi:start obss scan: obss scan is stopped
I (9671) wifi:20: WiFi Connected to W
I (9681) wifi:BcnInt:102400, DTIM:2
D (9681) wifi:set max rate: from <rate=130, phy=3, sig=0> to <rate=144, phy=3 sig=0>
D (9681) wifi:sig_b=0, sig_g=0, sig_n=0, max_b=22, max_g=108, max_n=144
D (9681) wifi:update trc
I (9751) wifi:<ba-add>idx:0 (ifx:0, [REMOVED]), tid:0, ssn:0, winSize:64
```

me-no-dev, 2022-12-19:
Disconnect reason 2 is `AUTH_EXPIRE` which can be due to timeout: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/wifi.html#wi-fi-reason-code

I suggest you try running the camera code on Core 1 and see if that changes the situation :)

wuyuanyi135, 2022-12-19:
@me-no-dev thanks for the suggestion. I actually changed WiFi task core affinity to core 1 once and it did not help. I will try to change the camera task later. But I guess the timeout limit posed by the windows hotspot is more tight than the normal router, causing the expiry failure only under this combination.

nishanth-radja, 2023-01-05:
@wuyuanyi135 can we have a sniffer capture during association with and without the camera code while connecting to the windows hotspot.We can compare the difference.

wuyuanyi135, 2023-01-06:
@nishanth-radja Hello. Could you please provide the procedures to perform the capture?

AxelLin, 2023-04-21:
> @nishanth-radja Hello. Could you please provide the procedures to perform the capture?

https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/wireshark-user-guide.html

AxelLin, 2023-10-19:
@wuyuanyi135 Any update regarding comment https://github.com/espressif/esp-idf/issues/10390#issuecomment-1371919397 ?

AxelLin, 2024-01-26:
@wuyuanyi135 
Do you still hit the issue with recent esp-idf and esp32-camera updates?

AxelLin, 2024-05-08:
@wuyuanyi135 
Do you still hit this issue with recent esp-idf version and latest esp32-camera-v2.0.8 ?

wuyuanyi135, 2024-05-19:
@AxelLin Thank you for continuing attention to this issue and sorry for the late response. Due to job reallocation I did not have the same setup to test this issue. I am pretty sure this issue could be tested using the widely available ESP-CAM module (despite the different mcu, esp32s3 vs esp32). If anyone could help validate this issue would be much appreciated.