espressif/esp-idf #10392: No bootable app partitions in the partition table after once debug

Yecgaa1, 2022-12-17:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I use ESP-WROVER-KIT-VE to write programs on esp-idf of v4.4.3-316-ge86181704a
Before starting debugging the program can burn normally (using serial port) and print normal run log. After starting debugging, I can stop at the first breakpoint, but when I step in it jumps to unhandled double exception and keeps looping, then the monitor shows the following and does not move (even if I close debug)
```shell
ets Jul 29 2019 12:21:46

rst:0x3 (SW_RESET),boot:0x1e (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:7020
ho 0 tail 12 room 4
load:0x40078000,len:15600
load:0x40080400,len:4492
0x40080400: _init at ??:?

entry 0x400806b4
I (29) boot: ESP-IDF v4.4.3-316-ge86181704a 2nd stage bootloader
I (29) boot: compile time 12:08:04
I (29) boot: chip revision: 3
I (33) boot_comm: chip revision: 3, min. bootloader chip revision: 0
I (40) qio_mode: Enabling default flash chip QIO
I (45) boot.esp32: SPI Speed      : 80MHz
I (50) boot.esp32: SPI Mode       : QIO
I (54) boot.esp32: SPI Flash Size : 4MB
I (59) boot: Enabling RNG early entropy source...
I (64) boot: Partition Table:
I (68) boot: ## Label            Usage          Type ST Offset   Length
I (75) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (83) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (90) boot:  2 factory          factory app      00 00 00010000 00100000
I (98) boot: End of partition table
I (102) boot_comm: chip revision: 3, min. application chip revision: 0
I (109) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=12c04h ( 76804) map
I (139) esp_image: segment 1: paddr=00022c2c vaddr=3ffb0000 size=0362ch ( 13868) load
I (143) esp_image: segment 2: paddr=00026260 vaddr=40080000 size=09db8h ( 40376) load
I (158) esp_image: segment 3: paddr=00030020 vaddr=400d0020 size=74670h (476784) map
I (289) esp_image: segment 4: paddr=000a4698 vaddr=40089db8 size=0d454h ( 54356) load
I (318) boot: Loaded app from partition at offset 0x10000
I (318) boot: Disabling RNG early entropy source...
I (330) cpu_start: Pro cpu up.
I (330) cpu_start: Starting app cpu, entry point is 0x40081454
0x40081454: call_start_cpu1 at C:/Users/22380/esp/esp-idf/components/esp_system/port/cpu_start.c:148

I (316) cpu_start: App cpu up.
I (344) cpu_start: Pro cpu start user code
I (344) cpu_start: cpu freq: 240000000
I (344) cpu_start: Application information:
I (349) cpu_start: Project name:     wireless_esp_dap
I (355) cpu_start: App version:      v0.2.1-1-geb14f79-dirty
I (361) cpu_start: Compile time:     Dec 17 2022 12:09:18
I (367) cpu_start: ELF file SHA256:  3884699297ccab2a...
I (373) cpu_start: ESP-IDF:          v4.4.3-316-ge86181704a
I (379) heap_init: Initializing. RAM available for dynamic allocation:
I (387) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (393) heap_init: At 3FFB76C0 len 00028940 (162 KiB): DRAM
I (399) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (405) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (412) heap_init: At 4009720C len 00008DF4 (35 KiB): IRAM
I (418) spi_flash: detected chip: generic
I (423) spi_flash: flash io: qio
I (427) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
```
[image]
[image]

When I restarted the monitor, I found that it could no longer run the program normally and kept outputting the following message
```shell
ets Jul 29 2019 12:21:46

rst:0x3 (SW_RESET),boot:0x1e (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:7020
ho 0 tail 12 room 4
load:0x40078000,len:15600
load:0x40080400,len:4492
0x40080400: _init at ??:?

entry 0x400806b4
I (56) boot: ESP-IDF v4.4.3-316-ge86181704a 2nd stage bootloader
I (56) boot: compile time 12:08:04
I (56) boot: chip revision: 3
I (61) boot_comm: chip revision: 3, min. bootloader chip revision: 0
I (68) qio_mode: Enabling default flash chip QIO
I (73) boot.esp32: SPI Speed      : 80MHz
I (78) boot.esp32: SPI Mode       : QIO
I (83) boot.esp32: SPI Flash Size : 4MB
I (87) boot: Enabling RNG early entropy source...
I (93) boot: Partition Table:
I (96) boot: ## Label            Usage          Type ST Offset   Length
I (103) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (111) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (119) boot:  2 factory          factory app      00 00 00010000 00100000
I (126) boot: End of partition table
I (130) boot_comm: chip revision: 3, min. application chip revision: 0
I (138) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=12c04h ( 76804) map
I (167) esp_image: segment 1: paddr=00022c2c vaddr=3ffb0000 size=0362ch ( 13868) load
I (172) esp_image: segment 2: paddr=00026260 vaddr=40080000 size=09db8h ( 40376) load
I (186) esp_image: segment 3: paddr=00030020 vaddr=400d0020 size=74670h (476784) map
I (318) esp_image: segment 4: paddr=000a4698 vaddr=40089db8 size=0d454h ( 54356) load
E (336) esp_image: Checksum failed. Calculated 0xef read 0x1a
E (336) boot: Factory app partition is not bootable
E (337) boot: No bootable app partitions in the partition table
ets Jul 29 2019 12:21:46

rst:0x3 (SW_RESET),boot:0x1e (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:7020
ho 0 tail 12 room 4
load:0x40078000,len:15600
load:0x40080400,len:4492
0x40080400: _init at ??:?

entry 0x400806b4
```
I can burn and debug normally in a minimal project Hello-World
Based on the above monitor I suspect that it has something to do with my partition table, but I don't know how to fix this problem. 
This is my vscode launch.json, this seems to be the default
```json
{
  "version": "0.2.0",
  "configurations": [
    {
      "type": "espidf",
      "name": "Launch",
      "request": "launch"
    }
  ]
}
```
I would be grateful if you could help me!

igrr, 2022-12-17:
> After starting debugging, I can stop at the first breakpoint, but when I step in it jumps to unhandled double exception and keeps looping, then the monitor shows the following and does not move (even if I close debug)

Could you please share the sample program for us to reproduce this behavior?

> E (336) esp_image: Checksum failed. Calculated 0xef read 0x1a
> E (336) boot: Factory app partition is not bootable
> E (337) boot: No bootable app partitions in the partition table

This issue has happened because the application binary in flash was modified by inserting the software breakpoint. When the debug session terminates normally, OpenOCD removes all software breakpoints, reverting the program binary to its original state. I suppose the debug session was not terminated normally in this case. If you flash the application again, it should start working.

Yecgaa1, 2022-12-17:
> This issue has happened because the application binary in flash was modified by inserting the software breakpoint. When the debug session terminates normally, OpenOCD removes all software [...]

Oh, I've learned now

> Could you please share the sample program for us to reproduce this behavior?

Yes, but since I really don't know what the problem is, please allow me to link the repository directly to you, I'm sorry if this causes you more work
[](https://github.com/xutongxin1/wireless-esp8266-dap/tree/xiaohui)