espressif/esp-idf #10394: Sniff ESPNOW traffic

AmirHmZz, 2022-12-18:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I have two questions : 

1. How does random bytes prevent relay attacks?
2. I know ESPNOW traffic can be inspected by tools like wireshark If I don't use action frame encryption. But, can an attacker generate and send ESPNow messages? Is there anything that verifies messages are surely sent from an esp device and they are not sent by an attacker with these [tools](https://wiki.wireshark.org/Tools#traffic-generators)?

AmirHmZz, 2023-02-11:
Any updates on this question?

zhangyanjiaoesp, 2023-04-10:
@AmirHmZz 
1. Currently, the random bytes  can only check the retransmit packets with the `Retry` bit set to 1 in the `Frame Control` field.
2. If the ESPNOW data is unencrypted, an attacker can generate and send ESPNOW, and there isn't a good way to tell if the packet was sent from the ESP device or from the attacker.

closed 2024-02-29 by Alvin1Zhang