espressif/esp-idf #10396: gpio_uninstall_isr_service appears to leak memory

kbsriram, 2022-12-18:
running the standalone heap tracer around this pair of calls:
```
gpio_install_isr_service(0);
gpio_uninstall_isr_service();
```
results in a reported missing free as follows:
```
1 allocations trace (10 entry buffer)
12 bytes (@ 0x3fc90fb8) allocated CPU 0 ccount 0x027bc30c caller 
12 bytes 'leaked' in trace (1 allocations)
total allocations 3 total frees 2
```
Additional information:
```
Target: esp32c3
Board: DFRobot Beetle ESP32 - C3 (https://www.dfrobot.com/product-2566.html)
idf.py --version: ESP-IDF v5.1-dev-1768-g6bb28c4cdc-dirty
Host os: Linux kbs-personal 4.15.0-200-generic #211-Ubuntu SMP
```

Sample code with configuration attached:
[memory_leak.zip](https://github.com/espressif/esp-idf/files/10254022/memory_leak.zip)

kayzen9, 2022-12-19:
it's by design to speed up further int alloc... (@kbsriram)

https://github.com/espressif/esp-idf/blob/edd815af2e7d541b25ff3a74c59a7fe0612df42d/components/esp_hw_support/intr_alloc.c#L752-L754

kbsriram, 2022-12-19:
Ah, interesting! Thanks for the pointer @kayzen9 - I'll close this out as WAI.

closed 2022-12-19 by kbsriram

AxelLin, 2022-12-23:
> it's by design to speed up further int alloc... (@kbsriram)

Do you expect the intr_alloc/intr_free to be called in critical path?
If not, why you need the hack for speedup?

kbsriram, 2022-12-23:
@songruo - gotcha - and thank you for the follow-up.