espressif/esp-idf #10398: Bug MQTT over TCP

DavidVentura, 2022-12-18:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1-dev-2356-gedd815af2e and v4.4.2

### Operating System used.

Linux

### How did you build your project?

Command line with CMake

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP-WROOM-32

### Power Supply used.

USB

### What is the expected behavior?

MQTT events are received and accepted by the broker

### What is the actual behavior?

ESP gets a FINACK immediately after sending an MQTT CONNECT

### Steps to reproduce.

Based on the combined examples from `wifi/get-started` and `protocols/mqtt/tcp`

main.c
```c
/*
 * SPDX-FileCopyrightText: 2010-2022 Espressif Systems (Shanghai) CO LTD
 *
 * SPDX-License-Identifier: CC0-1.0
 */

#include <stdio.h>
#include <inttypes.h>
#include "sdkconfig.h"
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "driver/gpio.h"
#include "esp_chip_info.h"
#include "esp_flash.h"
#include "wifi.c"
#include "mqtt.c"

static uint8_t s_led_state = 0;
static const char *TAG = "example";
#define BLINK_GPIO 2

static void blink_led(void)
{
    /* Set the GPIO level according to the state (LOW or HIGH)*/
    gpio_set_level(BLINK_GPIO, s_led_state);
}
[... 369 lines cut ...]
I (2229) MQTT_EXAMPLE: MQTT_EVENT_PUBLISHED, msg_id=24652
I (2229) MQTT_EXAMPLE: MQTT_EVENT_SUBSCRIBED, msg_id=64355
I (2239) MQTT_EXAMPLE: sent publish successful, msg_id=0
I (2249) MQTT_EXAMPLE: MQTT_EVENT_SUBSCRIBED, msg_id=62540
I (2249) MQTT_EXAMPLE: sent publish successful, msg_id=0
I (2259) MQTT_EXAMPLE: MQTT_EVENT_UNSUBSCRIBED, msg_id=12952
I (2269) MQTT_EXAMPLE: MQTT_EVENT_DATA
TOPIC=/topic/qos0
DATA=data
I (2439) MQTT_EXAMPLE: MQTT_EVENT_DATA
TOPIC=/topic/qos0
DATA=data
Restarting in 99 seconds...
Restarting in 98 seconds...
Restarting in 97 seconds...
Restarting in 96 seconds...
Restarting in 95 seconds...
```

### More Information.

I see the exact same issue on both v4.4.2 and current HEAD.
The MQTT daemon is `mosquitto`. There is no TLS, just plain TCP. The issue occurs every time.
I have captured the network packets 4 times:
[captures.zip](https://github.com/espressif/esp-idf/files/10254726/captures.zip)

ESP-YJM, 2022-12-19:
@DavidVentura The FIN sent by broker is different TCP seesion with the connect one. You can identify the port, it is different. The FIN is the previous one. And from your MQTT captured packet, it is ok and no problems.

DavidVentura, 2022-12-20:
:facepalm: after another debugging session I realized I was publishing to `/topic/qos` and subscribing to `topic/#`, sorry for the noise

closed 2022-12-20 by DavidVentura