espressif/esp-idf #10399: [ESP32S3] btdm_controller_init failing after deinit

KonssnoK, 2022-12-19:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4.3-319-g0fb32e11a58

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

PowerShell

### Development Kit.

ESP32S3-WROOM N8R2

### Power Supply used.

USB

### What is the expected behavior?

Switch on and off bluetooth multiple times on request to provision devices

### What is the actual behavior?

Device crash on ble init the second time provisioning is started.

This was not happening before, we are looking for the commit that changed the behavior.

From the documentation it seems that we are performing some forbidden handling

...
/**
 * @brief       Initialize BT controller to allocate task and other resource.
 *              This function should be called only once, before any other BT functions are called.
 * @param  cfg: Initial configuration of BT controller. Different from previous version, there's a mode and some
 *              connection configuration in "cfg" to configure controller work mode and allocate the resource which is needed.
 * @return      ESP_OK - success, other - failed
 */
esp_err_t esp_bt_controller_init(esp_bt_controller_config_t *cfg);

/**
 * @brief  De-initialize BT controller to free resource and delete task.
 *         You should stop advertising and scanning, as well as
 *         disconnect all existing connections before de-initializing BT controller.
 *
 * This function should be called only once, after any other BT functions are called.
 * This function is not whole completed, esp_bt_controller_init cannot called after this function.
 * @return  ESP_OK - success, other - failed
 */
esp_err_t esp_bt_controller_deinit(void);
...

but then how you are supposed to handle a device that needs to update it's own configuration via bluetooth but normally has only wifi?

### Steps to reproduce.

1. Start provisioning of one device
2. End provisioning -> BLE will be switched off
3. Restart provisioning without rebooting the device
...

### Debug Logs.

```plain
I (16:44:23.963) prov_BLE: Unhandled event 0
I (16:44:23.968) prov_BLE: Starting BLE provisioning
I (92942) wifi:mode : sta (7c:df:a1:e0:8b:7c)
btdm: bss start 0x3fcef180, len 36
btdm: data start 0x3fcef174, data start rom 0x40057350, len 12
MAGIC fadebead VERSION 00010006
I (16:44:23.989) BT_INIT: BT controller compile version [02bfb7f]
E (16:44:23.996) simple_ble: simple_ble_start enable controller failed 257
E (16:44:24.004) protocomm_ble: simple_ble_start failed w/ error code 0x101
E (16:44:24.011) wifi_prov_scheme_ble: Failed to start protocomm BLE service
E (16:44:24.019) wifi_prov_mgr: Failed to start service
ESP_ERROR_CHECK failed: esp_err_t 0xffffffff (ESP_FAIL) at 0x40381e80
0x40381e80: _esp_error_check_failed at C:/src/esp-idf/components/esp_system/esp_err.c:42

file: "./components/provisioning/bluetooth.c" line 480
func: provisioning_ble_start
expression: wifi_prov_mgr_start_provisioning(security, pop, service_name, service_key)
```

KonssnoK, 2022-12-19:
we currently applied the following patch to the provisioning app in order to continue to use it

[image]

xiewenxiang, 2022-12-30:
Did you only turn on and off the controller?
I have test init and deinit the bluetooth protocol stack, but no problems have been found so far. Could you please provide your sample code?

KonssnoK, 2022-12-30:
@xiewenxiang 
The sample code is using the Bluetooth stack of the provisioning app...

https://github.com/espressif/esp-idf/tree/release/v4.4/examples/provisioning/wifi_prov_mgr

Not sure you can reproduce tho, since the provisioning app doesn't allow to go back into provisioning mode 🤔.

xiewenxiang, 2023-02-01:
@KonssnoK 
This is the original demo of IDF, and I have add init & deinit code, tested with this demo, but I couldn't reproduce this issue.

KonssnoK, 2023-02-01:
i'm sorry @xiewenxiang we are following on other issues, we currently don't have time to test this one :(

But the code there is not enough i think, you need to changes in order to enable multiple provisioning without restarting the device.

KonssnoK, 2023-04-20:
after updating to latest v4.4 7eba5f80027e1648775b46f889cb4d9519afc965
we tried removing the patch and we are not having the failure anymore

closed 2023-04-20 by KonssnoK