espressif/esp-idf #10403: Assert failure in esp_bignum.c

mjeanrichard, 2022-12-19:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.0

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

PowerShell

### Development Kit.

ESP32-WROOM-32

### Power Supply used.

Battery

### What is the expected behavior?

No assert failure.

### What is the actual behavior?

My ESP connects to a MQTT server aends a few messages, goes to deep sleep for a while and then starts the process again. Once in a while (maybe one a day?) I get a panic reset. Since I store the Core Dumps on a SD card I could check the Stack trace and identify the assert here (https://github.com/espressif/esp-idf/blob/master/components/mbedtls/port/esp_bignum.c#L651) was causing the panic.

### Steps to reproduce.

Connect to an MQTT Broker over SSL and wait?

### Debug Logs.

```plain
#0  0x40081b71 in panic_abort (details=0x3ffca220 "assert failed: 0x40125699") at /opt/esp/idf/components/esp_system/panic.c:412
#1  0x4008ad34 in esp_system_abort (details=0x3ffca220 "assert failed: 0x40125699") at /opt/esp/idf/components/esp_system/esp_system.c:135
#2  0x40091a51 in __assert_func (file=0x0, line=0, func=0x0, expr=0x0) at /opt/esp/idf/components/newlib/assert.c:40
#3  0x4012569c in mpi_mult_mpi_failover_mod_mult (z_words=<optimized out>, Y=<optimized out>, X=<optimized out>, Z=<optimized out>) at /opt/esp/idf/components/mbedtls/port/esp_bignum.c:651
#4  mbedtls_mpi_mul_mpi (Z=<optimized out>, X=<optimized out>, Y=<optimized out>) at /opt/esp/idf/components/mbedtls/port/esp_bignum.c:522
#5  0x4012577c in mbedtls_mpi_mul_int (X=0x3ffca2d8, A=0x3ffca2f0, b=934139769) at /opt/esp/idf/components/mbedtls/port/esp_bignum.c:557
#6  0x4011a1f7 in mbedtls_mpi_div_mpi (Q=0x0, R=0x3ffd2ff8, A=0x3ffca390, B=0x3ffd2f98) at /opt/esp/idf/components/mbedtls/mbedtls/library/bignum.c:1690
#7  0x4011a3e1 in mbedtls_mpi_mod_mpi (R=0x3ffd2ff8, A=0x3ffca390, B=0x3ffd2f98) at /opt/esp/idf/components/mbedtls/mbedtls/library/bignum.c:1760
#8  0x40125158 in calculate_rinv (Rinv=0x3ffd2ff8, M=0x3ffd2f98, num_words=64) at /opt/esp/idf/components/mbedtls/port/esp_bignum.c:200
#9  0x40125389 in esp_mpi_exp_mod (_Rinv=0x3ffd2ff8, M=0x3ffd2f98, Y=0x3ffd2fa4, X=0x3ffca450, Z=0x3ffca450) at /opt/esp/idf/components/mbedtls/port/esp_bignum.c:388
#10 mbedtls_mpi_exp_mod (X=0x3ffca450, A=0x3ffca450, E=0x3ffd2fa4, N=0x3ffd2f98, _RR=0x3ffd2ff8) at /opt/esp/idf/components/mbedtls/port/esp_bignum.c:460
#11 0x401234d6 in mbedtls_rsa_public (ctx=0x3ffd2f90, input=<optimized out>, output=0x3ffd4c60 <error: Cannot access memory at address 0x3ffd4c60>) at /opt/esp/idf/components/mbedtls/mbedtls/library/rsa.c:763
#12 0x40124383 in mbedtls_rsa_rsassa_pkcs1_v15_verify (ctx=0x3ffd2f90, md_alg=MBEDTLS_MD_SHA256, hashlen=<optimized out>, hash=0x3ffca540 "u¯ÏÊG¯{tv&d~3q^¹b5÷þ~¯ƒN<)¼\024\231«9\377O\025ý?ü", sig=0x3ffcd469 <error: Cannot access memory at address 0x3ffcd469>)
    at /opt/esp/idf/components/mbedtls/mbedtls/library/rsa.c:2154
#13 0x401243ed in mbedtls_rsa_pkcs1_verify (ctx=0x3ffd2f90, md_alg=MBEDTLS_MD_SHA256, hashlen=32, hash=0x3ffca540 "u¯ÏÊG¯{tv&d~3q^¹b5÷þ~¯ƒN<)¼\024\231«9\377O\025ý?ü", sig=0x3ffcd469 <error: Cannot access memory at address 0x3ffcd469>)
    at /opt/esp/idf/components/mbedtls/mbedtls/library/rsa.c:2206
#14 0x40121a62 in rsa_verify_wrap (ctx=0x3ffd2f90, md_alg=<optimized out>, hash=0x3ffca540 "u¯ÏÊG¯{tv&d~3q^¹b5÷þ~¯ƒN<)¼\024\231«9\377O\025ý?ü", hash_len=32, sig=0x3ffcd469 <error: Cannot access memory at address 0x3ffcd469>, sig_len=256)
    at /opt/esp/idf/components/mbedtls/mbedtls/library/pk_wrap.c:247
#15 0x40121744 in mbedtls_pk_verify_restartable (ctx=0x3ffd29c8, md_alg=MBEDTLS_MD_SHA256, hash=0x3ffca540 "u¯ÏÊG¯{tv&d~3q^¹b5÷þ~¯ƒN<)¼\024\231«9\377O\025ý?ü", hash_len=32, sig=0x3ffcd469 <error: Cannot access memory at address 0x3ffcd469>, sig_len=256,
    rs_ctx=rs_ctx@entry=0x0) at /opt/esp/idf/components/mbedtls/mbedtls/library/pk.c:441
#16 0x40117143 in ssl_parse_server_key_exchange (ssl=0x3ffcab58) at /opt/esp/idf/components/mbedtls/mbedtls/library/ssl_tls12_client.c:2512
#17 mbedtls_ssl_handshake_client_step (ssl=0x3ffcab58) at /opt/esp/idf/components/mbedtls/mbedtls/library/ssl_tls12_client.c:3663
#18 0x40115282 in mbedtls_ssl_handshake_step (ssl=0x3ffcab58) at /opt/esp/idf/components/mbedtls/mbedtls/library/ssl_tls.c:3027
#19 mbedtls_ssl_handshake_step (ssl=0x3ffcab58) at /opt/esp/idf/components/mbedtls/mbedtls/library/ssl_tls.c:2984
#20 0x401152db in mbedtls_ssl_handshake (ssl=<optimized out>) at /opt/esp/idf/components/mbedtls/mbedtls/library/ssl_tls.c:3092
#21 mbedtls_ssl_handshake (ssl=0x3ffcab58) at /opt/esp/idf/components/mbedtls/mbedtls/library/ssl_tls.c:3068
#22 0x40111744 in esp_mbedtls_handshake (tls=0x3ffcab58, cfg=0x3ffc8f8c) at /opt/esp/idf/components/esp-tls/esp_tls_mbedtls.c:192
#23 0x40111408 in esp_tls_handshake (cfg=0x3ffc8f8c, tls=0x3ffcab58) at /opt/esp/idf/components/esp-tls/esp_tls.c:83
#24 esp_tls_low_level_conn (hostname=<optimized out>, hostlen=19, port=<optimized out>, cfg=0x3ffc8f8c, tls=0x3ffcab58) at /opt/esp/idf/components/esp-tls/esp_tls.c:437
#25 0x401114fa in esp_tls_conn_new_sync (hostname=0x3ffc8398 <error: Cannot access memory at address 0x3ffc8398>, hostlen=19, port=8883, cfg=0x3ffc8f8c, tls=0x3ffcab58) at /opt/esp/idf/components/esp-tls/esp_tls.c:464
#26 0x40112cee in ssl_connect (t=0x3ffc8508, host=0x3ffc8398 <error: Cannot access memory at address 0x3ffc8398>, port=8883, timeout_ms=<optimized out>) at /opt/esp/idf/components/tcp_transport/transport_ssl.c:109
#27 0x40176bd0 in esp_transport_connect (t=0x3ffc8508, host=0x3ffc8398 <error: Cannot access memory at address 0x3ffc8398>, port=8883, timeout_ms=10000) at /opt/esp/idf/components/tcp_transport/transport.c:123
#28 0x400e8885 in esp_mqtt_task (pv=0x3ffc8194) at /opt/esp/idf/components/mqtt/esp-mqtt/mqtt_client.c:1489
#29 0x4008dbe0 in vPortTaskWrapper (pxCode=0x400e866c <esp_mqtt_task>, pvParameters=0x3ffc8194) at /opt/esp/idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:151
```
```

laukik-hase, 2022-12-20:
Hello, @mjeanrichard!

Thank you for the issue report. Could you please attach the following -
- The `sdkconfig` file for your example
- Minimal example that produces the issue
- Disassembly of the function `mpi_to_mem_block`; this can be generated with the instructions given below.
  - Build your example and fire the command `xtensa-esp32-elf-gdb <path_to_elf_file>.elf`
  - On entering the `gdb` console, enter the sequence `set logging on` → `disassemble /m mpi_to_mem_block`
  - Quit the `gdb` console after viewing the entire disassembly. Please post the resulting text file here.

mjeanrichard, 2022-12-20:
Thanks for your response. 
The Sdkconfig : [sdkconfig.txt](https://github.com/espressif/esp-idf/files/10266397/sdkconfig.txt)

I cannot provide a minimal example to reporduce this. The Issue seems to happen about once a day, but since I do the deepsleep - connect cycle almost 300 times a day I cannot readily reproduce it.

Disassembly: [gdb.txt](https://github.com/espressif/esp-idf/files/10266405/gdb.txt)

As a workaround I disabled CONFIG_MBEDTLS_HARDWARE_MPI and so far it seems to work.

mjeanrichard, 2022-12-20:
v5.0

mjeanrichard, 2022-12-20:
Not sure if this helps, but I have a number of these devices around my house and it looks like this issue happens more often if the wifi singal is weaker (ie. for the devices further away from the access point). Could just be because some opertions happens more frequent if the connection is unstable and therefore is more likely to trigger this isseu...

mjeanrichard, 2022-12-26:
I managed to create a repro:
[repro.c.txt](https://github.com/espressif/esp-idf/files/10304562/repro.c.txt)
[sdkconfig.txt](https://github.com/espressif/esp-idf/files/10304563/sdkconfig.txt)

Basically it just a copy and pasted from your samples, combining WIFI and MQTT. It connects to MQTT and then goes to deepsleep for 1 sec. After the deep sleep it connects again. 

After running this for a few hours I get an assert failure:
``assert failed: mpi_mult_mpi_failover_mod_mult esp_bignum.c:651 (mpi_words(Z) == z_words)``

This only happens with the attached sdkconfig. Not, for example, when I disable optimization.

AxelLin, 2022-12-28:
@mahavirj,
Any comment about https://github.com/espressif/esp-idf/issues/10403#issuecomment-1365384526 ?
This assertion seems to be the same as https://github.com/espressif/esp-idf/issues/8710#issuecomment-1249178698 , but v5.0  already includes 84b0254 .

mahavirj, 2022-12-28:
@AxelLin We are looking into this issue. It may be an entirely different issue than the previous one discussed in #8710, difficult to say at this moment.

@mjeanrichard Thank you for sharing the reproducible code, we will try to set it up by early next week (given holiday season). Interim could you please try following:
 - Please enable code from https://github.com/espressif/esp-idf/blob/490216a2ace6dc3e1b9a3f50d265a80481b32f6d/components/mbedtls/port/esp32/bignum.c#L107 and see if that changes the assert location
 - Please confirm if you disable the deep sleep related workflow, you can still reproduce the issue
 - Please confirm if you can reproduce this issue at `COMPILER_OPTIMIZATION_SIZE`

mjeanrichard, 2022-12-28:
I am on holiday as well until the 3rd of january. But I will try as soon as I am back.
@mahavirj : What do you mean exactly by "disable the deep sleep related workflow"? Should I just disconnect/reconnect wifi and mqtt after a successful connect? Or just leave the connection running and send some MQTT messages?

I will try with COMPILER_OPTIMIZATION_SIZE. I know that with debug settings (ie. no optimization) it is not happening.

mahavirj, 2022-12-29:
> Should I just disconnect/reconnect wifi and mqtt after a successful connect?

Yes, I was suggesting this setup.

mjeanrichard, 2023-01-03:
New Findings:
- It does NOT seem to happen with "COMPILER_OPTIMIZATION_SIZE" (I tried for 12 hours and no panic)
- When I uncomment the lines in [esp32/bignum.c](https://github.com/espressif/esp-idf/blob/490216a2ace6dc3e1b9a3f50d265a80481b32f6d/components/mbedtls/port/esp32/bignum.c#L107) and remove the guard it will trigger that Assert (`bignum.c:106` is the for loop mention by @mahavirj):
```
0x40081f82: panic_abort at C:/Users/accou/esp5/esp-idf/components/esp_system/panic.c:412
0x40089749: esp_system_abort at C:/Users/accou/esp5/esp-idf/components/esp_system/esp_system.c:135
0x4008ff1a: __assert_func at C:/Users/accou/esp5/esp-idf/components/newlib/assert.c:78
0x400f8a2f: mpi_to_mem_block at C:/Users/accou/esp5/esp-idf/components/mbedtls/port/esp32/bignum.c:106
 (inlined by) mpi_to_mem_block at C:/Users/accou/esp5/esp-idf/components/mbedtls/port/esp32/bignum.c:75
0x400f8d09: esp_mpi_mult_mpi_failover_mod_mult_hw_op at C:/Users/accou/esp5/esp-idf/components/mbedtls/port/esp32/bignum.c:295
0x400f8892: mpi_mult_mpi_failover_mod_mult at C:/Users/accou/esp5/esp-idf/components/mbedtls/port/esp_bignum.c:640
 (inlined by) mbedtls_mpi_mul_mpi at C:/Users/accou/esp5/esp-idf/components/mbedtls/port/esp_bignum.c:522
0x400f89c9: mbedtls_mpi_mul_int at C:/Users/accou/esp5/esp-idf/components/mbedtls/port/esp_bignum.c:557
0x400ed440: mbedtls_mpi_div_mpi at C:/Users/accou/esp5/esp-idf/components/mbedtls/mbedtls/library/bignum.c:1690
0x400ed62a: mbedtls_mpi_mod_mpi at C:/Users/accou/esp5/esp-idf/components/mbedtls/mbedtls/library/bignum.c:1760
0x400f83a1: calculate_rinv at C:/Users/accou/esp5/esp-idf/components/mbedtls/port/esp_bignum.c:200 (discriminator 2)
0x400f85d2: esp_mpi_exp_mod at C:/Users/accou/esp5/esp-idf/components/mbedtls/port/esp_bignum.c:388
 (inlined by) mbedtls_mpi_exp_mod at C:/Users/accou/esp5/esp-idf/components/mbedtls/port/esp_bignum.c:460
0x400f671f: mbedtls_rsa_public at C:/Users/accou/esp5/esp-idf/components/mbedtls/mbedtls/library/rsa.c:763
0x400f75cc: mbedtls_rsa_rsassa_pkcs1_v15_verify at C:/Users/accou/esp5/esp-idf/components/mbedtls/mbedtls/library/rsa.c:2154
0x400f7636: mbedtls_rsa_pkcs1_verify at C:/Users/accou/esp5/esp-idf/components/mbedtls/mbedtls/library/rsa.c:2206
0x400f4cab: rsa_verify_wrap at C:/Users/accou/esp5/esp-idf/components/mbedtls/mbedtls/library/pk_wrap.c:247
0x400f498d: mbedtls_pk_verify_restartable at C:/Users/accou/esp5/esp-idf/components/mbedtls/mbedtls/library/pk.c:441
0x400ea38c: ssl_parse_server_key_exchange at C:/Users/accou/esp5/esp-idf/components/mbedtls/mbedtls/library/ssl_tls12_client.c:2512
 (inlined by) mbedtls_ssl_handshake_client_step at C:/Users/accou/esp5/esp-idf/components/mbedtls/mbedtls/library/ssl_tls12_client.c:3663
0x400e84cb: mbedtls_ssl_handshake_step at C:/Users/accou/esp5/esp-idf/components/mbedtls/mbedtls/library/ssl_tls.c:3027
 (inlined by) mbedtls_ssl_handshake_step at C:/Users/accou/esp5/esp-idf/components/mbedtls/mbedtls/library/ssl_tls.c:2984
0x400e8524: mbedtls_ssl_handshake at C:/Users/accou/esp5/esp-idf/components/mbedtls/mbedtls/library/ssl_tls.c:3092
 (inlined by) mbedtls_ssl_handshake at C:/Users/accou/esp5/esp-idf/components/mbedtls/mbedtls/library/ssl_tls.c:3068
0x400e5cc1: esp_mbedtls_handshake at C:/Users/accou/esp5/esp-idf/components/esp-tls/esp_tls_mbedtls.c:192
0x400e59c5: esp_tls_handshake at C:/Users/accou/esp5/esp-idf/components/esp-tls/esp_tls.c:83 (discriminator 15)
 (inlined by) esp_tls_low_level_conn at C:/Users/accou/esp5/esp-idf/components/esp-tls/esp_tls.c:437 (discriminator 15)
0x400e5aa6: esp_tls_conn_new_sync at C:/Users/accou/esp5/esp-idf/components/esp-tls/esp_tls.c:464
0x400e68ca: ssl_connect at C:/Users/accou/esp5/esp-idf/components/tcp_transport/transport_ssl.c:109
0x40152d61: esp_transport_connect at C:/Users/accou/esp5/esp-idf/components/tcp_transport/transport.c:123
0x400d971a: esp_mqtt_task at C:/Users/accou/esp5/esp-idf/components/mqtt/esp-mqtt/mqtt_client.c:1489
0x4008c889: vPortTaskWrapper at C:/Users/accou/esp5/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:151
```

I will check next if it happens without deep sleep as well (needs some code changes...)

laukik-hase, 2023-01-16:
Hello, @mjeanrichard!

Could you try applying the attached patch (on top of the tag `v5.0`) and check whether it fixes your issue? [esp32_mpi_alt_fix.zip](https://github.com/espressif/esp-idf/files/10424381/esp32_mpi_alt_fix.zip)

**Note:**
- We are still investigating this issue as it has been very difficult to reproduce on our side.
- More information about this patch in the comment https://github.com/espressif/esp-idf/issues/8710#issuecomment-1283534267.

mjeanrichard, 2023-01-21:
@laukik-hase the patch you provided seems to fix the problem (can not be 100% sure, but my device was running successfully for about 12h, while without the patch it only takes a few hours to panic). 

Is my sample code not crashing your devices?

ohitsderrick, 2023-01-30:
> Could you try applying the attached patch (on top of the tag `v5.0`) and check whether it fixes your issue? [esp32_mpi_alt_fix.zip](https://github.com/espressif/esp-idf/files/10424381/esp32_mpi_alt_fix.zip)

@laukik-hase, I've also run into this assertion a few times since moving to ESP-IDF v5.0 and applying your patch. I'll try to narrow down the details and see if there is a way to recreate it more consistently. We are serving up a simple webpage over HTTPS (and hammering it with GET requests) and running an MQTT client over TLS simultaneously to stress the device. It seems to always happen in the HTTP task and not the MQTT task (maybe just coincidence).

laukik-hase, 2023-01-31:
Hello, @ohitsderrick and @mjeanrichard!
Sorry for the late updates.

> Is my sample code not crashing your devices?

I was able to reproduce the issue with your sample code. Applying the above patch attached seems to fix it; no assertion was found even after 48+ hours of running.

> I've also run into this assertion a few times since moving to ESP-IDF v5.0 and applying your patch

- Does the patch not work for you?
- Could be please attach a minimal code sample and your `sdkconfig` file?
- Have you enabled the `CONFIG_COMPILER_OPTIMIZATION_PERF` option?

laukik-hase, 2023-02-06:
> Hello, @ohitsderrick and @mjeanrichard! Sorry for the late updates.
> 
> > Is my sample code not crashing your devices?
> 
> I was able to reproduce the issue with your sample code. Applying the above patch attached seems to fix it; no assertion was found even after 48+ hours of running.
> 
> > I've also run into this assertion a few times since moving to ESP-IDF v5.0 and applying your patch
> 
>     * Does the patch not work for you?
> 
>     * Could you please attach a minimal code sample and your `sdkconfig` file?
> 
>     * Have you enabled the `CONFIG_COMPILER_OPTIMIZATION_PERF` option?

Hello, @ohitsderrick!
Could you please update me on this issue? 

@mjeanrichard, could you please confirm whether the patch attached fixed the issue in your case?

mjeanrichard, 2023-02-06:
@laukik-hase I already did a few comments above. The patched device was running some time without crashing. So, yes, I think it works.

ohitsderrick, 2023-02-07:
@laukik-hase, I haven't had a chance to try reproducing the issue since I first reported it. It did seem to only happen in that specific stress test I described. I will try again today and get a snapshot of the stack trace and sdkconfig if it happens.

laukik-hase, 2023-02-13:
Hello, @ohitsderrick!

Any updates on the issue?

AxelLin, 2023-02-20:
> Hello, @ohitsderrick and @mjeanrichard! Sorry for the late updates.
> 
> > Is my sample code not crashing your devices?
> 
> I was able to reproduce the issue with your sample code. Applying the above patch attached seems to fix it; no assertion was found even after 48+ hours of running.

@laukik-hase 
Do you have the number of performance penalty if using DPORT_WRITE_REG (volatile writes) wrappers to write to the MPI peripheral?

laukik-hase, 2023-02-20:
Hello, @AxelLin!
PFA the performance analysis of all the fixes for this issue.

### Performance analysis of fixes (ESP32)

| Workaround                                  | Time (us)  |     | CPU Cycles (@240MHz) |       |
|---------------------------------------------|------------|-----|----------------------|-------|
|                                             | 1          | 2   | 1                    | 2     |
| None                                        | 103        | 168 | 24817                | 40249 |
| DPORT_WRITE_REG                             | 105        | 170 | 25243                | 40745 |
| Compiler flag (`master` and `release/v5.0`) | 102        | 166 | 24538                | 39750 |

**Note:** _1 & 2 correspond to 2048 and 4096-bit bignum multiplications respectively._

ohitsderrick, 2023-02-21:
> Hello, @ohitsderrick!
> 
> Any updates on the issue?

@laukik-hase, we haven't seen the issue these past few weeks while performing various validation tests. We'll call it resolved on our end for now. At this point, I'm questioning whether or not we saw the assert failure on a legitimate setup with your patch applied.

We updated our sdkconfig a while ago to force single-core mode. I'm assuming that shouldn't have any influence on this assert failiure though.

AxelLin, 2023-02-22:
> We updated our sdkconfig a while ago to force single-core mode. I'm assuming that shouldn't have any influence on this assert failiure though.

@ohitsderrick 
This is off topic.
But I'm wondering why you need to force single-core mode?  Did you hit any issues with dual-core?

ohitsderrick, 2023-02-22:
We were encountering some issues with [Cache disabled but cached memory region accessed](https://docs.espressif.com/projects/esp-idf/en/v5.0/esp32/api-guides/fatal-errors.html?highlight=cache%20disabled#cache-err-msg). Some of the sdkconfig configurations for various components appeared to be causing code to still be executed during calls to the NVS component. It was an quick and easy solution rather than having to pin down every sdkconfig setting and piece of related code that needed to be moved to IRAM. We were also experiencing a "lock" issue deep in the NVS calls where the core making the NVS call would get stuck permanently waiting on the other core to suspend its tasks. We will eventually revert back to dual-core once we have the time and reinvestigate the issue. I could not find any relevant, existing issues posted here.

closed 2023-02-24 by espressif-bot

AxelLin, 2023-03-06:
>  I could not find any relevant, existing issues posted here.

@ohitsderrick 
Please create new issues for the problems you encountered with detail information. Otherwise it wont get fixed.