espressif/esp-idf #10405: ESP32S3 (TG0WDT_SYS_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)

Mbded, 2022-12-19:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hello

I have a device basic on **ESP32-S3-WROOM-1-N8R2** module and programming on **IDF 4.4.3**. To this time all flashing works, but after last flashing device it wasn't bootable, even for simply project as blink, or hello_world

I get below result in monitor. 

```
--- idf_monitor on /dev/ttyUSB0 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:2
load:0x3fce3808,len:0x1664
load:0x3c970000,len:0xbb840
ets_loader.c 78 
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x7 (TG0WDT_SYS_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x400454d5
SPIWP:0xee
mode:DIO, clock div:2
load:0x3fce3808,len:0x1664
load:0x3c970000,len:0xbb840
ets_loader.c 78 
.....
```
I read some issue on this forum, and knew that's cause may be set some bit in flash register from unexpected reasons, but I don't know how can I solve this, and also don't know it is true.

I also check flash register and get result.

```
esptool.py read_flash_status --bytes 3
esptool.py v3.3.2
Found 1 serial ports
Serial port /dev/ttyUSB0
Connecting....
Detecting chip type... ESP32-S3
Chip is ESP32-S3
Features: WiFi, BLE
Crystal is 40MHz
MAC: f4:12:fa:d7:ad:30
Uploading stub...
Running stub...
Stub running...
Status value: **0xe37bfc**
Hard resetting via RTS pin...
```

Icarus113, 2022-12-20:
I use this [v4.4.3](6407ecb3f8d2cc07c4c230e7e64f2046af5c86f7) commit, and a same ESP32-S3-WROOM-1-N8R2 board with same flash vendor. But can't reproduce your issue.

Could you help attach more info such as your sdkconfig, etc. @Mbded

Mbded, 2022-12-20:
> I use this [v4.4.3](6407ecb3f8d2cc07c4c230e7e64f2046af5c86f7) commit, and a same ESP32-S3-WROOM-1-N8R2 board with same flash vendor. But can't reproduce your issue.
> 
> Could you help attach more info such as your sdkconfig, etc. @Mbded

First, I mistake on project which I run. This was **i2c_simple**, nor blink, different is small, but I want to sorry for my mistake. Anyway i2c_simple, blink, hello_world earlier just works.  Finally this sdkconfig basic on i2c_simple. 

Additional information is that this was happend when I try flashing other project basic on this sdkconfig, then  module stop works **after a few** flashing. Then back to previous version (i2c_simple) wasn't able.   

[sdkconfig.txt](https://github.com/espressif/esp-idf/files/10267865/sdkconfig.txt)

I discovered that, when I change configuration on octal flash by configuration then I get below results. Ofcorse this is not god configuration because according to manual this module has quad data flash so error is clear, but I put the this logs because may be has some critical information. 

SERIAL FLASHER CONFIG -> ENABLE OCTAL FLASH -> TRUE

```
Rebooting...
 �ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x9 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40375834
0x40375834: esp_restart_noos at /home/rafal/esp/esp-idf/components/esp_system/port/soc/esp32s3/system_internal.c:151 (discriminator 1)

SPIWP:0xee
mode:DOUT, clock div:2
load:0x3fce3808,len:0x1664
load:0x403c9700,len:0xbb8
load:0x403cc700,len:0x2eb0
entry 0x403c9954
I (33) boot: ESP-IDF v4.4.3-dirty 2nd stage bootloader
I (33) boot: compile time 12:23:00
I (34) boot: chip revision: 0
I (36) boot.esp32s3: Boot SPI Speed : 40MHz
I (40) boot.esp32s3: SPI Mode       : DOUT
I (45) boot.esp32s3: SPI Flash Size : 2MB
I (50) boot: Enabling RNG early entropy source...
I (55) boot: Partition Table:
I (59) boot: ## Label            Usage          Type ST Offset   Length
I (66) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (74) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (81) boot:  2 factory          factory app      00 00 00010000 00100000
I (89) boot: End of partition table
I (93) esp_image: segment 0: paddr=00010020 vaddr=3c020020 size=096d8h ( 38616) map
I (112) esp_image: segment 1: paddr=00019700 vaddr=3fc924f0 size=028b4h ( 10420) load
I (115) esp_image: segment 2: paddr=0001bfbc vaddr=40374000 size=0405ch ( 16476) load
I (123) esp_image: segment 3: paddr=00020020 vaddr=42000020 size=1a210h (107024) map
I (156) esp_image: segment 4: paddr=0003a238 vaddr=4037805c size=0a488h ( 42120) load
I (170) esp_image: segment 5: paddr=000446c8 vaddr=50000000 size=00010h (    16) load
I (176) boot: Loaded app from partition at offset 0x10000
I (176) boot: Disabling RNG early entropy source...
E (190) cpu_start: Octal Flash option selected, but EFUSE not configured!

abort() was called at PC 0x403753c8 on core 0
0x403753c8: call_start_cpu0 at /home/rafal/esp/esp-idf/components/esp_system/port/cpu_start.c:367 (discriminator 3)

Backtrace: 0x40375b96:0x3fceb260 0x4037a491:0x3fceb280 0x4038061e:0x3fceb2a0 0x403753c8:0x3fceb310 0x403cd7cc:0x3fceb340 0x403cdc35:0x3fceb380 0x403c99bd:0x3fceb4b0 0x40045c01:0x3fceb570 |<-CORRUPTED
0x40375b96: panic_abort at /home/rafal/esp/esp-idf/components/esp_system/panic.c:402

0x4037a491: esp_system_abort at /home/rafal/esp/esp-idf/components/esp_system/esp_system.c:128

0x4038061e: abort at /home/rafal/esp/esp-idf/components/newlib/abort.c:46

0x403753c8: call_start_cpu0 at /home/rafal/esp/esp-idf/components/esp_system/port/cpu_start.c:367 (discriminator 3)
```