espressif/esp-idf #10409: IDF V5.0 Offline installer needs to be online?

DspDave, 2022-12-20:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.0

### Operating System used.

Windows

### How did you build your project?

Command line with Make

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

I expected to be able to install the V5.0 IDF in an offline development environment.

### What is the actual behavior?

The installer only completes the installation if connected to the internet.

### Steps to reproduce.

My normal development system is offline so I used the V5.0 offline installer which failed after ~90% of the installation had completed.

I installed the latest Eclipse installer V2.7.0 with IDF V4.4.3  (_espressif-ide-setup-2.7.0-with-esp-idf-4.4.3.exe_) on new machine with internet disconnected, all ok builds example project.

Repeated the V5.0 offline installation on this new new an online machine with the network connection un plugged and it fails in exactly the same way.
If the network connection is restored then the installation succeeds.

I have attached the log so you can see that the installer does not cope when it can't download xtensa-esp-elf-gdb-11.2_20220823-x86_64-w64-mingw32.zip from git hub.

Regards Dave.

[Off line Setup Log 2022-12-19 #002.zip](https://github.com/espressif/esp-idf/files/10267075/Off.line.Setup.Log.2022-12-19.002.zip)

### Build or installation Logs.

```plain
Log added as zip file
```

### More Information.

I have submitted a Forum issue, https://www.esp32.com/viewtopic.php?f=13&t=31179

dobairoland, 2022-12-20:
@georgik PTAL. Maybe the tools.json file was out of sync with the included tool in the installer?

igrr, 2022-12-20:
Looks like we are missing the GDB tools (Xtensa and RISC-V) here: https://github.com/espressif/idf-installer/blob/ddda7acc33309272d8dcc99ade8c8b6a8e6164cc/src/InnoSetup/IdfToolsSetup.iss#L224-L237

georgik, 2022-12-22:
Thank you for reporting the problem. Confirmed, it's not possible to deploy all target's with Offline installer.

Offline installer is downloading tools by command:
```
idf_tools.py --tools-json tools/tools.json --non-interactive install
```
Then the installation is performed by
```
python.exe "C:\Espressif\frameworks\esp-idf-v5.0\tools\idf_tools.py" "--idf-path=C:\Espressif\frameworks\esp-idf-v5.0"   install --targets=esp32,esp32-c3,esp32-s3,esp32-s2
```

The first command does not download gdb, so it's missing in the final image (`xtensa-esp-elf-gdb-11.2_20220823-x86_64-w64-mingw32.zip`)

georgik, 2022-12-22:
The rebuild of the installer is ready.

@DspDave, please, re-download the latest version of the Offline installer. The installation should be successful.
Download: https://github.com/espressif/idf-installer/releases/tag/offline-5.0

closed 2022-12-23 by georgik

DspDave, 2023-01-04:
Thanks, the offline IDF-V5.0  now works correctly, however the Eclipse plugin V2.7.0 is now working correctly.
Bug issue raised https://github.com/espressif/idf-eclipse-plugin/issues/672.