espressif/esp-idf #10411: BLE GATT Server with 4 Characteristics in each

meakashrao, 2022-12-20:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

4.4.3

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP-WROOM-32

### Power Supply used.

USB

### What is the expected behavior?

I am using the gatt_server_service_table example, 
I am trying to replicate the current Heart Rate service and create a second service like it with 3 characteristics in the second service as well.

I replicated the code to the best of my understanding using the gatt_server_service_table walkthrough and the code that I have compiles properly without any errors or warnings. 
Still the BLE part doesn't starts which is evident using the debug logs below.

### What is the actual behavior?

I need 2 services with 3 characteristics each, 

the default example gives me 1 service with 3 characteristics.
Please help me with solving this. 

### Steps to reproduce.

I copy the required code snippets of the example to create 2 services using tables but am not able to do so.

The main file of the code is:

/*
 * SPDX-FileCopyrightText: 2021 Espressif Systems (Shanghai) CO LTD
 *
 * SPDX-License-Identifier: Unlicense OR CC0-1.0
 */

/****************************************************************************
*
* This demo showcases creating a GATT database using a predefined attribute table.
* It acts as a GATT server and can send adv data, be connected by client.
* Run the gatt_client demo, the client demo will automatically connect to the gatt_server_service_table demo.
* Client demo will enable GATT server's notify after connection. The two devices will then exchange
* data.
*
****************************************************************************/

#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "freertos/event_groups.h"
[... 808 lines cut ...]
I (438) cpu_start: App version:      1
I (442) cpu_start: Compile time:     Dec 20 2022 17:55:23
I (448) cpu_start: ELF file SHA256:  f114e42d17a0472e...
I (454) cpu_start: ESP-IDF:          v4.4.3
I (459) heap_init: Initializing. RAM available for dynamic allocation:
I (466) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (472) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (478) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (485) heap_init: At 3FFC6B10 len 000194F0 (101 KiB): DRAM
I (491) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (497) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (504) heap_init: At 400970CC len 00008F34 (35 KiB): IRAM
I (511) spi_flash: detected chip: generic
I (514) spi_flash: flash io: dio
I (520) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (572) BTDM_INIT: BT controller compile version [67ee4f0]
I (582) system_api: Base MAC address is not set
I (582) system_api: read default base MAC address from EFUSE
I (582) phy_init: phy_version 4670,719f9f6,Feb 18 2021,17:07:07
```

### More Information.

I am trying this on ESP-IDF extension running on VS Code

meakashrao, 2022-12-20:
Power Supply used.

USB+External Supply to prevent brownout detection

xiewenxiang, 2022-12-30:
Does it the same issue https://github.com/espressif/esp-idf/issues/10424?

meakashrao, 2022-12-30:
Yes, pretty much. Thank you for the help.

closed 2022-12-30 by meakashrao