espressif/esp-idf #10415: Nothing scanned when enabling Nimble extended advertising flasg

QuentinFarizon, 2022-12-20:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.0

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32S3-DevkitM1

### Power Supply used.

USB

### What is the expected behavior?

I have flashed example `nimble/blecent`.
When I flash using default settings, I receive multiple BLE_GAP_EVENT_DISC events (I have dozens of BLE devices around me)

But when I set the build variable CONFIG_BT_NIMBLE_EXT_ADV to manage extended advertising (`CONFIG_BT_NIMBLE_EXT_ADV=y`), I receive NO BLE_GAP_EVENT_DISC events.

Shouldn't I still scan all the devices ? 

### What is the actual behavior?

I should still be able to scan all BLE devices.

### Steps to reproduce.

Add a log on scan :
```
case BLE_GAP_EVENT_DISC:
        MODLOG_DFLT(INFO, "Discover event %s", addr_str(event->disc.addr.val));
```

Set `CONFIG_BT_NIMBLE_EXT_ADV=y` in sdkconfig.defaults
Flash example `nimble/blecent`
Wait for scan data in monitor

### More Information.

Here is the interesting part of my sdkconfig :

```
CONFIG_BT_NIMBLE_50_FEATURE_SUPPORT=y
CONFIG_BT_NIMBLE_LL_CFG_FEAT_LE_2M_PHY=y
CONFIG_BT_NIMBLE_LL_CFG_FEAT_LE_CODED_PHY=y
CONFIG_BT_NIMBLE_EXT_ADV=y
CONFIG_BT_NIMBLE_MAX_EXT_ADV_INSTANCES=1
CONFIG_BT_NIMBLE_EXT_ADV_MAX_SIZE=1650
CONFIG_BT_NIMBLE_ENABLE_PERIODIC_ADV=y
CONFIG_BT_NIMBLE_PERIODIC_ADV_SYNC_TRANSFER=y
CONFIG_BT_NIMBLE_MAX_PERIODIC_SYNCS=1
```

rahult-github, 2022-12-21:
Hi @QuentinFarizon ,

Can you please share your complete sdkconfig file ?

Thanks.

QuentinFarizon, 2022-12-21:
Ok, indeed I'm getting BLE_GAP_EVENT_EXT_DISC instead of BLE_GAP_EVENT_DISC events :+1:

closed 2022-12-21 by QuentinFarizon