espressif/esp-idf #10416: ESP32S3 <-> ADC Data Face Issues

ReneSuch, 2022-12-20:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

**Setup:**
-	ESP32S3 is SPI-Slave
-	between ESP32S3 and ADC there is a voltage level translator (TXB0104 and TXU0104)
-	ADC (AD7134) is SPI-Master
-	SPI setting for data transfer (ADC -> ESP):
      o	quad-channel (data0, data1, data2, data3)
-	SPI interface works fine for writing the ADC registers (ESP -> ADC)

**Problems with the data-interface between ESP and ADC:**
**Problem 1 with the data-interface (ADC -> ESP):**
-	spi_slave_transmit() only receives 3 Bytes (24bit ADC), but should have  received12 Bytes (4x24 bit, as 4 channels are active and SPI is in quad-channel mode)
   o	the received data (3 Bytes) are probably only from data0
   o	according to the datasheet the slave driver also works for quad-channel
Question 1.1: How can I receive quad channel data, when the ESP is in slave mode? Any examples or additional settings?

**Problem 2 with the data-interface (ADC -> ESP):**
-	spi_slave_transmit() is only executed every 10 ms but should be executed every 12.2 us. 
Question 2.1: Is there a lower limit on how fast data can be received? 
Question 2.2: The ESP CS Signal is set by the ADC (Output Data Rate (ODR) pin). It is set to HIGH before a data transmission but for only 4 us. Is this too short for the ESP to recognize a new read command? Should the CS HIGH before a data transmission be longer?

wanckl, 2022-12-21:
@ReneSuch Hi, usually slave device need some specific `CMD Bytes` to enter qaud line mode, for esp32 serials, you need trans  '0x23' first to start a esp32 qaud receive with dma, then trans you data once with 4-line mode.    
So if your ADC(AD7134) is a programmable device, and your esp32 only receive data from SPI, you can reference a half duplex test case in `esp-idf/components/driver/test_apps/spi/slave_hd/main/test_spi_slave_hd.c::"SPI quad hd test "` to configure your SPI master.
more `CMD` value for esp32 serials, see `technical_reference_manual->SPI Controller (SPI)->Functional Description->GP­SPI Works as a Slave :Supported CMD Values in SPI Mode`

And esp32 GPSPI slave can work at frequency up to 60MHz, so 4us of CS HIGH is enough, but 0.5 sclk period of CS active before and after transaction is suggested, and freertos tick and if there is some print in your task, it will cost times.

ReneSuch, 2022-12-23:
@wanckl Hi and thank you very much for your answer. We checked the technical reference and the code sample you sent, which already helped! Some questions remain:

So basically, before every data transfer, the ADC needs to send '0x23' to the ESP32? This cannot be done by the ADC in our understanding. Is there an alternative to configure the ESP in a way that it always assumes quad-channel (just like you would hardcode ADC registers before using/operating it)? Concerning your reference to the technical_reference_manual-> SPI Controller (SPI) chapter: to what sub-chapter are you referring to specifically in your text (30.5.9 / Table 30-15)? Can the ESP32 read data directly from an ADC at all, if the ADC is not specially tuned for handling the ESP32 interface (send additional data first)?

Moreover, so far we used the following driver:
../driver/spi_slave.h
The example you referenced uses:
../driver/spi_slave_hd.h
Why is this driver used in the example? To our understanding the basic spi_slave.h driver should also support half-duplex.

What still remains unclear however, is why data is only received every 10 ms? Is there any link you can make to a specific setting? The CS HIGH signal appears every 12,2 us and is longer than 0.5 SCLK periods. Find attached a screenshot of our communication.

Merry Christmas and thanks again for your highly appreciated support.
[image]

wanckl, 2022-12-28:
@ReneSuch Sorry, I have a covid passed days :rofl: 
1. If use one line (not qaud) mode, CMD Byte is not needed, ESP32 work as a standard spi slave can read your data. 
2. ESP32 slave support a QPI mode that you send another special CMD Byte first then it enter qaud mode, data can transfered without additional CMD Bytes, But you need send CMD Byes again if esp32 is reboot. the special CMD Byte can found normally in the chapter `SPI Controller` of your chip's technical_reference_manual, usually a sheet like this    
[image]

3. what the `channel` you mean, is the adc channel or spi data channel ? I think you need ensure that what the data order of your adc chips output, is it a 4 channel data or 4-line SPI data, because 4-line spi (qaud) is a parallel IO, 1 Byte transfer use 4 data lines simutansly (figure bellow). but  4 channel normally means 1 Byte use only 1 data line, and 4 bytes transferring simutansly, they are different.   
[image]

4.  `spi_slave.h` driver can also support half-duplex, but qaud mode only supported in half-duplex driver of IDF now.
5.  What the freertos tick freq of your project, max is up to 1kHz, and the cpu freq can be 240MHz for esp32 and esp32s3.

ReneSuch, 2022-12-30:
Hi @wanckl ,
oh no! I hope you are back to full health now and thanks again for commenting. You are really helpful!

Okay, thanks for your explanation. We had a misunderstanding on what quad mode meant. As our ADC has 4 separate data lines (1 bit on each of the 4 output channels), we thought this is exactly what we want (in the screenshot above, we only showed 1 channel). However, we can also configure the ADC in a way, that it sends the data of the 4 channels only in one channel (and then we don't need the CMD byte). 

Our goal is that the ADC then sends these 4x3 Bytes (or 4x24 bit) succesively on one channel at a rate of approx. 82 kHz (4 Channels x 3 Byte x 82 kHz = 984 kB/s). Would this be possible? In our code there are no delays (it is a while-loop pinned to one core); are there any dependencies in the driver itself upon the freertos (in other words, are there any delays in the driver)? So how fast can we read 4 x 24 bit data on one channel with the ESP32-S3?

wanckl, 2023-01-03:
@ReneSDot I have tried your situation and got the result bellow:
1. the bit rate of spi on one channel is enough (984 kB/s * 8 = 7.8 Mb/s), you can set your spi master work at this freq or upper. but the limitation is that freertos spend about 70us to deal with signals in my test, so that the package rate(3*4 byte) just up to about 13k, which seems that your goal is unreachable.
2. you can change something in esp-idf to reduce cost of freertos as little as passable, but I have not test how many upgrade of this way.
3. but normally, small pied data can't got a good transaction through a OS, the OS cost more time than actual data transaction of a period, so would you buffer your ADC data as a package about 1kB then transaction once, transaction rate will got improvement.

you can have a try if possible, and I'll found more ways, any question you can at me here, good luck.

ReneSDot, 2023-01-07:
Hey @wanckl,
thank you once more for the valuable insights! 

1. We are a little surprised that transaction takes so long (we would need less than 12 us) and needs to involve the OS, as there is a dedicated hardware interface for transactions. 
2. Did you test it with an ADC or another ESP32? 
3. It would be great if you can share your test code, to see, where we possibly go wrong.

Thank you and best wishes!

wanckl, 2023-01-11:
@ReneSDot thank you for your wishes!
I'm sorry to tell you that 82kHz of package rate maybe hard to realize for ESP32 depend on my test.
1. For slave side, I have tested with a new added API to reduce OS cost as less as passable, although it is a private API now, you can find it in the master branch of esp-idf next week.
2. I tested with another ESP32, and similarly the SPI master of ESP32 is hard to output a package rate at approx 82 kHz, so I just use a GPIO level connect to the CS line to my slave, as SPI slave can detect a transaction by CS line, you can ignore this an replace with your ADC device.
3. In my best, the SPI slave can receive a dummy package approx every 14us,  so the package rate reached about 70kHz, but this rate may go down after add your process code.
4. To ensure ADC data transaction as early as passable, could you set your SPI master at a clock freq upper than 20MHz, then, to use this new added API, you should notice that:   
> 1). With this new API, you can get your transaction result within the `post_trans_cb`, refer detail usage in test case `esp-idf/components/driver/test_apps/spi/slave/main/test_spi_slave.c: "SPI_Slave: Test_Queue_Trans_in_ISR"`.
> 2). Notice that the call back func is run within an SPI ISR, so you shouldn't do mass process in this func, and leave quickly.   
5. My test code:   
```c
#include "driver/gpio.h"
#include "esp_private/spi_slave_internal.h"
static void test_short_master(void){
    gpio_config_t io_conf = {};
    io_conf.pin_bit_mask = 1ULL<<PIN_NUM_CS;
    io_conf.mode = GPIO_MODE_OUTPUT;
    gpio_config(&io_conf);
    gpio_set_level(PIN_NUM_CS, 1);  // set init level of CS in SPI mode 0
    esp_rom_delay_us(1);

    unity_send_signal("Master ready");
    unity_wait_for_signal("Slave ready");
    for (int i = 0; i < 1000; i++) {
        gpio_set_level(PIN_NUM_CS, 0);
        esp_rom_delay_us(3);
        gpio_set_level(PIN_NUM_CS, 1);
        esp_rom_delay_us(10);
    }
}

uint16_t cnt=0;
static void test_slave_short_post_trans_cbk(spi_slave_transaction_t *curr_trans)
{
    cnt ++;
    if (ESP_OK == spi_slave_queue_trans_isr(TEST_SPI_HOST, curr_trans)) {
    } else {
        esp_rom_printf(DRAM_STR("SPI Add trans in isr fail, Queue full\n"));
    }
}

static void test_short_slave(void){
    uint8_t slave_send  [12];
    uint8_t slave_recive[12];

    spi_bus_config_t buscfg = SPI_BUS_TEST_DEFAULT_CONFIG();
    spi_slave_interface_config_t slvcfg = SPI_SLAVE_TEST_DEFAULT_CONFIG();
    slvcfg.post_trans_cb = test_slave_short_post_trans_cbk;
    TEST_ESP_OK(spi_slave_initialize(TEST_SPI_HOST, &buscfg, &slvcfg, SPI_DMA_DISABLED));

    spi_slave_transaction_t trans_cfg = {
        .tx_buffer = &slave_send,
        .rx_buffer = &slave_recive,
        .length = 12 * 8,
    };
    unity_wait_for_signal("Master ready");
    TEST_ESP_OK(spi_slave_queue_trans(TEST_SPI_HOST, &trans_cfg, portMAX_DELAY));

    vTaskDelay(1000);   //waitting trans finish by master

    printf("cnt %d\n", cnt);
    TEST_ESP_OK(spi_slave_free(TEST_SPI_HOST));
}

TEST_CASE_MULTIPLE_DEVICES("TEST_SPI_Mass_Short_pkg", "[spi_ms][timeout=30]", test_short_master, test_short_slave);
```

wanckl, 2023-01-13:
@ReneSDot The new API is now available on Github, you can have a try

ReneSDot, 2023-01-15:
Dear @wanckl,
thank you so much for both your posts. It helped very much (again!). I need some more time with my answer / possible new questions this time, as I am on vacation right now. But I will not forget!
Have a good day and hear you soon!

wanckl, 2023-02-06:
@ReneSDot How are you recently, I find a issue seems similarly with your one, maybe you can take a look https://github.com/espressif/esp-idf/pull/9363

ReneSDot, 2023-02-11:
Dear @wanckl,
thank you for reaching out! I am fine and hope you are too! This post looks indeed interesting! To conquer our issue, we will test 3 different approaches now:

1. We will test the I2S interface, maybe this will be faster.
 
2. We will try to manipulate the ODR/CS Signal with another chip: for example intercept the ODR signal (frequency of 81.92 kHz) with another chip and then only give to the ESP32 every 10th real ODR signal; then on the data channel the ESP will see 10 x 24 bit for each received (via the other chip) ODR signal/transmission and only after these 10 x 24 bit we actually handle the data, which will then call the OS --> so basically reduce the number of OS interventions and hence increase speed.

3. We will check the new API with the new commit/command (#9363) you just referenced! Sounds quite promising!

Let me know what you think of option 1 and 2? It would again be very valuable to get an expert opinion! The issue is not solved (yet!).

wanckl, 2023-02-13:
I'm sorry can't resolve your problem for long time.
For your test, IMO:
1. If your ADC device support i2s interface, it is a good idea, because i2s is a continuous peripheral, while if not, it will same as SPI because i2s still transaction data through freertos.
2. The pr #9363 add new APIs for SPI master driver, and the private APIs I mentioned on #10416#issuecomment-1378209902 is SPI slave driver, they are both the same type if ISR version APIs which used within a ISR, I may try to merge that PR, it may be the most easy way for you to have a try.

Thank you for your trust, waiting for your reply !

ReneSDot, 2023-02-28:
Dear @wanckl,
thanks again for bringing us forward! Your help is unbelievable valuable. In the last 2 weeks we progressed by using the I2S interface and feel that we almost reach our goal. Very close! However, we started a new issue concerning using the I2S interface: #10863. We hope it is fine, as our new question is not related to this original problem anymore. Please feel free to comment there too! We would be really grateful for your help again! All the best to you and thanks for your great support!

closed 2023-02-28 by ReneSDot