espressif/esp-idf #10417: PSRAM ID read error:  0x00ffffff

Nickobellic15X, 2022-12-21:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hardware Description:

ESP32 S3 N16R8
Uart0 speed : 115200bps
------------------------------------------
I'm trying to achieve maximum SPI flash speeds for my application but only 80Mhz Octal Mode on PSRAM and 80Mhz Quad Mode on SPI flash seems to work.

Tried the following configurations:
Flash in Quad Mode and PSRAM in Quad Mode: 
Flash 80Mhz PSRAM 80Mhz : Same error as title.
Flash 120Mhz PSRAM 120Mhz: Same error as title.

Debug output:

I (666) boot: Loaded app from partition at offset 0x10000
I (666) boot: Disabling RNG early entropy source...
D (669) boot: Mapping segment 0 as DROM
D (673) boot: Mapping segment 2 as IROM
D (677) boot: calling set_cache_and_start_app
D (681) boot: configure drom and irom and start
V (686) boot: d mmu set paddr=00010000 vaddr=3c0b0000 size=179264 n=3
V (692) boot: rc=0
V (694) boot: i mmu set paddr=00040000 vaddr=42000000 size=680132 n=11
V (701) boot: rc=0
D (703) boot: start: 0x40375548
0x40375548: call_start_cpu0 at C:/ESP/esp-idf/esp-idf/components/esp_system/port/cpu_start.c:267

W (717) FLASH: DO NOT USE FOR MASS PRODUCTION! Timing parameters will be updated in future IDF version.
E (718) psram: PSRAM ID read error: 0x00ffffff
E (721) cpu_start: Failed to init external RAM!

abort() was called at PC 0x40375628 on core 0
0x40375628: call_start_cpu0 at C:/ESP/esp-idf/esp-idf/components/esp_system/port/cpu_start.c:414 (discriminator 3)

Backtrace:0x40375e22:0x3fceb2500x4037d979:0x3fceb270 0x4038583a:0x3fceb290 0x40375628:0x3fceb300 0x403cd97c:0x3fceb330 0x403cdedd:0x3fceb380 0x403c9a05:0x3fceb4b0 0x40045c01:0x3fceb570  |<-CORRUPTED
0x40375e22: panic_abort at C:/ESP/esp-idf/esp-idf/components/esp_system/panic.c:402

0x4037d979: esp_system_abort at C:/ESP/esp-idf/esp-idf/components/esp_system/esp_system.c:128

0x4038583a: abort at C:/ESP/esp-idf/esp-idf/components/newlib/abort.c:46

0x40375628: call_start_cpu0 at C:/ESP/esp-idf/esp-idf/components/esp_system/port/cpu_start.c:414 (discriminator 3)

ELF file SHA256: 2880fcd75277bc44

CPU halted.

Not sure what I'm doing wrong.

Also tring to execute the efuse burn command to enable octal mode on flash yields this: 
> py ./espefuse.py -p /dev/<serial_device> --do-not-confirm burn_efuse FLASH_TYPE 1     

C:\Users\username\AppData\Local\Programs\Python\Python310\python.exe: can't open file 'C:\\Users\\revun\\OneDrive\\Documents\\The IFB-CAM Project\\ESP-IDF\\IFB_Smart_Camera_S3_Chip\\espefuse.py': [Errno 2] No such file or directory

Any help would be appriciated. Thanks

Nickobellic15X, 2022-12-21:
Flash tool for octal worked. I had to use the command like this: 
python -m espefuse -p "COMX" --do-not-confirm burn_efuse FLASH_TYPE 0

Anyway, that did not resolve my issue.
My current worry is I am not able to run both PSRAM and SPI Flash at 120Mhz. Both are in quad mode. Here is the output:

ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x2374
load:0x403c9700,len:0x10a8
load:0x403cc700,len:0x35c8
entry 0x403c99e4
I (27) boot: ESP-IDF v4.4.2-dirty 2nd stage bootloader
I (27) boot: compile time 15:40:01
D (28) bootloader_flash: non-XMC chip detected by SFDP Read (C8), skip.
D (33) bootloader_flash: mmu set block paddr=0x00000000 (was 0xffffffff)
I (39) boot: chip revision: 0
D (43) qio_mode: Probing for QIO mode enable...
D (48) qio_mode: Raw SPI flash chip id 0xc84018
D (52) qio_mode: Manufacturer ID 0xc8 chip ID 0x4018
I (57) qio_mode: Enabling default flash chip QIO
D (63) qio_mode: Initial flash chip status 0x2
D (67) qio_mode: QIO mode already enabled in flash
D (72) qio_mode: Enabling QIO mode...
D (76) boot.esp32s3: magic e9
D (79) boot.esp32s3: segments 03
D (82) boot.esp32s3: spi_mode 02
D (85) boot.esp32s3: spi_speed 0f
D (89) boot.esp32s3: spi_size 04
I (92) boot.esp32s3: Boot SPI Speed : 80MHz
I (97) boot.esp32s3: SPI Mode       : QIO
I (101) boot.esp32s3: SPI Flash Size : 16MB
D (106) boot: Enabling RTCWDT(9000 ms)
I (110) boot: Enabling RNG early entropy source...
D (116) bootloader_flash: mmu set paddr=00000000 count=1 size=c00 src_addr=8000 src_addr_aligned=0
D (125) boot: mapped partition table 0x8000 at 0x3c008000
D (130) flash_parts: partition table verified, 5 entries
I (135) boot: Partition Table:
I (139) boot: ## Label            Usage          Type ST Offset   Length
D (147) boot: load partition table entry 0x3c008000
D (152) boot: type=1 subtype=2
I (155) boot:  0 nvs              WiFi data        01 02 00009000 00006000
D (162) boot: load partition table entry 0x3c008020
D (167) boot: type=1 subtype=1
I (170) boot:  1 phy_init         RF data          01 01 0000f000 00001000
D (178) boot: load partition table entry 0x3c008040
D (183) boot: type=0 subtype=0
I (186) boot:  2 factory          factory app      00 00 00010000 00100000
D (193) boot: load partition table entry 0x3c008060
D (198) boot: type=1 subtype=82
I (202) boot:  3 littlefs         Unknown data     01 82 00110000 00400000
I (209) boot: End of partition table
D (213) boot: Trying partition index -1 offs 0x10000 size 0x100000
D (220) esp_image: reading image header @ 0x10000
D (224) bootloader_flash: mmu set block paddr=0x00010000 (was 0xffffffff)
D (231) esp_image: image header: 0xe9 0x06 0x02 0x04 4037555c
V (237) esp_image: loading segment header 0 at offset 0x10018
V (243) esp_image: segment data length 0x2bc40 data starts 0x10020
V (249) esp_image: segment 0 map_segment 1 segment_data_offs 0x10020 load_addr 0x3c0b0020
I (257) esp_image: segment 0: paddr=00010020 vaddr=3c0b0020 size=2bc40h (179264) map
D (266) esp_image: free data page_count 0x0000003f
D (271) bootloader_flash: mmu set paddr=00010000 count=3 size=2bc40 src_addr=10020 src_addr_aligned=10000
V (308) esp_image: loading segment header 1 at offset 0x3bc60
D (308) bootloader_flash: mmu set block paddr=0x00030000 (was 0xffffffff)
V (309) esp_image: segment data length 0x43b0 data starts 0x3bc68
V (315) esp_image: segment 1 map_segment 0 segment_data_offs 0x3bc68 load_addr 0x3fc985c0
I (323) esp_image: segment 1: paddr=0003bc68 vaddr=3fc985c0 size=043b0h ( 17328) load
[... 44 lines cut ...]
V (700) boot: rc=0
V (702) boot: i mmu set paddr=00040000 vaddr=42000000 size=680132 n=11
V (709) boot: rc=0
D (711) boot: start: 0x4037555c
0x4037555c: call_start_cpu0 at C:/ESP/esp-idf/esp-idf/components/esp_system/port/cpu_start.c:267

W (726) FLASH: DO NOT USE FOR MASS PRODUCTION! Timing parameters will be updated in future IDF version.
E (726) psram: PSRAM ID read error: 0x00ffffff
E (730) cpu_start: Failed to init external RAM!

abort() was called at PC 0x4037563c on core 0
0x4037563c: call_start_cpu0 at C:/ESP/esp-idf/esp-idf/components/esp_system/port/cpu_start.c:414 (discriminator 3)

Backtrace:0x40375e36:0x3fceb2500x4037d98d:0x3fceb270 0x4038584e:0x3fceb290 0x4037563c:0x3fceb300 0x403cdadc:0x3fceb330 0x403ce03d:0x3fceb380 0x403c9a4d:0x3fceb4b0 0x40045c01:0x3fceb570  |<-CORRUPTED  
0x40375e36: panic_abort at C:/ESP/esp-idf/esp-idf/components/esp_system/panic.c:402

0x4037d98d: esp_system_abort at C:/ESP/esp-idf/esp-idf/components/esp_system/esp_system.c:128

0x4038584e: abort at C:/ESP/esp-idf/esp-idf/components/newlib/abort.c:46

0x4037563c: call_start_cpu0 at C:/ESP/esp-idf/esp-idf/components/esp_system/port/cpu_start.c:414 (discriminator 3)

ELF file SHA256: c103d478caed5bc1

CPU halted.

kayzen9, 2022-12-26:
try enabling octal mode:

https://github.com/espressif/esp-idf/blob/cfef24863f1acd4418227a727dbb6e356438a385/components/esp_psram/esp32s3/Kconfig.spiram#L18

Nickobellic15X, 2022-12-28:
My final hardware will not support octal SPI flash, therefore I'm trying to avoid it. :/

Icarus113, 2022-12-28:
> My current worry is I am not able to run both PSRAM and SPI Flash at 120Mhz (QIO). Both are in quad mode.

Hi @Nickobellic15X , I see you're using an ESP32 S3 N16R8, I think this one is with an Octal Flash chip and an Octal PSRAM chip.
For this module:
- the PSRAM always works under DDR mode. 
- the Flash can be switched between DDR mode or SDR mode.

The limitation is:
- 120MHz DDR mode isn't supported yet. 

Therefore to use 120Mhz on this module, you should not enable PSRAM (if you enable it, you'll not see 120MHz option in menuconfig). Because 120MHz flash cannot work with a 80MHz PSRAM. (refer to the [doc](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-guides/flash_psram_config.html#flash-psram-combination:~:text=Flash%20in%20A%20works%20with%20PSRAM%20in%20A/C/D) you posted previously). And you can set the Flash to 120MHz SDR mode.

If you use a module with a Quad Flash and Quad PSRAM (for example ESP32-S3-WROOM-1-N8R2), then you'll be able to set both of them to 120MHz, and under this condition, they're all working under SDR mode.

Icarus113, 2023-01-12:
@Nickobellic15X  Hi, does my comment clarify the situation?

closed 2023-03-07 by Alvin1Zhang