espressif/esp-idf #10418: [Help] ESP32 connected to RTL8201F(PHY) chip, network auto-negotiation timeout, unable to communicate

garytsai168, 2022-12-21:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

  Use the ESP32-WROOM-32E-N16 module to communicate with RTL8201F through the RMII interface, 
and use the \examples\ethernet\basic routine that comes with the IDF programming environment for testing. 
After obtaining the Mac Address, it cannot Link Up Ethernet.
( RMII clock mode (Input RMII clock from external) )

    I (425) cpu_start: Starting scheduler on PRO CPU.
    I (0) cpu_start: Starting scheduler on APP CPU.
    I (445) system_api: Base MAC address is not set
    I (445) system_api: read default base MAC address from EFUSE
    I (465) esp_eth.netif.glue: 10:97:bd:ec:33:fb
    I (465) esp_eth.netif.glue: ethernet attached to netif
    W (4465) rtl8201: auto negotiation timeout
    I (4465) eth_example: Ethernet Starte
    ...

This is the source code and circuit diagram I use, looking forward to someone can help,
Grateful !

[SME__ESP32+RTL8201F.pdf](https://github.com/espressif/esp-idf/files/10275936/SME__ESP32%2BRTL8201F.pdf)
[main.txt](https://github.com/espressif/esp-idf/files/10275951/main.txt)

    #include <stdio.h>
    #include <string.h>
    #include <tcpip_adapter.h>
    #include "freertos/FreeRTOS.h"
    #include "freertos/task.h"
    #include "esp_netif.h"
    #include "esp_eth.h"
    #include "esp_event.h"
    #include "esp_log.h"
    #include "driver/gpio.h"
    #include "sdkconfig.h"
    
    static const char *TAG = "eth_example";
    
    /** Event handler for Ethernet events */
    static void eth_event_handler(void *arg, esp_event_base_t event_base,
                                  int32_t event_id, void *event_data) {
        uint8_t mac_addr[6] = {0};
        /* we can get the ethernet driver handle from event data */
        esp_eth_handle_t eth_handle = *(esp_eth_handle_t *) event_data;
    
        switch (event_id) {
            case ETHERNET_EVENT_CONNECTED:
                esp_eth_ioctl(eth_handle, ETH_CMD_G_MAC_ADDR, mac_addr);
                ESP_LOGI(TAG, "Ethernet Link Up");
                ESP_LOGI(TAG, "Ethernet HW Addr %02x:%02x:%02x:%02x:%02x:%02x",
                         mac_addr[0], mac_addr[1], mac_addr[2], mac_addr[3], mac_addr[4], mac_addr[5]);
                break;
            case ETHERNET_EVENT_DISCONNECTED:
                ESP_LOGI(TAG, "Ethernet Link Down");
                break;
            case ETHERNET_EVENT_START:
                ESP_LOGI(TAG, "Ethernet Started");
                break;
            case ETHERNET_EVENT_STOP:
                ESP_LOGI(TAG, "Ethernet Stopped");
                break;
            default:
                break;
        }
    }
    
    /** Event handler for IP_EVENT_ETH_GOT_IP */
    static void got_ip_event_handler(void *arg, esp_event_base_t event_base,
                                     int32_t event_id, void *event_data) {
        ip_event_got_ip_t *event = (ip_event_got_ip_t *) event_data;
        const esp_netif_ip_info_t *ip_info = &event->ip_info;
    
        ESP_LOGI(TAG, "Ethernet Got IP Address");
        ESP_LOGI(TAG, "~~~~~~~~~~~");
        ESP_LOGI(TAG, "ETHIP:" IPSTR, IP2STR(&ip_info->ip));
        ESP_LOGI(TAG, "ETHMASK:" IPSTR, IP2STR(&ip_info->netmask));
        ESP_LOGI(TAG, "ETHGW:" IPSTR, IP2STR(&ip_info->gw));
        ESP_LOGI(TAG, "~~~~~~~~~~~");
    }
    
    void app_main(void) {
        ESP_ERROR_CHECK(esp_netif_init());
        // Create default event loop that running in background
        ESP_ERROR_CHECK(esp_event_loop_create_default());
        // Create new default instance of esp-netif for Ethernet
        esp_netif_config_t cfg = ESP_NETIF_DEFAULT_ETH();
        esp_netif_t *eth_netif = esp_netif_new(&cfg);
        /*esp_netif_set_hostname(eth_netif,"test");*/
        ESP_ERROR_CHECK(esp_eth_set_default_handlers(eth_netif));
        // Init MAC and PHY configs to default
        eth_mac_config_t mac_config = ETH_MAC_DEFAULT_CONFIG();
        eth_phy_config_t phy_config = ETH_PHY_DEFAULT_CONFIG();
    
        phy_config.phy_addr = 0;
        phy_config.reset_gpio_num = 33;
        mac_config.smi_mdc_gpio_num = 23;
        mac_config.smi_mdio_gpio_num = 18;
        esp_eth_mac_t *mac = esp_eth_mac_new_esp32(&mac_config);
        esp_eth_phy_t *phy = esp_eth_phy_new_rtl8201(&phy_config);
        esp_eth_config_t config = ETH_DEFAULT_CONFIG(mac, phy);
        esp_eth_handle_t eth_handle = NULL;
        ESP_ERROR_CHECK(esp_eth_driver_install(&config, &eth_handle));
        /* attach Ethernet driver to TCP/IP stack */
        ESP_ERROR_CHECK(esp_netif_attach(eth_netif, esp_eth_new_netif_glue(eth_handle)));
        ESP_ERROR_CHECK(esp_event_handler_register(ETH_EVENT, ESP_EVENT_ANY_ID, &eth_event_handler, NULL));
        ESP_ERROR_CHECK(esp_event_handler_register(IP_EVENT, IP_EVENT_ETH_GOT_IP, &got_ip_event_handler, NULL));
        ESP_ERROR_CHECK(esp_eth_start(eth_handle));
    }

garytsai168, 2022-12-23:
hardware circuit problem
done.

closed 2022-12-23 by garytsai168

lebies, 2022-12-25:
Hi @garytsai168, if you don't mind, what was the circuit problem exactly please?
Thanks in advance, appreciate...

mercenaruss, 2023-11-07:
> hardware circuit problem
> 
> done.

Can you share with us updated schematic?