espressif/esp-idf #10419: Nondeterministic order of components in sdkconfig

nx518, 2022-12-21:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.0

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

"git status" should always be:
working tree clean

### What is the actual behavior?

output of "git status"
sometimes: 
working tree clean 
sometimes: 
modified: sdkconfig

### Steps to reproduce.

1. Have a project with 2 components like idf_component.yml containing:
...
espressif/mdns: "1.0.7"
espressif/esp-modbus: "1.0.5"
...

2. idf.py build
3. git status
working tree clean
4. idf.py fullclean
5. idf.py build
6. git status
modified: sdkconfig

### Build or installation Logs.

```plain
diff --git a/sdkconfig b/sdkconfig
index b43ade7..2c2d987 100644
--- a/sdkconfig
+++ b/sdkconfig
@@ -1715,32 +1715,6 @@ CONFIG_WPA_MBEDTLS_TLS_CLIENT=y
 #
 # end of esp_mse1021 Konfiguration
 
-#
-# mDNS
-#
-CONFIG_MDNS_MAX_INTERFACES=3
-CONFIG_MDNS_MAX_SERVICES=10
-CONFIG_MDNS_TASK_PRIORITY=1
-CONFIG_MDNS_TASK_STACK_SIZE=4096
-# CONFIG_MDNS_TASK_AFFINITY_NO_AFFINITY is not set
-CONFIG_MDNS_TASK_AFFINITY_CPU0=y
-# CONFIG_MDNS_TASK_AFFINITY_CPU1 is not set
-CONFIG_MDNS_TASK_AFFINITY=0x0
-CONFIG_MDNS_SERVICE_ADD_TIMEOUT_MS=2000
-# CONFIG_MDNS_STRICT_MODE is not set
-CONFIG_MDNS_TIMER_PERIOD_MS=100
-# CONFIG_MDNS_NETWORKING_SOCKET is not set
-CONFIG_MDNS_MULTIPLE_INSTANCE=y
-
-#
-# MDNS Predefined interfaces
-#
-CONFIG_MDNS_PREDEF_NETIF_STA=y
-CONFIG_MDNS_PREDEF_NETIF_AP=y
-CONFIG_MDNS_PREDEF_NETIF_ETH=y
-# end of MDNS Predefined interfaces
-# end of mDNS
-
 #
 # Modbus configuration
 #
@@ -1771,6 +1745,32 @@ CONFIG_FMB_EVENT_QUEUE_TIMEOUT=20
 # CONFIG_FMB_TIMER_PORT_ENABLED is not set
 # CONFIG_FMB_TIMER_USE_ISR_DISPATCH_METHOD is not set
 # end of Modbus configuration
+
+#
+# mDNS
+#
+CONFIG_MDNS_MAX_INTERFACES=3
+CONFIG_MDNS_MAX_SERVICES=10
+CONFIG_MDNS_TASK_PRIORITY=1
+CONFIG_MDNS_TASK_STACK_SIZE=4096
+# CONFIG_MDNS_TASK_AFFINITY_NO_AFFINITY is not set
+CONFIG_MDNS_TASK_AFFINITY_CPU0=y
+# CONFIG_MDNS_TASK_AFFINITY_CPU1 is not set
+CONFIG_MDNS_TASK_AFFINITY=0x0
+CONFIG_MDNS_SERVICE_ADD_TIMEOUT_MS=2000
+# CONFIG_MDNS_STRICT_MODE is not set
+CONFIG_MDNS_TIMER_PERIOD_MS=100
+# CONFIG_MDNS_NETWORKING_SOCKET is not set
+CONFIG_MDNS_MULTIPLE_INSTANCE=y
+
+#
+# MDNS Predefined interfaces
+#
+CONFIG_MDNS_PREDEF_NETIF_STA=y
+CONFIG_MDNS_PREDEF_NETIF_AP=y
+CONFIG_MDNS_PREDEF_NETIF_ETH=y
+# end of MDNS Predefined interfaces
+# end of mDNS
 # end of Component config
 
 # Deprecated options for backward compatibility
```

### More Information.

Order of components in sdkconfig seems to change randomly, leading to dirty builds etc.
Changing the order of sdkconfig manually and changing order in idf_component.yaml does not help.

igrr, 2022-12-21:
@nx518 Thanks for reporting this issue. Could you please try the attached patch and see if it works on your side?
[0001-cmake-sort-managed-components-list.patch.txt](https://github.com/espressif/esp-idf/files/10276931/0001-cmake-sort-managed-components-list.patch.txt)

nx518, 2022-12-21:
Thank for the quick reply, however this patch does not fix it neither on v5.0 nor on master.

What I found out is, that it is enough to run "idf.py reconfigure" to get random changes in the component order. 
So need for fullclean etc. to reproduce the issue, just "watch idf.py reconfigure" and "watch git status".

igrr, 2023-01-10:
@nx518 Thank you, I was able to reproduce the issue. We'll fix this in https://github.com/espressif/idf-component-manager soon.

igrr, 2023-01-11:
The fix was merged, there will be a bugfix release [of the component manager](https://github.com/espressif/idf-component-manager/releases) this week. After that you can run `pip install --upgrade idf-component-manager` to get the latest version.

mickeyl, 2024-04-16:
I have the same problem. https://github.com/espressif/esp-idf/issues/13490 is probably also an artefact of this.