espressif/esp-idf #10421: esp32-c3 flashing with esphome issue

evlo, 2022-12-21:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

So this is most likely issue on my side, but I just do not know anymore, so I'm asking here.

Some time before I did get the t-01c3 working in my application using UART.

I wanted to update to newer framework so I can get bluetooth proxy working (I'm using esphome), but i ended up in bootloop and I can't even get the thing to work with my previously working build settings (that is the weirdest thing). It might be espHome issue as that is what i'm using

[image]

I found this https://docs.espressif.com/projects/esp-idf/en/release-v4.4/esp32s3/api-guides/flash_psram_config.html#error-handling about efuses, but before everything worked without it, i do not understand what exactly it is about. Should I run the efuse command?

related esphome issue https://github.com/esphome/issues/issues/3924

checklist is not true as i'm actually using esphome, so please do not pay it any mind.

igrr, 2023-01-10:
@evlo The error happens because the binary file flashed into the ESP32-C3 is malformed: one of the segments has size not divisible by 4.

To troubleshoot this, could you please run and show the output of:

* `esptool.py --chip esp32c3 image_info bootloader.bin` (or whatever is the name of the binary file generated by esphome). You can get esptool here: https://docs.espressif.com/projects/esptool/en/latest/esp32c3/installation.html.
* `riscv32-esp-elf-readelf -S YOUR_APP.elf` (where YOUR_APP.elf is the ELF file of the application)

evlo, 2023-01-13:
https://github.com/espressif/esp-idf/issues/10421#issuecomment-1377634458
```
PS C:\esphome> esptool version
esptool.py v4.4
4.4
```
```
PS C:\esphome> pip index versions esptool
WARNING: pip index is currently an experimental command. It may be removed/changed in a future release without prior warning.
esptool (4.4)
Available versions: 4.4, 4.3, 4.2.1, 4.2, 4.1, 4.0.1, 4.0, 3.3.2, 3.3.1, 3.3, 3.2, 3.1, 3.0, 2.8, 2.7, 2.6, 2.5.1, 2.5.0, 2.4.1, 2.4.0, 2.3.1, 2.3, 2.2.1, 2.2, 2.1, 2.0.1, 2.0, 1.3, 1.2.1, 1.2, 1.1
  INSTALLED: 4.4
  LATEST:    4.4
```

```
PS C:\esphome> esptool --chip esp32c3  image_info "C:\esphome\.esphome\build\eh-elektromer\.pioenvs\eh-elektromer\bootloader.bin"
esptool.py v4.4
Image version: 1
Entry point: 403ce000
3 segments

Segment 1: len 0x00438 load 0x3fcd6100 file_offs 0x00000018 [DRAM,BYTE_ACCESSIBLE]
Segment 2: len 0x00918 load 0x403ce000 file_offs 0x00000458 [IRAM]
Segment 3: len 0x024e4 load 0x403d0000 file_offs 0x00000d78 [IRAM]
Checksum: bf (valid)
Validation Hash: fd190de0d8b02ec50eb7cd38cbb5aa3ce593622bb240387b879e77a6c6fd39dc (valid)
```

```
PS C:\esphome> esptool --chip esp32c3  image_info "C:\esphome\.esphome\build\eh-elektromer\.pioenvs\eh-elektromer\firmware.bin"
esptool.py v4.4
Image version: 1
Entry point: 40382196
6 segments

Segment 1: len 0x1f5f8 load 0x3c0b0020 file_offs 0x00000018 [DROM]
Segment 2: len 0x009f8 load 0x3fc8e600 file_offs 0x0001f618 [DRAM,BYTE_ACCESSIBLE]
Segment 3: len 0xaddfc load 0x42000020 file_offs 0x00020018 [IROM]
Segment 4: len 0x029f4 load 0x3fc8eff8 file_offs 0x000cde1c [DRAM,BYTE_ACCESSIBLE]
Segment 5: len 0x0e5c4 load 0x40380000 file_offs 0x000d0818 [IRAM]
Segment 6: len 0x00010 load 0x50000010 file_offs 0x000dede4 [RTC_IRAM,RTC_DRAM]
Checksum: 6f (valid)
Validation Hash: 65bcb8fb195798906eea59af47c1eadd7979f2b7e86caf8295b903226f6b1d3e (valid)
```

```
PS C:\esphome> py c:\python\scripts\readelf.py C:\esphome\.esphome\build\eh-elektromer\.pioenvs\eh-elektromer\firmware.elf
PS C:\esphome>
```
That last command i'm still trying to figure out on win

closed 2024-07-12 by dobairoland