espressif/esp-idf #10422: Support the component manager for dependencies of components

Delzye, 2022-12-21:
### Is your feature request related to a problem?

In our environment we have multiple components, which are needed for (nearly) every project, so they reside in a system folder and are included via the EXTRA_COMPONENT_DIRS variable in CMake.

One of these components is responsible for handling wifi (let's call it Component A).
Any project of ours includes a component (say, Component B), which depends on Component A, since it starts wifi etc.

Component A uses mdns, which is not part of IDF since v5.0.

So I wanted to include it in Component A via the component manager. This does not seem to work, since the list that is searched for manifests only contains local components and anything listed in the project's CMakeLists.txt, so anything included by an extra component is not searched for an idf_components.yml.

### Describe the solution you'd like.

An option to recursively search dependencies for component manifests would be nice.

### Describe alternatives you've considered.

1. Adding mdns to every project's idf_components.yml (awful) 
2. adding mdns to component B's idf_components.yml (not optimal, since one can theoretically use Component A without Component B and to enable that, one would have to maintain both idf_component.yml's (possible version missmatch and other issues)).
3. Downloading the mdns component to the extra component directory and skipping the manager alltogether. Not nice for version management and since the manager already exists, I would like to use it.

igrr, 2022-12-21:
Would you consider adding dependency on your common components via idf_component.yml as well? The manifest file 
 supports dependencies on components in local directories as well. If there are two many components, you can create one "wrapper" component which depends on all other common components, and only include the "wrapper" via idf_component.yml of your project.

Delzye, 2022-12-21:
Hm, that does sound like an option, too. One Question: Does the idf_component.yml support only including components, if a certain sdkconfig-variable is set? That would be the best thing i've heard all week!

igrr, 2022-12-21:
> Does the idf_component.yml support only including components, if a certain sdkconfig-variable is set?

That's not possible, unfortunately, because the list of components needs to be known before Kconfig options are evaluated. Same applies to adding components via EXTRA_COMPONENT_DIRS.

Delzye, 2022-12-21:
That's unfortunate, but I understand why. Thanks anyway!

closed 2022-12-21 by Delzye