espressif/esp-idf #10423: ESP32-C6 support status

igrr, 2022-12-21:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

ESP-IDF v5.0, currently in development, will include initial support for ESP32-C6 chip.

For each feature, the status (Supported/Not supported) is given. An internal issue reference (such as "IDF-1234") is listed at the end of the feature description to help us keep this list up to date.

This list will be updated as the support for ESP32-C6 is being developed.

- Wi-Fi
    - 802.11bgn, HT40 – Not supported (WIFI-5174)
    - 802.11ax: 20MHz-only non-AP mode – Not supported (WIFI-5175)
    - Station, SoftAP, sniffer mode – Not supported (WIFI-5176)
    - UL/DL OFDMA (26/52/106/242 tone) – Not supported (WIFI-5185)
    - DL MU-MIMO，DL MU-MIMO within OFDMA – Not supported (WIFI-5188)
    - HE ER (HE Extended Range), DCM and ESP LR (ESP Long Range) – Not supported (WIFI-5193)
    - BSS Color – Not supported (WIFI-5141)
    - CSI (Channel State Information) – Not supported (WIFI-5157)
    - ESP-NOW – Not supported (WIFI-5154)
    - ESP-WiFi-Mesh – Not supported (WIFI-5155)
    - ESP-Touch v1/v2 – Not supported (WIFI-5156)
    - Wi-Fi Security – Not supported (WIFI-5200)
    - DPP (Device Provisioning Protocol) – Not supported (WIFI-5168)
    - Network-assisted roaming using 802.11k (Radio Measurements), 802.11v (BTM) specification and 802.11r (Fast BSS Transition) – Not supported (WIFI-5160)
    - TWT– Not supported (WIFI-5138) 
        - Individual TWT
        - Broadcast TWT 
    - Advanced DTIM Sleep – Not supported  (WIFI-5151）
    - Power Save: modem sleep, light sleep – Not supported  (WIFI-5150）
    - FTM (Fine Time Measurement)  – Not supported  (WIFI-5144)
- Coexistence
    - Coexistence of Wi-Fi and Bluetooth LE - Not supported (FCC-49)
    - Coexistence of Wi-Fi and 802.15.4 - Not supported (FCC-50)
    - Coexistence of Bluetooth LE and 802.15.4 - Not supported (FCC-51)
    - External coexistence - Not supported (FCC-52)
- Bluetooth LE
    - Bluetooth 5 Controller (Bluetooth 5.3 Certified) - Not supported（IDF-6524）
    - Bluedroid Host - Not supported（IDF-6525）
    - NimBLE Host - Not supported（IDF-6526）
    - ESP-BLE-MESH - Not supported（IDF-6527）
    - BLE Modem Sleep and Light Sleep - Not supported（IDF-6528）
    - Bluetooth Power Control - Not supported（IDF-6529）
    - HCI - Not supported（IDF-6530）
- 802.15.4
    - 802.15.4 Mac - Not supported (TZ-38)
    - 802.15.4 CSL - Not supported (TZ-41)
- Thread
    - Thread FTD/MTD - Not supported (TZ-38)
    - Thread RCP - Not supported (TZ-38)
    - Thread Border Router - Not supported (TZ-42)
- Zigbee
    - Zigbee 3.0 Home Automation - Not supported (TZ-39)
    - Zigbee RCP - Not supported (TZ-39)
    - Zigbee Gateway - Not supported (TZ-39)
- Peripherals
    - ADC
        - ADC Calibration - Not supported (IDF-5312)
        - ADC continuous mode driver - Not supported (IDF-5311)
        - ADC oneshot mode driver - Not supported (IDF-5310)
    - Dedicated GPIO driver - Supported
    - Ethernet driver
        - SPI Ethernet driver - Not supported (IDF-5325)
    - Event Task Matrix - Supported
    - GDMA - Supported
    - GPIO driver - Supported
        - Glitch Filter - Supported
        - RTC IO driver - Not supported (IDF-6027)
    - GPIO sigma delta driver - Supported
    - GPSPI
        - SPI Master driver - Not supported (IDF-5334)
        - SPI Slave HD driver - Supported
        - SPI Slave driver - Supported
    - GPTimer driver - Supported
    - I2C
        - I2C master driver - Supported
        - I2C slave driver - Supported
    - I2S driver - Supported
        - I2S PDM rx mode
        - I2S PDM tx mode
        - I2S STD mode
        - I2S TDM mode
    - LCD driver - Supported
        - I2C LCD driver
        - SPI LCD driver
    - LEDC driver - Supported
        - Gamma Curve Generation - Not supported (IDF-5811)
    - MCPWM driver - Supported
    - PCNT driver - Supported
    - Parallel IO driver - Not supported (IDF-5805)
    - RMT driver - Supported
    - SD/SDIO/MMC driver
        - SDSPI Host driver - Not supported (IDF-5860)
    - SDIO Slave driver - Not supported (IDF-5809)
    - SPI Flash driver - Supported
    - Systimer driver
        - Esptimer implementation over systimer - Supported
        - OS tick port over systimer - Not supported (IDF-5324)
    - TWAI driver - Supported
    - Temperature Sensor driver - Supported
    - UART
        - UART FIFO mode driver - Supported
        - UHCI (UART DMA mode) driver - Not supported (IDF-5337)
    - USB Serial JTAG - Supported
    - memory copy by DMA - Supported
- Power management
    - Auto Light Sleep - Not supported (IDF-5350)
    - Deep Sleep - Not supported (IDF-5349)
    - Dynamic Frequency Switch - Not supported (IDF-5347)
    - Light Sleep - Not supported (IDF-5348)
- Security Features
    - AES accelerator - Not supported (IDF-5356)
    - Digital Signature driver - Not supported (IDF-5360)
    - ECC accelerator - Supported
    - Flash encryption - Not supported (IDF-5358)
    - HMAC driver - Not supported (IDF-5355)
    - Permission Control - Not supported (IDF-5354)
    - RNG - Not supported (IDF-5352)
    - RSA hardware accelerator driver - Not supported (IDF-5359)
    - SHA accelerator - Not supported (IDF-5353)
    - Secure boot - Not supported (IDF-5357)
    - Trusted Execution Environment
        - Access Permission Management - Not supported (IDF-5818)
        - TEE Controller - Not supported (IDF-5819)
- System Features
    - Brownout Detector - Not supported (IDF-5345)
    - Cache - Not supported (IDF-5342)
    - Efuse controller driver - Supported
    - MMU - Not supported (IDF-5339)
    - MSPI
        - MSPI tuning (Flash & PSRAM DDR and/or over 80 MHz) - Not supported (IDF-5343)
        - SPI Flash auto suspend - Not supported (IDF-5344)
    - Timeout Module (New) - Not supported (IDF-5814)

*notes: Nested features can be considered as supported, if the parent feature is shown as `Supported`, and it's not explicitly mentioned as `Not supported`. *

igrr, 2023-01-02:
Update 2023/01/02: added Wi-Fi features, ADC oneshot and continuous mode, RTC IO, SPI Master, SDSPI Host drivers.

igrr, 2023-01-13:
Update 2023/01/13: added BLE features, Brownout Detector support

igrr, 2023-01-25:
Update 2023/01/25: added HMAC, Digital Signature, SDIO slave, Bluetooth Power Control.

mahavirj, 2023-02-20:
Update 2023/02/20: added RSA, Secure Boot, Flash Encryption, APM.

gojimmypi, 2023-03-05:
I've confirmed wolfSSL encryption libraries work with the ESP-IDF v5.1 and the ESP32-C6. It is operational on my [fork](https://github.com/gojimmypi/wolfssl/tree/ED25519_SHA2_fix) and I plan to submit a PR soon to address https://github.com/wolfSSL/wolfssl/issues/6163 for full support.

tom-borcin, 2023-03-15:
Update 2023/03/15: added Coexistence of Wi-Fi and Bluetooth LE, and Coexistence of Bluetooth LE and 802.15.4

jack0c, 2023-03-15:
Update 2023/03/15: added Coexistence features and Power save features.

rotilho, 2023-03-16:
Is it be possible to ZigBee / Matter, BLE and WiFi coexist?

Hedda, 2023-03-20:
> Will it be possible ZigBee / Matter, BLE and WiFi to coexist?

Wondering this too about Zigbee and Thread? However as I understand it, Zigbee and Thread will intially be mutually exclusive, so can not currently use Zigbee and Thread simultaneously (at the same time)?

Not sure if feasable nativly on a single-core MCU but perhaps in future Zigbee and Thread could instead be timeshifted/timesliced to run concurrently using RCP (Radio Co-Processor) architecture design in where the Zigbee and Thread libraries/processes are off-loaded from the 802.15.4 SoC radio SoC to a host computer? 

https://openthread.io/platforms/co-processor

That is anyway how Silicon Labs and some others support using two different protocols at the same time on its 802.15.4 SoC radios.

PS: Off-topic but FYI; Home Assistant has experimental add-on for a such Zigbee and Therad implementation using off-loading running concurrently on their Silicon Labs based Home Assistant SkyConnect USB dongle with 802.15.4 SoC radio (and Home Assistent Yellow onboard radio based on the same 802.15.4 SoC radio SoC) in RCP mode.

loraahd, 2023-03-21:
Hello,
Unfortunately I am not able to use my ESP32-C6 DEVKITC-1. That is because I am not able to install ESP-IDF V5.0.1 which is the only version that supports C6 targets. Please find attached the error message and the log file.

Anyone could give support please?
BR
 
[ADH Setup Log 2023-03-21 #001.txt](https://github.com/espressif/esp-idf/files/11026811/ADH.Setup.Log.2023-03-21.001.txt)
[image]

RuslanSpartak, 2023-03-21:
> Hello, Unfortunately I am not able to use my ESP32-C6 DEVKITC-1. That is because I am not able to install ESP-IDF V5.0.1 which is the only version that supports C6 targets. Please find attached the [...]

SDK 5.1 supports esp32c6, but it is not ready. You can use master brach for start of project to use esp32c6, but you will need to make some manuals manupulations to add for examples esp32c6 support in eclipse (as example).

loraahd, 2023-03-21:
> > Hello, Unfortunately I am not able to use my ESP32-C6 DEVKITC-1. That is because I am not able to install ESP-IDF V5.0.1 which is the only version that supports C6 targets. Please find attached the error message and the log file.
> > Anyone could give support please? BR
> > [ADH Setup Log 2023-03-21 #001.txt](https://github.com/espressif/esp-idf/files/11026811/ADH.Setup.Log.2023-03-21.001.txt) [image]
> 
> SDK 5.1 supports esp32c6, but it is not ready. You can use master brach for start of project to use esp32c6, but you will need to make some manuals manipulations to add for examples esp32c6 support in eclipse (as example).

Thanks for you fast response @RuslanSpartak . would you kindly send me the link to the master branch with the needed manual manipulations for C6? do you have any idea on the date SDK 5.1 will begin supporting esp32c6, anytime soon?
BR
Achraf

RuslanSpartak, 2023-03-21:
> Thanks for you fast response @RuslanSpartak . would you kindly send me the link to the master branch with the needed manual manipulations for C6? do you have any idea on the date SDK 5.1 will begin [...]

I'm also a customer that uses ESP32C6 module . You can get master branch from github. Which IDE you use, depends on that I can try to help you. About SDK5.1 I really don't know as I'm not emedded engineer of espressif.

loraahd, 2023-03-21:
> 

I use IAR. I have no problem to use another one.
all I need is to flash my C6 and send commands to connect the C6 wifi to an access point. I don't really need to developpe any SW or FW.
thanks again for your kind support.

RuslanSpartak, 2023-03-21:
> > 
> 
> I use IAR. I have no problem to use another one. all I need is to flash my C6 and send commands to connect the C6 wifi to an access point. I don't really need to developpe any SW or FW. thanks again for your kind support.

Are you able to install master branch instead of 5.0.1?

tom-borcin, 2023-03-21:
Hello @rotilho, @Hedda, @loraahd and @RuslanSpartak,

this issue tracks only ESP32-C6 support status and for better visibility of update comments specifying what support was added, I had to hide your comments.

If you have any questions or need support, please feel free to open new issues [here](https://github.com/espressif/esp-idf/issues/new/choose) on GitHub or on [esp32.com](https://www.esp32.com) forum. Our developers and support team will be happy to help you.

Thank you for understanding, 

Tomas

loraahd, 2023-03-21:
Don't know how to install it. Do you have a link to it?

mrninhvn, 2023-03-21:
> Update 2023/03/15: added Coexistence of Wi-Fi and Bluetooth LE, and Coexistence of Bluetooth LE and 802.15.4

Hi, can Wi-Fi, BLE and 802.15.4 coexist?

Espressif-liuuuu, 2023-03-29:
> > Update 2023/03/15: added Coexistence of Wi-Fi and Bluetooth LE, and Coexistence of Bluetooth LE and 802.15.4
> 
> Hi, can Wi-Fi, BLE and 802.15.4 coexist?

hi @mrninhvn , coexist of three modems together is not scheduled yet

a-d-r-i-a-n-d, 2023-04-16:
Hi @Espressif-liuuuu does it mean that once the modems will coexist together we could have a Thread Border Router just with the ESP32-C6?

igrr, 2023-04-16:
@a-d-r-i-a-n-d please see https://github.com/espressif/esp-thread-br/issues/9#issuecomment-1506402241 and leave a comment there if you have further questions regarding the Border Router solution.

igrr, 2023-04-18:
Update 2023/04/18:  802.15.4 CSL, Thread Border Router, Parallel IO driver, LEDC gamma curve, systimer, cache and MMU are now supported.

loraadh, 2023-05-22:
Hello,
I am not able to use my ESP32-C6 DEVKITC-1. That is because ESP-IDF V5.0.1 still does not supports C6 targets. Please find attached the error message.
Is there an upgrade that support C6 targets? any workaround?
[image]

BR,

mickeyl, 2023-05-22:
@loraadh Espressif have stated that support for the C6 will debut with ESP-IDF 5.1. You can check out the `release/5.1` branch and it will already work pretty well with the C6.

loraadh, 2023-05-22:
> @loraadh Espressif have stated that support for the C6 will debut with ESP-IDF 5.1. You can check out the `release/5.1` branch and it will already work pretty well with the C6.

thanks @mickeyl 
is the 5.1 version the same as the master version?
BR

Hedda, 2023-05-22:
> is the 5.1 version the same as the master version?

No, 5.1 is the "release/v5.1" branch -> https://github.com/espressif/esp-idf/tree/release/v5.1

loraadh, 2023-05-22:
> > is the 5.1 version the same as the master version?
> 
> No, 5.1 is the "release/v5.1" branch -> https://github.com/espressif/esp-idf/tree/release/v5.1
> 
> PS: Also follow [espressif/arduino-esp32#7852](https://github.com/espressif/arduino-esp32/issues/7852) and [espressif/arduino-esp32#7713](https://github.com/espressif/arduino-esp32/issues/7713) for updates on ESP-IDF 5.1 and ESP32-C6

thanks @Hedda,
in previous versions I used to do the installations via the online installer: https://dl.espressif.com/dl/esp-idf/?idf=4.4
Version ESP-IDF 5.1 is still not supported by that installer. Is there a user guide on how to install, flash/upload, and command the c6?
thanks 
BR

Hedda, 2023-05-22:
For end-user questions and want support then you should post them to the esp32.com forum instead -> https://www.esp32.com

Headcrabed, 2023-05-23:
Might be off-topic, but why c6's riscv trace support isn't on this table?

igrr, 2023-05-24:
@Headcrabed indeed we did not include CPU Trace support into the plan, as we saw relatively few people using the similar feature on the previous Xtensa chips. It will get supported, eventually, especially if more users express interest.

mickeyl, 2023-05-30:
Why was https://github.com/espressif/esp-idf/issues/10423#issuecomment-1549680635 marked as being offtopic?

Renatomusica, 2023-06-07:
I Cant see the option to choose the ESP32-C6 Dev Kit in the very first VS code blink example. I am used to Program Texas Instruments And Microchip devices with C compilers and the approach is so different. It is my first time attempting with a ESP device that I just bought, so far this issue is really unfriendly already.. any help on this?
[image]

igrr, 2023-06-07:
@Renatomusica it is possible that the version of IDF you are using is one which doesn't contain C6 support. (For example, v5.0.x doesn't support C6). Please check that you have v5.1-rc1 installed, as that's the latest release candidate of v5.1. If you still run into an issue with VS Code after installing v5.1-rc1, please open an issue here: https://github.com/espressif/vscode-esp-idf-extension/issues

Renatomusica, 2023-06-07:
> @Renatomusica it is possible that the version of IDF you are using is one which doesn't contain C6 support. (For example, v5.0.x doesn't support C6). Please check that you have v5.1-rc1 installed, [...]

Thank you I downloaded v5.1-rc1 and i can see device listed..
Now trying to build an example i run into all kinds of problems like asking for container location (i already set the folder where in this case the blink example should be set )  and also this
"Unable to determine what CMake generator to use. Please install or configure a preferred generator, or update settings.json, your Kit configuration or PATH variable." message that i tried to solve generating files from CMake extension but not working... do you have any source of guidance in order to accurately build an example w this system?

emotasHil, 2023-06-09:
TWAI is marked as supported but actually only 1 of 2 TWAI controllers is supported.

jack0c, 2023-06-20:
updated coexistence part, sorry since external coexistence is not supported in v5.1

nunojpg, 2023-06-20:
With release-v5.1 or master I see 10-15ºC with the temperature sensor example, at a room at about 25ºC.

AxelLin, 2023-06-21:
> updated coexistence part, sorry since external coexistence is not supported in v5.1

Is there any other progress update for C6?
Last update was 2 Months ago (https://github.com/espressif/esp-idf/issues/10423#issuecomment-1512679789),
I'm surprised the only update here is to mention that external coexistence is not supported.

@igrr

Hedda, 2023-06-21:
@AxelLin Follow the status of ESP-IDF v5.1 as that upcoming will be first  ESP-IDF version to have initial support for the ESP32-C6:

https://github.com/espressif/arduino-esp32/issues/7713

https://github.com/espressif/arduino-esp32/issues/7852

Espressif ESP-IDF v5.1 RC 1 (ESP-IDF version 5.1 Release Candidate 1) was tagged in this GiHub repository just two weeks ago:

https://github.com/espressif/esp-idf/releases

https://github.com/espressif/esp-idf/releases/tag/v5.1-rc1

As both ESP32-C6 and ESP32-H2 are at the very top of the list of major features in that I think that you can safely assume that when the final ESP-IDF v5.1 release (i.e. "GA" release for "General Availability") is released as available then it will be officially supported by Espressif in that ESP-IDF v5.1 release. That is at least the industry standard for most software release life cycles:

https://en.wikipedia.org/wiki/Software_release_life_cycle

[image]

igrr, 2023-06-21:
ADC Calibration is now also supported on ESP32-C6.

The remaining features which aren't yet marked as supported will not be available in IDF v5.1 release. We will update the support plan for the remaining features before the next v5.1.1 release. In the meantime, if your project requires any of the features not listed as supported, please contact Espressif via business support channels.

micbis, 2023-06-21:
Are there any plans to support also the second twai controller any time in the near future?

AxelLin, 2023-06-23:
> With release-v5.1 or master I see 10-15ºC with the temperature sensor example, at a room at about 25ºC.

@nunojpg Please create a new issue to track this.

marciogranzotto, 2023-07-03:
Missing from feature list: `OpenThread Sleepy End Device (SED)`

jack0c, 2023-08-17:
updated:
- [x] External coexistence - Supported
- [x]  BLE Modem Sleep and Light Sleep - Supported

embenix, 2023-09-14:
Sorry if I am repeating the same question as already asked by 2 other guys,
What are your plans for implementing the drivers for 2nd TWAI (`TWAI1`) port on ESP32C6? This could be a breakthrough in my opinion if 2nd TWAI port gets working sooner.

wanckl, 2023-09-19:
@embenix sorry for confusing you, a new suit of TWAI driver is coming, 2nd TWAI will supported on new driver in new version of IDF at end 2023. sorry for long time waiting.

kavehsgh, 2023-09-28:
Hi, Do you have any plan for the coexistence of Wifi-BLE-Zigbee?

IsraelMek, 2023-09-28:
Hi,
Not really. Most likely we will be using the BLE. Since now we are at product development stage, we want to have the option of having more than one communication protocol, in case we want to go for another option. You  are very welcome if you have any suggestion.

Kind regards,
Israel Mekonnen

HX2003, 2023-10-07:
Please refer to espressif/arduino-esp32#7713 regarding the BLE unified provisioning, so that the issue can be fixed. Thanks.

AxelLin, 2023-10-11:
@igrr @jack0c 

Do you have update for the upcoming 5.1.2 release?
Do you have target timeline for feature complete? (Only 6 items are not yet supported)

Massimiliano-solutiontech, 2023-10-30:
Hi,
I would like to know when the support for the `LP UART` will be available because I don't see the code in the `v5.2` but only on the `master` branch.

Thanks

TRusselo, 2023-11-14:
Above it says I2S support is complete, but not working for me in current dev branch
[image]
[image]

gojimmypi, 2023-11-20:
Happy to announce wolfSSL Hardware Acceleration Support for the ESP32-C6!  See [wolfSSL #6990](https://github.com/wolfSSL/wolfssl/pull/6990).  

I also have a [wolfSSL Managed Component](https://components.espressif.com/components/wolfssl/wolfssl) update in the works that will include this change once the PR is merged. The [wolfSSH](https://github.com/wolfSSL/wolfssh/tree/master/ide/Espressif) and [wolfMQTT](https://github.com/wolfSSL/wolfMQTT/tree/master) Espressif Managed components be added soon as well.

jack0c, 2023-11-28:
> @igrr @jack0c
> 
> Do you have update for the upcoming 5.1.2 release? Do you have target timeline for feature complete? (Only 6 items are not yet supported)

The features supported in v5.1.2 were updated.
Here are some comments for the features not supported:
 Broadcast TWT – Not supported (WIFI-5215) - Still WIP, will support in IDF v5.3.
 FTM (Fine Time Measurement) – Not supported (WIFI-5144) - Still WIP, doing an ECO to support it, plan to support in IDF v5.3
 MSPI tuning (Flash & PSRAM DDR and/or over 80 MHz) - Not supported (IDF-5343) - Still WIP, not sure when to release
 SPI Flash auto suspend - Not supported (IDF-5344) - looks done, looks already in master 66bba5694e611980d2671b10ad0839ff1ad6e3e4
 Timeout Module (New) - Not supported (IDF-5814) - No plan yet

Hedda, 2023-11-28:
Could you please add Zigbee NCP Host to the list? See -> https://github.com/espressif/esp-zigbee-sdk/issues/124

txf-, 2024-01-16:
RTC Watchdog issues:

in menuconfig there is this option and description:
[image]

As far as I can tell there `rtc_wdt_disable() `and `rtc_wdt_feed()`, do not apply to anything but the ESP32 and ESP32-S2 as those functions seem to be guarded by ifdefs for those devices.

Will this API ever be supported on other devices?

tom-borcin, 2024-01-17:
Update 2024/01/17: 
- SPI Flash auto suspend - is supported
- Broadcast TWT - is in progress

Only remaining items are:
- FTM (Fine Time Measurement)
- UHCI (UART DMA mode) driver
- MSPI tuning (Flash & PSRAM DDR and/or over 80 MHz)
- Timeout Module (New)

These remaining items will be supported based on customer requests. We are closing the issue and unpinning it. Please open new issue if you have a feature request, question or any other problem.

closed 2024-01-17 by tom-borcin

gojimmypi, 2024-01-17:
Woohoo! Congratulations to everyone that contributed to this massive update for ESP32C6!!

AxelLin, 2024-01-17:
I'm wondering if all the supported features of C6 will be back port to 5.1 and 5.2 releases? (People mainly use release branches for real projects)

Suxsem, 2024-01-17:
Hi, I'm using the relase/5.2 branch and I'm trying to use the ADC_ONESHOT with light sleep.

On boot, I initialize the ADC:
1) adc_oneshot_new_unit
2) adc_oneshot_config_channel
3) adc_cali_create_scheme_curve_fitting

Periodically, it wakes up from deep sleep and read the voltage:
4) adc_oneshot_read
5) adc_cali_raw_to_voltage

Unfortunately, only the first read is correct (in my case 1.8v); reads the occurs after light sleep are totally wrong (4v).

Is calibrated oneshoot adc supported after light sleep 1for esp32-c6?

[11 more replies]