espressif/esp-idf #10424: Multiple characteristics and multiple services in each for ESP32 Dev Kit

meakashrao, 2022-12-21:
### Is your feature request related to a problem?

Hello All,

Is it possible to create Multiple services with Multiple characteristics in each?
I am trying to create 4 characteristics with 3 services in each.

I am trying to merge the Service Table and Gatt_Client_Demo Example. 

Please any suggestions are welcome, please help with the same.

Regards,
Akash Rao

### Describe the solution you'd like.

I'd like an example where users learn to create Multiple services and Multiple characteristics in each service (3 Services with 3 characteristics in each would be great)

### Describe alternatives you've considered.

I tried merging the Gatt_Client_Demo with Gatt_Server_Table_Demo, but I keep encountering runtime bugs and my BLE App wont start and ESP would keep rebooting itself.

xiewenxiang, 2022-12-30:
Here is a simple example of two services that you can refer to for modification.
[ble_ftms.c.zip](https://github.com/espressif/esp-idf/files/10323307/ble_ftms.c.zip)

meakashrao, 2022-12-30:
Thank You, this solves it.

Regards,
Akash Rao

closed 2022-12-30 by meakashrao