espressif/esp-idf #10425: WAKE UP FORM BLE (ESP32C3) WITH EXTERNAL OSCILLATOR

naltamura, 2022-12-21:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

HOW IS IT POSSIBLE TO WAKE UP WITH BLE FROM SLEEP MODE?
I HAVE ALREADY INSTALLED A 32MHZ OSCILLATOR AND CONFIGURED THE SETTINGS IN MENUCONFIG.
I AM NOT USING ARDUINO LIBRARIES BUT JUST PURE ESP32 CODE.
I HAVE AN ESP32C3 BOARD AND MY GOAL IS TO WAKE UP THE BOARD WITH BLE. A GATT SERVER IS IMPLEMENTED ON THE BOARD AND I NEED TO WAKE UP THE BOARD FROM SLEEP LIGTH MODE. HELP ME.
 I GOING TO ATTACH MY CODE.
[image]
[image]
[image]

xiewenxiang, 2022-12-29:
To use bluetooth modem sleep with light sleep, please follow the below instructions.

ESP-IDF menuconfig options:
Enable Power Management:
menuconfig ---> Component config ---> Power management --->
[*] Support for power management

Enable Tickless Idle:
menuconfig ---> Component config ---> FreeRTOS --->
[*] Tickless idle support
(3) Minimum number of ticks to enter sleep mode for (NEW)

Note: Tickless idle needs to be enabled to allow automatic light sleep. FreeRTOS will enter light sleep if no tasks need to run
for 3(by default) ticks, that is 30ms if tick rate is 100Hz. Configure the FreeRTOS tick rate to be higher if you want to allow
shorter duration light sleep, for example:
menuconfig ---> Component config ---> FreeRTOS ->
(1000) Tick rate (Hz)

Configure external 32.768Hz crystal as RTC clock source:
menuconfig ---> Component config ---> ESP32-specific --->
RTC clock source (External 32kHz crystal)
[*] Additional current for external 32kHz crystal
Note that the "additional current" option is a workaround for a hardware issue on ESP32 that the crystal can fail in oscillating.
Please enable this option when you use external 32kHz crystal. This hardware issue will be resolved in the next ECO chip.

Enable Bluetooth modem sleep with external 32.768kHz crystal as low power clock:
menuconfig ---> Component config ---> Bluetooth ---> Bluetooth controller ---> MODEM SLEEP Options --->
[*] Bluetooth modem sleep
Bluetooth Modem sleep mode (ORIG mode(sleep with low power clock))
Bluetooth low power clock (External 32kHz crystal)

naltamura, 2022-12-29:
Thanks but I already followed these instructions.
The wakeup seems enabled but the current consumption (medured) is around
0.15mA when the adv is off
I wanna that the device have to wake up only when second board send it a
message. The use case is : monitoring temperature at defined intervals in
light sleep and the device has to be wake up on request for external device.

xiewenxiang, 2022-12-29:
There are many ways to wake up the chip. Please reference this example.
https://github.com/espressif/esp-idf/tree/master/examples/system/light_sleep

naltamura, 2022-12-29:
I already seen also these examples and there aren't cases about BLE wake up...
It is needed call esp_light_sleep_start(); in BLE controller ?

This is my configuration but BLE seems always UP.  

esp_pm_config_esp32c3_t pm_config = {
.max_freq_mhz = 80, // e.g. 80, 160, 240
.min_freq_mhz = 40, // e.g. 40
.light_sleep_enable = true, // enable light sleep
};
esp_err_t err=ESP_FAIL;

ESP_ERROR_CHECK(esp_pm_configure(&pm_config));

esp-cjh, 2024-02-27:
Hi, @naltamura BLE will enter modem sleep only when the interval between BLE events is greater than 7.5ms.
So you should make sure that:
1 adv interval is greater than 7.5ms. 
2 (scan interval - scan window) is greater than 7.5ms.
3  connection interval is greater than 7.5ms.

closed 2024-12-17 by Alvin1Zhang