espressif/esp-idf #10426: Debugger Error: Target not halted

ddomnik, 2022-12-21:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

esp-idf-v5.0 ((v5.0))

### Operating System used.

Windows

### How did you build your project?

Eclipse IDE

### If you are using Windows, please specify command line type.

PowerShell

### What is the expected behavior?

Running debug mode

### What is the actual behavior?

[image]

### Steps to reproduce.

The debugger has worked before (reinstall did not help)

1. Using ESP-PROG
2. Connect all wires
3. Replacing FTDIBUS with WinUSB 
[image]
5. Setup debugger configuration 
[image]

Additional information:

If the monitor is connected while trying to run debug mode, the OpenOCD times out while starting and the UART logs display the following:
```
...
[normal UART logs]
...
[after starting debugger via eclipse]:
Guru Meditation Error: Core  0 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x403b2468: 00000000 00000000 00000000
Setting breakpoint at 0x403b246c and returning...
Guru Meditation Error: Core  0 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x403b2468: 00000000 00000000 00000000
Setting breakpoint at 0x403b246c and returning...
Guru Meditation Error: Core  0 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x403b2468: 00000000 00000000 00000000
Setting breakpoint at 0x403b246c and returning...
Guru Meditation Error: Core  0 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x403b2468: 00000000 00000000 00000000
Setting breakpoint at 0x403b246c and returning...
Guru Meditation Error: Core  0 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x403b2468: 00000000 00000000 00000000
Setting breakpoint at 0x403b246c and returning...
Guru Meditation Error: Core  0 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x403b2468: 00000000 00000000 00000000
Setting breakpoint at 0x403b246c and returning...
Guru Meditation Error: Core  0 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x403b2468: 00000000 00000000 00000000
Setting breakpoint at 0x403b246c and returning...
Guru Meditation Error: Core  0 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x403b2468: 00000000 00000000 00000000
Setting breakpoint at 0x403b246c and returning...```

kolipakakondal, 2022-12-22:
> If the monitor is connected while trying to run debug mode, the OpenOCD times out while starting and the UART logs display the following:

Hi @ddomnik  I am sorry that it's not working. What happens if you close the serial monitor and start the OpenOCDDebugging, does it work?

I think we come across a similar [issue](https://github.com/espressif/idf-eclipse-plugin/issues/638), [this](https://github.com/espressif/idf-eclipse-plugin/pull/661) might fix the problem. Keep you posted as soon as it's out for release.

ddomnik, 2023-01-02:
> Hi @ddomnik I am sorry that it's not working. What happens if you close the serial monitor and start the OpenOCDDebugging, does it work?
>
> I think we come across a similar [...]

Thank you for the support, I still have the same issue after installing the update.

Without active serial monitor:
```Open On-Chip Debugger v0.11.0-esp32-20221026 (2022-10-26-14:48)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : ftdi: if you experience problems at higher adapter clocks, try the command "ftdi tdo_sample_edge falling"
Info : clock speed 20000 kHz
Info : JTAG tap: esp32s3.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32s3.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : starting gdb server for esp32s3.cpu0 on 3333
Info : Listening on port 3333 for gdb connections
Info : [esp32s3.cpu0] Target halted, PC=0x403B0CE7, debug_reason=00000000
Info : [esp32s3.cpu0] Reset cause (3) - (Software core reset)
Info : [esp32s3.cpu1] Target halted, PC=0x40000400, debug_reason=00000000
Info : [esp32s3.cpu1] Reset cause (3) - (Software core reset)
Info : Set GDB target to 'esp32s3.cpu1'
Error: Target not halted
```

with active serial monitor:
```Open On-Chip Debugger v0.11.0-esp32-20221026 (2022-10-26-14:48)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : ftdi: if you experience problems at higher adapter clocks, try the command "ftdi tdo_sample_edge falling"
Info : clock speed 20000 kHz
Info : JTAG tap: esp32s3.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32s3.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : starting gdb server for esp32s3.cpu0 on 3333
Info : Listening on port 3333 for gdb connections
Info : [esp32s3.cpu0] Debug controller was reset.
Info : [esp32s3.cpu0] Core was reset.
Info : [esp32s3.cpu1] Debug controller was reset.
Info : [esp32s3.cpu1] Core was reset.
Info : [esp32s3.cpu1] Target halted, PC=0x4037FCA2, debug_reason=00000000
Info : [esp32s3.cpu1] Reset cause (1) - (Power on reset)
Info : Set GDB target to 'esp32s3.cpu1'
Info : [esp32s3.cpu0] Target halted, PC=0x4037FCA2, debug_reason=00000000
Info : [esp32s3.cpu0] Reset cause (1) - (Power on reset)
Info : [esp32s3.cpu0] Target halted, PC=0x403B246C, debug_reason=00000001
Error: Failed to get flash maps (5)!
Warn : Failed to get flash mappings (-4)!
Info : [esp32s3.cpu0] Target halted, PC=0x403B246C, debug_reason=00000001
Info : Using flash bank 'esp32s3.cpu0.irom' size 0 KB
Info : [esp32s3.cpu0] Target halted, PC=0x403B246C, debug_reason=00000001
Error: Failed to get flash maps (5)!
Warn : Failed to get flash mappings (-4)!
Info : [esp32s3.cpu0] Target halted, PC=0x403B246C, debug_reason=00000001
Info : Using flash bank 'esp32s3.cpu0.drom' size 0 KB
Info : [esp32s3.cpu0] Target halted, PC=0x403B246C, debug_reason=00000001
Error: Failed to get flash maps (5)!
Warn : Failed to get flash mappings (-4)!
Info : [esp32s3.cpu0] Target halted, PC=0x403B246C, debug_reason=00000001
Info : Using flash bank 'esp32s3.cpu1.irom' size 0 KB
Info : [esp32s3.cpu0] Target halted, PC=0x403B246C, debug_reason=00000001
Error: Failed to get flash maps (5)!
Warn : Failed to get flash mappings (-4)!
Info : [esp32s3.cpu0] Target halted, PC=0x403B246C, debug_reason=00000001
Info : Using flash bank 'esp32s3.cpu1.drom' size 0 KB
Flashing C:/Users/DominikDufner/Documents/esp/ksc-2n/build/ksc2n_Appl.bin at 0x10000
Info : JTAG tap: esp32s3.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32s3.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : [esp32s3.cpu0] requesting target halt and executing a soft reset
Info : [esp32s3.cpu0] Target halted, PC=0x4200414F, debug_reason=00000000
[... 100 lines cut ...]
Info : [esp32s3.cpu1] Target halted, PC=0x500000EF, debug_reason=00000000
Info : [esp32s3.cpu1] Reset cause (3) - (Software core reset)
Info : [esp32s3.cpu1] requesting target halt and executing a soft reset
Info : [esp32s3.cpu0] Core was reset.
Info : [esp32s3.cpu0] Target halted, PC=0x40000400, debug_reason=00000000
Info : [esp32s3.cpu1] Core was reset.
Info : [esp32s3.cpu1] Target halted, PC=0x40000400, debug_reason=00000000
Info : [esp32s3.cpu1] Reset cause (3) - (Software core reset)
Info : [esp32s3.cpu0] Reset cause (3) - (Software core reset)
Warn : address 0x4200dca7 not writable
Error: [esp32s3.cpu0] Error writing instruction to memory: -4
Error: [esp32s3.cpu0] Failed to write breakpoint instruction (-4)!
Error: [esp32s3.cpu0] Failed to add SW breakpoint!
Error: can't add breakpoint: unknown reason
Warn : address 0x42012a90 not writable
Error: [esp32s3.cpu0] Error writing instruction to memory: -4
Error: [esp32s3.cpu0] Failed to write breakpoint instruction (-4)!
Error: [esp32s3.cpu0] Failed to add SW breakpoint!
Error: can't add breakpoint: unknown reason
Warn : address 0x42034867 not writable
Error: [esp32s3.cpu0] Error writing instruction to memory: -4
Error: [esp32s3.cpu0] Failed to write breakpoint instruction (-4)!
Error: [esp32s3.cpu0] Failed to add SW breakpoint!
Error: can't add breakpoint: unknown reason
```

ddomnik, 2023-02-02:
@kolipakakondal are there any updates? I am still facing the issues 2-4. This issues were not present in 01d014c42def8d0c19e1ce55c07de6761e092ffa