espressif/esp-idf #10427: undefined reference to `_bt_bss_start'

sidwarkd, 2022-12-21:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4.3-316-ge86181704a

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

I expect incremental builds to work as well as see the same behavior from one build to the next.

### What is the actual behavior?

When doing an incremental build I will get the following output from the link portion of the build 100% of the time
```txt
/opt/esp/tools/xtensa-esp32-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/bt/libbt.a(bt.c.obj):(.literal.esp_bt_mem_release+0x0): undefined reference to `_bt_bss_start'
/opt/esp/tools/xtensa-esp32-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: esp-idf/bt/libbt.a(bt.c.obj):(.literal.esp_bt_mem_release+0x4): undefined reference to `_bt_bss_end'
```

I then have to run `idf.py fullclean` and do a build from scratch and it will succeed. Sometimes even after running `fullclean` I will get the same error. I run `fullclean` again, build again and everything compiles and links properly.

I recently migrated from 4.4.1 to 4.4.3. I never experienced this issue on 4.4.1 and have not changed anything in our codebase surrounding the bluetooth functionality. From the code it looks like the error is generated from the statements [here](https://github.com/espressif/esp-idf/blob/release/v4.4/components/bt/controller/esp32/bt.c#L244) but the linker can't find definitions for them.

The weird thing is that it works by simply doing a `fullclean` and building again....most of the time. That is from one build to the next without changing a line of code.

### Steps to reproduce.

I will try to create a repo that reproduces this issue but wanted to get the issue posted to see if the team has any insight. I, unfortunately, cannot share the code due to an NDA but will attempt to recreate it with a bare bones project.

o-marshmallow, 2022-12-23:
Hello @sidwarkd ,

Thank you very much for the example and the explanation, it let me reproduce the issue easily on my side.

It seems like `bt` component's linker fragment is unstable when `ESP_ALLOW_BSS_SEG_EXTERNAL_MEMORY` is enabled, which is your case. It doesn't generate the proper linker symbols. Can you please try this fix on your end:
[bt_linker.diff.txt](https://github.com/espressif/esp-idf/files/10294016/bt_linker.diff.txt)

You can apply it with the following commands:
```
cd $IDF_PATH
patch -p1 < /path/to/bt_linker.diff.txt
```

Please have a try and tell me if this fixes your issue (and works properly in your case). If it does, I will create an internal merge request to patch IDF versions

sidwarkd, 2022-12-23:
@o-marshmallow Thank you for the quick reply. I tested this and was able to build and link without any issues. I did several full builds as well as incremental builds and no longer see the error. I also did a functional test of the firmware using bluetooth and all worked as expected.

o-marshmallow, 2022-12-23:
@sidwarkd Thank you for the quick feedback! I have created an internal merge request to make this fix permanent in IDF versions.

sidwarkd, 2022-12-23:
@o-marshmallow will this be in the 4.4 release as well as the stable 5.0?

o-marshmallow, 2022-12-23:
@sidwarkd As soon as it is merged, it will be backported to `release/v5.0`. I also plan to backport it to `release/v4.4`. I will keep you informed

xanthome, 2023-02-23:
on a clean install of esp-idf 5.0.1 windows I get this error of the esp-box factory demo example.

If I apply the patch above and flash the esp32-s3 box and sample will be stuck in a reboot loop panicking on some bluetooth code.

should I remove CONFIG_ESP_ALLOW_BSS_SEG_EXTERNAL_MEMORY? Doing fullclean at the moment I have to fullclean and build and once in a while it will succeed.

-- Adding linker script C:/Espressif/frameworks/esp-idf-v5.0.1/components/bootloader/subproject/main/ld/esp32s3/bootloader.rom.ld
-- Components: bootloader bootloader_support efuse esp_app_format esp_common esp_hw_support esp_rom esp_system esptool_py freertos hal log main micro-ecc newlib partition_table soc spi_flash xtensa
-- Component paths: C:/Espressif/frameworks/esp-idf-v5.0.1/components/bootloader C:/Espressif/frameworks/esp-idf-v5.0.1/components/bootloader_support C:/Espressif/frameworks/esp-idf-v5.0.1/components/efuse C:/Espressif/frameworks/esp-idf-v5.0.1/components/esp_app_format C:/Espressif/frameworks/esp-idf-v5.0.1/components/esp_common C:/Espressif/frameworks/esp-idf-v5.0.1/components/esp_hw_support C:/Espressif/frameworks/esp-idf-v5.0.1/components/esp_rom C:/Espressif/frameworks/esp-idf-v5.0.1/components/esp_system C:/Espressif/frameworks/esp-idf-v5.0.1/components/esptool_py C:/Espressif/frameworks/esp-idf-v5.0.1/components/freertos C:/Espressif/frameworks/esp-idf-v5.0.1/components/hal C:/Espressif/frameworks/esp-idf-v5.0.1/components/log C:/Espressif/frameworks/esp-idf-v5.0.1/components/bootloader/subproject/main C:/Espressif/frameworks/esp-idf-v5.0.1/components/bootloader/subproject/components/micro-ecc C:/Espressif/frameworks/esp-idf-v5.0.1/components/newlib C:/Espressif/frameworks/esp-idf-v5.0.1/components/partition_table C:/Espressif/frameworks/esp-idf-v5.0.1/components/soc C:/Espressif/frameworks/esp-idf-v5.0.1/components/spi_flash C:/Espressif/frameworks/esp-idf-v5.0.1/components/xtensa
-- Configuring done
-- Generating done
-- Build files have been written to: C:/dev/esp-box/examples/factory_demo/build/bootloader
[105/106] Generating binary image from built executableesptool.py v4.5rmat.a←[Kort.a←[Kj←[Kt.c.obj←[Kj←[KKt.c.obj←[KK←[Ke_api_key.c.obj←[K_esp32s3.c.obj←[K
Creating esp32s3 image...
Merged 1 ELF section
Successfully created esp32s3 image.
Generated C:/dev/esp-box/examples/factory_demo/build/bootloader/bootloader.bin
[106/106] cmd.exe /C "cd /D C:\dev\esp-box\examples\factory_demo\build\bootloader\esp-idf\esptool_p...-offset 0x8000 bootloader 0x0 C:/dev/esp-box/examples/factory_demo/build/bootloader/bootloader.bin"Bootloader binary size 0x55f0 bytes. 0x2a10 bytes (33%) free.
[1585/1587] Linking CXX executable factory_demo.elf←[31mFAILED: ←[0mfactory_demo.elf .c.obj←[K
cmd.exe /C "cd . && C:\Espressif\tools\xtensa-esp32s3-elf\esp-2022r1-11.2.0\xtensa-esp32s3-elf\bin\xtensa-esp32s3-elf-g++.exe -mlongcalls  @CMakeFiles\factory_demo.elf.rsp -o factory_demo.elf  && cd ."
c:/espressif/tools/xtensa-esp32s3-elf/esp-2022r1-11.2.0/xtensa-esp32s3-elf/bin/../lib/gcc/xtensa-esp32s3-elf/11.2.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: esp-idf/bt/libbt.a(bt.c.obj):(.literal.esp_bt_mem_release+0x0): undefined reference to `_bt_bss_end'
c:/espressif/tools/xtensa-esp32s3-elf/esp-2022r1-11.2.0/xtensa-esp32s3-elf/bin/../lib/gcc/xtensa-esp32s3-elf/11.2.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: esp-idf/bt/libbt.a(bt.c.obj):(.literal.esp_bt_mem_release+0x8): undefined reference to `_bt_bss_start'
collect2.exe: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
ninja failed with exit code 1, output of the command is in the c:\dev\esp-box\examples\factory_demo\build\log\idf_py_stderr_output_928 and c:\dev\esp-box\examples\factory_demo\build\log\idf_py_stdout_output_928

xanthome, 2023-02-23:
the error that happens when it's stuck in a reboot loop. After I've added the device to rainmaker, given it wifi credentials and the ESP32-S3-BOX wants to boot up again.

I (20417) esp_rmaker_system_service: System service enabled.
I (20418) esp_rmaker_core: Starting RainMaker Work Queue task
I (20418) esp_rmaker_work_queue: RainMaker Work Queue task started.
I (20419) wifi_prov_scheme_ble: BT memory released
I (20419) app_wifi: Already provisioned, starting Wi-Fi STA
ESP_ERROR_CHECK failed: esp_err_t 0xffffffff at 0x420831fb
0x420831fb: esp_bt_mem_release at C:/Espressif/frameworks/esp-idf-v5.0.1/components/bt/controller/esp32s3/bt.c:1011 (discriminator 1)

file: "/IDF/components/bt/controller/esp32s3/bt.c" line 1011
func: esp_bt_mem_release
expression: try_heap_caps_add_region(mem_start, mem_end)

abort() was called at PC 0x40387eff on core 0
0x40387eff: _esp_error_check_failed at C:/Espressif/frameworks/esp-idf-v5.0.1/components/esp_system/esp_err.c:47

Backtrace: 0x4037a3ca:0x3fcc1590 0x40387f09:0x3fcc15b0 0x4038d12a:0x3fcc15d0 0x40387eff:0x3fcc1640 0x420831fb:0x3fcc1670 0x42072d4d:0x3fcc1690 0x42072183:0x3fcc16b0 0x4200e50d:0x3fcc16d0 0x4200d347:0x3fcc1770 0x403883f5:0x3fcc17a0
0x4037a3ca: panic_abort at C:/Espressif/frameworks/esp-idf-v5.0.1/components/esp_system/panic.c:423

0x40387f09: esp_system_abort at C:/Espressif/frameworks/esp-idf-v5.0.1/components/esp_system/esp_system.c:153

0x4038d12a: abort at C:/Espressif/frameworks/esp-idf-v5.0.1/components/newlib/abort.c:38

0x40387eff: _esp_error_check_failed at C:/Espressif/frameworks/esp-idf-v5.0.1/components/esp_system/esp_err.c:47

0x420831fb: esp_bt_mem_release at C:/Espressif/frameworks/esp-idf-v5.0.1/components/bt/controller/esp32s3/bt.c:1011 (discriminator 1)

0x42072d4d: wifi_prov_scheme_ble_event_cb_free_btdm at C:/Espressif/frameworks/esp-idf-v5.0.1/components/wifi_provisioning/src/scheme_ble.c:212

0x42072183: wifi_prov_mgr_deinit at C:/Espressif/frameworks/esp-idf-v5.0.1/components/wifi_provisioning/src/manager.c:1396

0x4200e50d: app_wifi_start at C:/dev/esp-box/examples/factory_demo/main/app/app_wifi.c:248 (discriminator 13)

0x4200d347: rmaker_task at C:/dev/esp-box/examples/factory_demo/main/rmaker/app_rmaker.c:175 (discriminator 2)

0x403883f5: vPortTaskWrapper at C:/Espressif/frameworks/esp-idf-v5.0.1/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:154

ELF file SHA256: 6f196fd4842b4c62

Rebooting...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x3 (RTC_SW_SYS_RST),boot:0xa (SPI_FAST_FLASH_BOOT)
Saved PC:0x40379ced

xanthome, 2023-03-09:
I have to just try, try and try over and over to get it to compile. I'm making no real changes from failure case to success case here. 
I'll press spacebar, then delete it again.

`PS C:\dev\esp-box\examples\factory_demo> idf.py app-flash monitor
Executing action: app-flash
Serial port COM4
Connecting...
Detecting chip type... ESP32-S3
Running ninja in directory c:\dev\esp-box\examples\factory_demo\build
Executing "ninja app-flash"...
[1/7] Building C object esp-idf/main/CMakeFiles/__idf_main.dir/gui/ui_device_ctrl.c.obj
[2/7] Linking C static library esp-idf\main\libmain.a
[3/7] Generating ld/sections.ld
[4/7] Linking CXX executable factory_demo.elf
FAILED: factory_demo.elf
cmd.exe /C "cd . && C:\Espressif\tools\xtensa-esp32s3-elf\esp-2022r1-11.2.0\xtensa-esp32s3-elf\bin\xtensa-esp32s3-elf-g++.exe -mlongcalls  @CMakeFiles\factory_demo.elf.rsp -o factory_demo.elf  && cd ."
c:/espressif/tools/xtensa-esp32s3-elf/esp-2022r1-11.2.0/xtensa-esp32s3-elf/bin/../lib/gcc/xtensa-esp32s3-elf/11.2.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: esp-idf/bt/libbt.a(bt.c.obj):(.literal.esp_bt_mem_release+0x0): undefined reference to `_bt_bss_end'
c:/espressif/tools/xtensa-esp32s3-elf/esp-2022r1-11.2.0/xtensa-esp32s3-elf/bin/../lib/gcc/xtensa-esp32s3-elf/11.2.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: esp-idf/bt/libbt.a(bt.c.obj):(.literal.esp_bt_mem_release+0x8): undefined reference to `_bt_bss_start'
collect2.exe: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
ninja failed with exit code 1, output of the command is in the c:\dev\esp-box\examples\factory_demo\build\log\idf_py_stderr_output_26676 and c:\dev\esp-box\examples\factory_demo\build\log\idf_py_stdout_output_26676`

then magically, it will compile. And it will also not crash freeing bt memory.

`PS C:\dev\esp-box\examples\factory_demo> idf.py app-flash monitor
Executing action: app-flash
Serial port COM4
Connecting...
Detecting chip type... ESP32-S3
Running ninja in directory c:\dev\esp-box\examples\factory_demo\build
Executing "ninja app-flash"...
[1/7] Building C object esp-idf/main/CMakeFiles/__idf_main.dir/gui/ui_device_ctrl.c.obj
[2/7] Linking C static library esp-idf\main\libmain.a
[3/7] Generating ld/sections.ld
[4/7] Linking CXX executable factory_demo.elf
[5/7] Generating binary image from built executable
esptool.py v4.5
Creating esp32s3 image...
Merged 2 ELF sections
Successfully created esp32s3 image.
Generated C:/dev/esp-box/examples/factory_demo/build/factory_demo.bin
[6/7] cmd.exe /C "cd /D C:\dev\esp-box\examples\factory_demo\build\esp-idf\esptool_py && C:\Espressif\python_env\idf5.0_py3.8_env\Scripts\python.exe C:/Espressif/frameworks/esp-idf-v5.0.1/components/partition_table/check_sizes.py --offset 0x8000 partition --type app C:/dev/esp-box/examples/factory_demo/build/partition_table/partition-table.bin C:/dev/esp-box/examples/factory_demo/build/factory_demo.bin"
factory_demo.bin binary size 0x1c6390 bytes. Smallest app partition is 0x41a000 bytes. 0x253c70 bytes (57%) free.
[6/7] cmd.exe /C "cd /D C:\Espressif\frameworks\esp-idf-v5.0.1\components\esptool_py && C:\Espressif\tools\cmake\3.24.0\bin\cmake.exe -D IDF_PATH=C:/Espressif/frameworks/esp-idf-v5.0.1 -D SERIAL_TOOL=C:/Espressif/python_env/idf5.0_py3.8_env/Scripts/python.exe;;C:/Espressif/frameworks/esp-idf-v5.0.1/components/esptool_py/esptool/esptool.py;--chip;esp32s3 -D SERIAL_TOOL_ARGS=--before=default_reset;--after=hard_reset;write_flash;@app-flash_args -D WORKING_DIRECTORY=C:/dev/esp-box/examples/factory_demo/build -P C:/Espressif/frameworks/esp-idf-v5.0.1/components/esptool_py/run_serial_tool.cmake"
esptool esp32s3 -p COM4 -b 460800 --before=default_reset --after=hard_reset write_flash --flash_mode dio --flash_freq 80m --flash_size 16MB 0x20000 factory_demo.bin
esptool.py v4.5
Serial port COM4
Connecting...
Chip is ESP32-S3 (revision v0.1)
Features: WiFi, BLE
Crystal is 40MHz
MAC: 7c:df:a1:e7:a5:94
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Flash will be erased from 0x00020000 to 0x001e6fff...
Compressed 1860496 bytes to 1080709...
Writing at 0x00020000... (1 %)`

tttonyyy, 2023-03-15:
> @sidwarkd As soon as it is merged, it will be backported to `release/v5.0`. I also plan to backport it to `release/v4.4`. I will keep you informed

Hi @o-marshmallow I am encountering the same issue - did a PR get raised for this change?  It looks like 4.4.4/5.01/master don't have changes like in your patch.

Alternatively is there a way I can associate a linker fragment with my project as an interim measure, to avoid patching the esp-idf toolchain?

Here I see significant internal memory free'd between 4.4.2 and 4.4.4 so it is highly desirable for us to move to 4.4.4.

tttonyyy, 2023-03-15:
I have created a top-level linker fragment that looks like this:

```
# This fixes an ESP-IDF 4.4.4/5.0.1 linker issue on incremental builds
#   "undefined reference to '_bt_bss_start'" for Bluetooth component library function 'esp_bt_mem_release'
# References:
#   https://github.com/espressif/esp-idf/issues/10427
#   https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/linker-script-generation.html

[sections:bt_bss_fix]
entries:
    .bss+

[scheme:bt_extram_bss_fix]
entries:
    bt_bss_fix -> extern_ram

[mapping:bt_fix]
archive: libbt.a
entries:
    if ESP_ALLOW_BSS_SEG_EXTERNAL_MEMORY = y:
        esp_bt_mem_release (bt_extram_bss_fix);
            bt_bss_fix -> extern_ram ALIGN(4) ALIGN(4, post) SURROUND(bt_bss)
```

I have added it using LDFRAGMENTS in main/CMakeLists.txt

This seems to fix the problem of the missing _bt_bss_start and _bt_bss_end for this symbol as an interim workaround; ESP-IDF remains unmodified.

xanthome, 2023-03-15:
I get the same reboot loop when using with the factory demo for esp-box:

ESP_ERROR_CHECK failed: esp_err_t 0xffffffff at 0x42062df3
0x42062df3: esp_bt_mem_release at C:/Espressif/frameworks/esp-idf-v5.0.1/components/bt/controller/esp32s3/bt.c:1011 (discriminator 1)

file: "/IDF/components/bt/controller/esp32s3/bt.c" line 1011
func: esp_bt_mem_release
expression: try_heap_caps_add_region(mem_start, mem_end)

abort() was called at PC 0x4038703b on core 0
0x4038703b: _esp_error_check_failed at C:/Espressif/frameworks/esp-idf-v5.0.1/components/esp_system/esp_err.c:47

Backtrace: 0x4037a36a:0x3fcb3030 0x40387045:0x3fcb3050 0x4038c266:0x3fcb3070 0x4038703b:0x3fcb30e0 0x42062df3:0x3fcb3110 0x42045df9:0x3fcb3130 0x4204522f:0x3fcb3150 0x4200cac1:0x3fcb3170 0x4200b12f:0x3fcb3210 0x40387531:0x3fcb3240
0x4037a36a: panic_abort at C:/Espressif/frameworks/esp-idf-v5.0.1/components/esp_system/panic.c:423

0x40387045: esp_system_abort at C:/Espressif/frameworks/esp-idf-v5.0.1/components/esp_system/esp_system.c:153

0x4038c266: abort at C:/Espressif/frameworks/esp-idf-v5.0.1/components/newlib/abort.c:38

0x4038703b: _esp_error_check_failed at C:/Espressif/frameworks/esp-idf-v5.0.1/components/esp_system/esp_err.c:47

0x42062df3: esp_bt_mem_release at C:/Espressif/frameworks/esp-idf-v5.0.1/components/bt/controller/esp32s3/bt.c:1011 (discriminator 1)

0x42045df9: wifi_prov_scheme_ble_event_cb_free_btdm at C:/Espressif/frameworks/esp-idf-v5.0.1/components/wifi_provisioning/src/scheme_ble.c:212

0x4204522f: wifi_prov_mgr_deinit at C:/Espressif/frameworks/esp-idf-v5.0.1/components/wifi_provisioning/src/manager.c:1396

0x4200cac1: app_wifi_start at C:/dev/esp-box/examples/factory_demo/main/app/app_wifi.c:256 (discriminator 13)

0x4200b12f: rmaker_task at C:/dev/esp-box/examples/factory_demo/main/rmaker/app_rmaker.c:146 (discriminator 2)

0x40387531: vPortTaskWrapper at C:/Espressif/frameworks/esp-idf-v5.0.1/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:154

o-marshmallow, 2023-03-16:
Hi folks,
I reported the issue to the team responsible for this component. Further investigations are needed to make sure that BSS can indeed be put in external RAM for this component.

If you are having problems at link time, can you try removing the following lines from the `component/bt/linker.lf` file:
```
    if ESP_ALLOW_BSS_SEG_EXTERNAL_MEMORY = y:
        * (extram_bss)
```
and try compiling again.

xanthome, 2023-03-16:
thank you very much! commenting out those two lines in `components/bt/linker.lf` fixed the linker issue for me and the esp-box factory demo example works correctly as well.

0x0fe, 2023-08-03:
@o-marshmallow 
Hello, i am encountering this issue too, can you confirm commenting these two line will still allow the BT to link to SPIRAM? In our case it is mandatory to link to the SPIRAM  (we use BT, BLE and wifi at same time).

0x0fe, 2023-08-03:
@tttonyyy 
I tried your method, in platformio, however it does solve the error, do you know what could be wrong?

i add the **LDFRAGMENTS "bt_link.lf**" to the already existing **idf_component_register(SRCS ${app_sources})** in src/CMakeLists.txt 

`idf_component_register(SRCS ${app_sources} LDFRAGMENTS "bt_link.lf")`

and i created bt_link.lf in src/ with the content you provided.

lnayman, 2024-01-10:
Hi guys. I have also been using the patch from https://github.com/Eltako/esp-idf/commit/c886e1578297faedcf00b9378f734fc0e1c4a6a0 and wondering when this will make it into esp-idf repo?

wmy-espressif, 2024-03-04:
Hi @AntoineSX   @lnayman  Sorry for the late response. We are considering a fix for this linker issue which will support the option `ESP_ALLOW_BSS_SEG_EXTERNAL_MEMORY` as well as the API `esp_bt_mem_release` considering all of the Espressif SoCs in esp-idf, which will be different from the patch you are using. Hopefully the formal fix can be provided before the end of this month.

Also thanks for your bug report @AntoineSX , I will evaluate the issue you pointed and provide my feedback later on.

wmy-espressif, 2024-04-01:
Hi @AntoineSX , thanks for your [reported issue](https://github.com/espressif/esp-idf/issues/10427#issuecomment-1917459707), I have added the fix of this issue in my proposed formal fix which is being reviewed.

closed 2024-04-17 by espressif-bot

wmy-espressif, 2024-04-22:
The fix for this issue is merged in to master branch of esp-idf in [commit d4552c62](https://github.com/espressif/esp-idf/commit/d4552c6274ac8119acef610b34276fe56dc2a4fd).

And the fix will be backported to other release branches soon.

michaelboeding, 2024-04-23:
Timeline for a release on 5.2.1? Should I just move to the master? @wmy-espressif

wmy-espressif, 2024-04-23:
Hi @michaelboeding , backport for branch "release/v5.2 " is ongoing, the fix is expected to be included in v5.2.2. Here is the patch if you use latest commit branch release/v5.2: 
[patch_for_v5.2.tar.gz](https://github.com/espressif/esp-idf/files/15075890/patch_for_v5.2.tar.gz)

michaelboeding, 2024-04-23:
I may be doing something wrong here but I moved the file and extracted it to ~/esp/esp-idf/

Then I run the `git apply patch_for_v5.2/*.patch`

(base) macbook:esp-idf michaelboeding$ git apply patch_for_v5.2/*.patch
error: patch failed: components/bt/CMakeLists.txt:72
error: components/bt/CMakeLists.txt: patch does not apply
error: patch failed: components/bt/controller/esp32/bt.c:233
error: components/bt/controller/esp32/bt.c: patch does not apply
(base) macbook:esp-idf michaelboeding$ 

git status shows im on  HEAD detached at v5.2.1

michaelboeding, 2024-04-23:
I got the patch applied by pulling the latest 5.2 branch, but now im getting new linked error see below.

`bash/Users/michaelboeding/esp/esp-idf/tools/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: esp-idf/bt/libbt.a(bt.c.obj):(.literal.patch_apply+0x0): undefined reference to `config_btdm_funcs_reset'
/Users/michaelboeding/esp/esp-idf/tools/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: esp-idf/bt/libbt.a(bt.c.obj):(.literal.patch_apply+0x4): undefined reference to `config_ble_funcs_reset'
/Users/michaelboeding/esp/esp-idf/tools/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: esp-idf/bt/libbt.a(bt.c.obj): in function `patch_apply':
/Users/michaelboeding/esp/esp-idf/components/bt/controller/esp32/bt.c:1714:(.text.patch_apply+0x3): undefined reference to `config_btdm_funcs_reset'
/Users/michaelboeding/esp/esp-idf/tools/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: /Users/michaelboeding/esp/esp-idf/components/bt/controller/esp32/bt.c:1721:(.text.patch_apply+0x9): undefined reference to `config_ble_funcs_reset'
/Users/michaelboeding/esp/esp-idf/tools/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: esp-idf/wpa_supplicant/libwpa_supplicant.a(esp_wpa_main.c.obj):(.literal.wpa_sta_connect+0x0): undefined reference to `esp_wifi_sta_connect_internal'
/Users/michaelboeding/esp/esp-idf/tools/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: esp-idf/wpa_supplicant/libwpa_supplicant.a(esp_wpa_main.c.obj): in function `wpa_sta_connect':
/Users/michaelboeding/esp/esp-idf/components/wpa_supplicant/esp_supplicant/src/esp_wpa_main.c:213:(.text.wpa_sta_connect+0x32): undefined reference to `esp_wifi_sta_connect_internal'
/Users/michaelboeding/esp/esp-idf/tools/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: esp-idf/wpa_supplicant/libwpa_supplicant.a(esp_wpas_glue.c.obj):(.literal.wpa_supplicant_transition_disable+0x0): undefined reference to `esp_wifi_enable_sae_pk_only_mode_internal'
/Users/michaelboeding/esp/esp-idf/tools/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: esp-idf/wpa_supplicant/libwpa_supplicant.a(esp_wpas_glue.c.obj):(.literal.wpa_supplicant_transition_disable+0x4): undefined reference to `esp_wifi_sta_disable_owe_trans_internal'
/Users/michaelboeding/esp/esp-idf/tools/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: esp-idf/wpa_supplicant/libwpa_supplicant.a(esp_wpas_glue.c.obj): in function `wpa_supplicant_transition_disable':
/Users/michaelboeding/esp/esp-idf/components/wpa_supplicant/esp_supplicant/src/esp_wpas_glue.c:114:(.text.wpa_supplicant_transition_disable+0x27): undefined reference to `esp_wifi_enable_sae_pk_only_mode_internal'
/Users/michaelboeding/esp/esp-idf/tools/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: /Users/michaelboeding/esp/esp-idf/components/wpa_supplicant/esp_supplicant/src/esp_wpas_glue.c:123:(.text.wpa_supplicant_transition_disable+0x48): undefined reference to `esp_wifi_sta_disable_owe_trans_internal'`

Any idea what caused this?

wmy-espressif, 2024-04-25:
Hi @michaelboeding  I think the cause is that you don't update the git submodules for esp-idf. Please use `git submodule update --init` in your esp-idf directory.

wmy-espressif, 2024-05-06:
Fix for branch release/v5.2 is merged in commit [dc3e9531](https://github.com/espressif/esp-idf/commit/dc3e9531c0333652aea236f5cfb663d707c460c8).

Fixes for release/v4.4, release/v5.0, release/v5.1 are queuing to be merged.

Here is the patch for v4.4.7: 
[patch_for_v4.4.7.tar.gz](https://github.com/espressif/esp-idf/files/15219096/patch_for_v4.4.7.tar.gz)

wmy-espressif, 2024-05-20:
Fix for branches release/v5.1, release/v5.0 and release/v4.4  were merged last week and are expected to be synchronized to github within serveral days. The following table lists the commits of the fixes:

| branch | commit |
| -- | -- |
| master | [d4552c6274](https://github.com/espressif/esp-idf/commit/d4552c6274ac8119acef610b34276fe56dc2a4fd) |
| release/v5.2 | [dc3e9531c0](https://github.com/espressif/esp-idf/commit/dc3e9531c0333652aea236f5cfb663d707c460c8): expected in v5.2.2 |
| release/v5.1 | a79bd17e39910a7b659466dd99262584928a1b18: expected in v5.1.5 |
| release/v5.0 | 6a17b9c1c7ab8d0d0df7244276e795acb8f98156: expected in v5.0.7 |
| release/v4.4 | 9160abc5c5e5e4162278109dd217450eef7b4bfa: expected in v4.4.8 |