espressif/esp-idf #10428: Wrong chip type detected when using SPIRAM support and virtual efuses

listout, 2022-12-22:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4.3

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-PICO-DevKitM-2

### Power Supply used.

USB

### What is the expected behavior?

I was testing Secure boot (version 2) and flash encryption, to reduce the chances of bricking my device I decided to use virtual efuses (stored in flash). With all these three options enabled I can't run any application (in this case hello_world example from v4.4.3); the device would boot showing device informations and in the **psram** section it would show **This chip is ESP32-D0WD** whereas my chip is ESP32 Pico Mini V3 02, and after some time the device would reboot. 

When I turn off SPIRAM support, the code/application runs as expected.

### What is the actual behavior?

Application run and correct chip is detected with secure boot, flash encryption and virtual efuses enabled.

### Steps to reproduce.

1. Clone/checkout v4.4.3 tag of ESP-IDF
2. Create/Copy a project
3. Run menuconfig and change the following settings (I've attached the minimal config below)
4. Change chip revision to 3
5. Turn on secure boot version 2
6. Turn on flash encryption (development mode)
7. Enable virtual efuses and keep efuses in flash

### Debug Logs.

```plain
W (91) boot.esp32: eFuse virtual mode is enabled. If Secure boot or Flash encryption is enabled then it does not provide any security. FOR TESTING ONLY!
I (46) boot: ESP-IDF v4.4.3 2nd stage bootloader
I (46) boot: compile time 17:47:15
I (46) boot: chip revision: 3
I (49) boot.esp32: SPI Speed      : 40MHz
I (54) boot.esp32: SPI Mode       : DIO
I (58) boot.esp32: SPI Flash Size : 8MB
I (63) boot: Enabling RNG early entropy source...
I (68) boot: Partition Table:
I (72) boot: ## Label            Usage          Type ST Offset   Length
I (79) boot:  0 nvs              WiFi data        01 02 00023000 00005000
I (87) boot:  1 otadata          OTA data         01 00 00028000 00002000
I (94) boot:  2 phy_init         RF data          01 01 0002a000 00001000
I (101) boot:  3 factory          factory app      00 00 00030000 00200000
W (109) efuse: Loading virtual efuse blocks from real efuses
EFUSE_BLKx:
0) 0x00000000 0xfb6cf2ac 0x00aeac0b 0x0000ad00 0x00001334 0x0015de26 0x00000004 
[... 253 lines cut ...]
    /* Print chip information */
    esp_chip_info_t chip_info;
    esp_chip_info(&chip_info);
    printf("This is %s chip with %d CPU core(s), WiFi%s%s, ",
            CONFIG_IDF_TARGET,
            chip_info.cores,
            (chip_info.features & CHIP_FEATURE_BT) ? "/BT" : "",
            (chip_info.features & CHIP_FEATURE_BLE) ? "/BLE" : "");

    printf("silicon revision %d, ", chip_info.revision);

    printf("%dMB %s flash\n", spi_flash_get_chip_size() / (1024 * 1024),
            (chip_info.features & CHIP_FEATURE_EMB_FLASH) ? "embedded" : "external");

    printf("Minimum free heap size: %d bytes\n", esp_get_minimum_free_heap_size());

    for (int i = 10; i >= 0; i--) {
        printf("Restarting in %d seconds...\n", i);
        vTaskDelay(1000 / portTICK_PERIOD_MS);
    }
    printf("Restarting now.\n");
    fflush(stdout);
    esp_restart();
}
```

listout, 2022-12-23:
Any help or ideas?

mahavirj, 2022-12-26:
@listout 

Probably `esp_efuse_get_pkg_ver` refers to the virtual EFuse block and reads an incorrect version as highlighted below:

https://github.com/espressif/esp-idf/blob/9ee3c8337d3c4f7914f62527e7f7c78d7167be95/components/efuse/src/esp_efuse_utility.c#L264-L268

As highlighted in the help for `CONFIG_EFUSE_VIRTUAL`, this features is primarily for the testing purpose and it is difficult to ensure functionality of all features with this config. 

I would recommend that you try our [qemu port](https://github.com/espressif/qemu/wiki) for verifying security features on ESP32.

Hope this helps!

listout, 2022-12-26:
Thank you @mahavirj. I've been using qemu for testing secure boot, unfortunately for flash encryption, it's not possible and the wiki says info added later.

mahavirj, 2022-12-26:
Thanks for the feedback! I will update the wiki for the flash enc instructions and notify you.

closed 2022-12-26 by listout